gphoto/libgphoto2 #84: Support for Canon Digital Ixus 180

vollbier, 2016-09-02:
Hi,
the  Canon Digital Ixus 180 is not listed in the supported cameras list. 
I tried the following commands:
```
$  lsusb |grep Canon
Bus 001 Device 012: ID 04a9:32c0 Canon, Inc.
```
```
$ LANG=C gphoto2 --auto-detect
Model                          Port                                            
----------------------------------------------------------
USB PTP Class Camera           usb:001,011
```     
Doing `$ LANG=C gphoto2 --list-all-config > allconfig.txt` resulted in the file
[allconfig.txt](https://github.com/gphoto/libgphoto2/files/452880/allconfig.txt).

Trying `env LANG=C gphoto2 --debug --debug-logfile=my-logfile.txt --capture-image-and-download` did not work. The log file: [my-logfile.txt](https://github.com/gphoto/libgphoto2/files/452882/my-logfile.txt)

Any more testing I can do to help?

msmeissn, 2016-09-04:
all the canon ixus/powershots from the last 5+ years have no capture support over USB in the regular firmware.

over wireless it might work (did not get any success rfeports though) and using the CHDK addon firmware it can work over usb.

I added the ID to our code to recognize it by name, but it will not gain capture abilities  by this.

vollbier, 2016-09-04:
Thanks for the quick feedback.

xl94, 2017-12-28:
Hi where can i find the complete list of all supported cameras? From [http://www.gphoto.org/proj/libgphoto2/support.php](http://www.gphoto.org/proj/libgphoto2/support.php), i see that the Canon Digital Ixus 180 was listed.

msmeissn, 2018-01-01:
That is the list of the known camera USB ids we have in the support.php page.

In general we support all PTP cameras, even if not listed by the generic class handling.

msmeissn, 2018-11-03:
can you try update to libgphoto2 2.5.20 and then try

gphoto2 --capture-image-and-download

? It seems this camera reports capture opcodes ...

vollbier, 2018-11-04:
In `Debian/unstable` there's only `gphoto2 2.5.17-1` and `libpghoto2-6 2.5.20-2`, so I build `gphoto2` from source but used libgphoto2 from Debian:

    $ ./gphoto2 --version
    gphoto2 2.5.20
    
    Copyright (c) 2000-2018 Lutz Mueller and others
    
    gphoto2 comes with NO WARRANTY, to the extent permitted by law. You may
    redistribute copies of gphoto2 under the terms of the GNU General Public
    License. For more information about these matters, see the files named COPYING.
    
    This version of gphoto2 is using the following software versions and options:
    gphoto2         2.5.20         gcc, popt(m), exif, cdk, aa, jpeg, readline
    libgphoto2      2.5.20         all camlibs, gcc, ltdl, EXIF
    libgphoto2_port 0.12.0         iolibs: disk ptpip serial usb1 usbdiskdirect usbscsi, gcc, ltdl, USB, serial without locking

Trying to capture an image resulted in:

    $ sudo env LANG=C ./gphoto2 --debug --debug-logfile=my-logfile.txt --capture-image-and-download
                                                                                   
    *** Error ***              
    Canon EOS Half-Press failed (0x2019: PTP Device Busy)
    ERROR: Could not capture image.
    ERROR: Could not capture.
    *** Error (-110: 'I/O in progress') ***       

I attached the [my-logfile.txt](https://github.com/gphoto/libgphoto2/files/2545934/my-logfile.txt). 
Hope this helps.

msmeissn, 2019-03-09:
I have added PowerShot / IXUS Canon capture support for some brands in current releases, can you check libgphoto2 2.5.22?

YO3HCM, 2019-06-05:
hi msmeissn

I have tried to install on my raspberry virtual machine latest libgphoto and gphoto (2.5.23) from source, ( using these instructions: https://hyfrmn.wordpress.com/2015/02/03/install-libgphoto2-and-gphoto2-from-source-on-raspberry-pi/  but 2.5.23 versions for both libgphoto2 and gphoto2... )
when I run gphoto2 -v   .... gphoto2 is version 2.5.23... but libgphoto2 is version  2.5.12 ..not .23 as I was expecting. 

Unfortunately we know already IXUS180 will not work with this version.

Do you have any recommendation for a linux noob like me how to proceed to install the latest libgphoto2 please on my raspberry 

Thank you very much

Andrei