gphoto/libgphoto2 #85: ptp_sony_9280() causes Sony DSC-HX100V rest

supernobby, 2016-09-03:
Hello!
I have a Sony DSC HX100V and tried gphoto2 2.5.10 form Arch Linux to get the images from it via PTP. However, this fails. The camera does seem to make some kind of reset. In the debug log the last thing that happens is "Sending PTP_OC 0x9280 (Unknown PTP_OC) data..." (probably added for issue #16) and then USB hangs. See the attached Sony_DSC-HX100V-logfile.txt for reference:

[Sony_DSC-HX100V-logfile.txt](https://github.com/gphoto/libgphoto2/files/453526/Sony_DSC-HX100V-logfile.txt)

I then got the source and managed rebuild the ptp2.so which having the calls to ptp_sony_9280() and ptp_sony_9281() disabled. Like that, things seem to work fine, I can e. g. list the files on the camera.

I have so far no knowledge, what ptp_sony_9280() really does. But my DSC-HX100V does not seem to like it, even while claiming, that it supports it. So, I wonder, if something can be done about it.

Andreas

msmeissn, 2016-09-07:
hmm, thats not so good :(

abrodkin, 2017-01-09:
So it looks like Sony NEX-3n is affected by similar issue. Is there a need for more logs or any other actions from my side?

mleise, 2017-04-05:
Sony DSC-HX9V seems to have issues with the command as well: [Sony_DSC-HX9V.txt](https://github.com/gphoto/libgphoto2/files/900965/Sony_DSC-HX9V.txt)
(using libgphoto2-2.5.12)
Unlike the OP nothing hangs. The camera just resets and becomes available again via MTP/PTP.

msmeissn, 2020-02-29:
I blank out the HX and NEX from sending this opcode combination now

closed 2020-02-29 by msmeissn