gphoto/libgphoto2 #87: gp_camera_capture() taking more than one photo Sony ILCE-5000

lluk12, 2016-09-13:
Is it normal that when setting the Sony ILCE-5000 in Spd. Priority Cont or Cont. Shooting it takes more than one photo?

Instead of taking one photo and continuing the execution, it takes two really fast, so when I want to get another picture, I get the second photo instead of taking a new shot.

The problem is that, in Single Shooting mode I get 2second delay between images, but in the other modes, I get 2 images with less than 0'5 seconds between them, and then 1 second gap until the next two photos.

Thanks!

msmeissn, 2016-09-14:
can you test current git, as the strategy for sony capture was changed?

lluk12, 2016-09-14:
Yes, just installed the new version. I'll give you feedback as soon as I can try it (I don't have the camera right now).

lluk12, 2016-09-14:
It does work now. When I call gp_camera_capture() it just makes one photo. Apart from that, it's much more easy to control the times now.

Thanks for the quick fix.

msmeissn, 2016-09-14:
the summary has "Sony Capture" listed ...are you expecting something different ? 

thanks to @elijahparker for the sony improvements :)

closed 2016-09-15 by lluk12

lluk12, 2016-09-16:
Note that "Speed Priority Cont" mode still has the same issue, even increasing the usleep() time to 2 seconds in line 3675