gphoto/libgphoto2 #89: gp_camera_autodetect() finding 'ghost' camera

lluk12, 2016-09-15:
Recently, I downloaded the last version of libgphoto2 and it works fine in my computer. When trying to port my program to a Tegra TK1 I come across some problems.

I follow the steps from the INSTALL file, to install the library and I compile my program with the command: g++ main.cpp -lgphoto2 -std=c++11
When running my program after installing the library, a camera appears in usb:001,001 although it doesn't exist even when running the command lsusb -t. No more cameras are detected, and I'm not able to take any photos (although my camera is connected).
When uninstalling the library (sudo make uninstall) and running the last compiled program, I can find the correct cameras! But when testing with different cameras some take photos and some don't. A Sony A5000 won't take photos, while a Nikon Coolpix S2700 does.

Then, I tried using gphoto2 directly, and it does detect the cameras correctly weather the libgphoto2 is installed or not, but can only take photos with the Sony A5000 when the library is installed, otherwise it says that the operation is not supported.

I believe that fixing knowing where this problems are coming from will help me understand why it doesn't work:
1- When listing the cameras, the Tegra TK1 shows a Nikon DSC in port usb:001,001 which I don't have connected (I don't even own it). But when listing the cameras after uninstalling libgphoto2, I can find my cameras.

2- gphoto2 always find the cameras, but can only take pictures from the Sony A5000 when the library is installed, although it does take photos from the Nikon Coolpix when it's not.

Code to list the cameras:
```
CameraList *list;
GPContext *context;
context = gp_context_new();
gp_list_new (&list);
int count = gp_camera_autodetect(list, context);
std::cout << "Count: " << count << std::endl;

for (int i=0; i < gp_list_count(list); i++) {
    const char *name, *value;
    gp_list_get_name(list, i, &name);
    gp_list_get_value(list, i, &value);
    std::cout << name << std::endl;
    std::cout << value << std::endl;
}
gp_list_free (list);
```
Code to init the camera:
```
gp_camera_new(&camera_);
context_ = gp_context_new();

int retval = gp_camera_init(camera_, context_);
if(retval != GP_OK)
{
	std::cout << "Error: " << gp_result_as_string(retval) << std::endl;
}
else
{
	std::cout << "Camera initialized" << std::endl;
}
```
Code to take photos:
```
CameraFilePath camera_file_path;
int retval = gp_camera_capture(camera_, GP_CAPTURE_IMAGE, &camera_file_path, context_);

if(retval)
{
    std::cout << "Capture error: " << gp_result_as_string(retval) << std::endl;
}
else
{
    std::cout << "Pathname on the camera: " << camera_file_path.folder << camera_file_path.name << std::endl;
}
```

Keep in mind that this works perfectly in my Ubuntu 14.04 x64 laptop, but not in the Tegra TK1 with Ubuntu 14.04.

Thanks!

msmeissn, 2016-09-15:
What you see is the builtin virtual camera of libgphoto2.

It only becomes available if you built without LIBUSB support.

gphoto2 --version

will show "no USB".

Install libusb development packages reconfigure libgphoto2 and rebuild.

lluk12, 2016-09-16:
Oh okey, good to know. Installed the libusb-dev package and it works fine now.

closed 2016-09-16 by lluk12