gphoto/libgphoto2 #90: Unable to build on Mac OS X

seanfuture, 2016-09-19:
Via git clone https://github.com/gphoto/libgphoto2 and then running `autoreconf --install --symlink` .. and then `./configure --prefix=/usr/local`, the following error is seen:

```
checking that we can compile and link with libltdl... no
configure: error: cannot compile and link against libltdl
libgphoto2 requires libltdl (the libtool dl* library),
but cannot compile and link against it.
Aborting.
```

Also per https://github.com/gphoto/libgphoto2/issues/19, tried downloading from http://sf.net/projects/gphoto/, then unpackaging and performing the same steps received the same error message.  Version from the tarball was **libgphoto2-2.5.10**.

Results from autoreconf: https://gist.github.com/seanfuture/720f56e6e0d8877fd2d990422ed5fa44
Results from configure: https://gist.github.com/seanfuture/53dab80b5d2c2728a7992caa7f5f3296

Any insight appreciated

msmeissn, 2016-09-29:
cant you install the libtool helper libraries?

seanfuture, 2016-09-29:
libtool itself was installed via `xcode-select --install` however, reports such as this one seem to indicate that the "devel" version is needed in order to resolve this error:  http://superuser.com/questions/567233/configure-gphoto2-error-libgphoto2-requires-libltdl-the-libtool-dl-library .. Can this be confirmed and/or can the documentation related to "libtool  (possibly from -dev or -devel package)" in https://github.com/gphoto/libgphoto2/blob/master/INSTALL be expanded upon?

Will research in the background when time permits getting libtool dev/devel installed w/o interfering highly w/ the existing Xcode command line utilities

Thank you kindly, @msmeissn

Sollex-21412, 2016-12-17:
Got the same error during configure.

OS: Mac OS El Capitan
libgphoto: latest from git

haysclark, 2018-01-18:
# Installation on OSX
> Successful tested with commit [1042987ae](https://github.com/gphoto/libgphoto2/commit/1042987ae2ccb071711c47e7d4052c00adae706a) using OSX 10.13.2 (High Sierra).

**Requirements:** Homebrew
**Resource:** Homebrew's [libgphoto2 formula](https://github.com/Homebrew/homebrew-core/blob/master/Formula/libgphoto2.rb)

1. Install [Homebrew](https://brew.sh/)
 ```/usr/bin/ruby -e "$(curl -fsSL https://raw.githubusercontent.com/Homebrew/install/master/install)"```

1. Install _libgphoto2_ compile dependencies
```brew install automake autoconf gettext pkg-config libtool libusb-compat gd libexif```

> Note: See Homebrew output when installing _gettext_, some version of OSX require that _gettext_ be added to PATH, like so:
```export PATH="/usr/local/opt/gettext/bin:$PATH"```

3. Clone _libgphoto2_ repo
```git clone git@github.com:gphoto/libgphoto2.git```

1. Change directory to _libgphoto2_
```cd libgphoto2```

1. Autoreconfing packages
```autoreconf -fvi```

1. Configure and verify output and settings
```./configure --prefix=/usr/local```

1. Make and install _libgphoto2_
```make install```

For any additional help make sure to read the official [INSTALL](https://github.com/gphoto/libgphoto2/blob/master/INSTALL) help file.

@msmeissn Consider **closing** this issue.

msmeissn, 2018-01-19:
i think we have a solution...closing

closed 2018-01-19 by msmeissn

seth-cohen, 2019-01-29:
One other step is required here (at least on MacOS High Sierra). `autopoint` is added via `gettext` formula but is not symlinked into /usr/local to prevent bins provided from `gettext` colliding with those provided by the OS. But the OS doesn't provide autopoint. So...

```
ln -s `brew ls gettext | grep bin/autopoint` /usr/local/bin  
```

Works for me.