gphoto/libgphoto2 #91: How to set AFL(autofocus locking) in libgphoto2

lwj0216, 2016-09-21:
Hi,  I am developing a application and using Nikon D3300 controlled by libgphoto2 on ubuntu14.04. I found that the camera couldn't auto focus successfully in the darkness, so that the method capture_to_file() return failed code and  print error message: "Context Error, Out Of Focus". I wonder if there is a way to set AFL by linux shell command or c/c++ code. My idea is that calling AFL method when calling autofocus method return GP_OK in brightness and then turn to darkness so the camera can take a picture successfully.
Thank you!

msmeissn, 2016-09-29:
there is the "aelaflmode" config variable, ... However it seems as if the D3300 does not expose it via its API . :/

lwj0216, 2016-11-11:
Thank you for your reply and i am so sorry for that i didn't notice your anwser. It has been confirmed that the D3300 does not support AFL via its API. So i tried to capture the image in LiveView mode and it did work for me. But now i meet some problem when i using gp_camera_capture_preview to get preview image data in a while loop and i get the error message below:
// error log start
errordumper error:Writing 16 = 0x10 bytes to port failed: Error writing to the port (-35)
errordumper error:PTP_OC 0x1015 sending req failed: Error writing to the port (-35)
errordumper error:Writing 16 = 0x10 bytes to port failed: Error writing to the port (-35)
errordumper error:PTP_OC 0x1015 sending req failed: Error writing to the port (-35)
errordumper error:Writing 16 = 0x10 bytes to port failed: Error writing to the port (-35)
errordumper error:PTP_OC 0x1016 sending req failed: Error writing to the port (-35)
errordumper error:Writing 16 = 0x10 bytes to port failed: Error writing to the port (-35)
errordumper error:PTP_OC 0x1016 sending req failed: Error writing to the port (-35)
errordumper error:'ptp_setdevicepropvalue (params, PTP_DPC_NIKON_RecordingMedia, &value, PTP_DTC_UINT8)' failed: PTP I/O Error (0x02ff)
errordumper error:Writing 12 = 0xc bytes to port failed: Error writing to the port (-35)
errordumper error:PTP_OC 0x9201 sending req failed: Error writing to the port (-35)
errordumper error:Writing 12 = 0xc bytes to port failed: Error writing to the port (-35)
errordumper error:PTP_OC 0x9201 sending req failed: Error writing to the port (-35)
errordumper error:'ret' failed: Nikon enable liveview failed (0x02ff: PTP I/O Error)
errordumper error:Nikon enable liveview failed (0x02ff: PTP I/O Error)

*** Contexterror ***              
Nikon enable liveview failed (0x02ff: PTP I/O Error)
errordumper error:'camera->functions->capture_preview ( camera, file, context)' failed: -7
// error log end
The code is sometimes normal but sometimes i just meet the error presented above.
That is my code snippet:
while(this->mCaptureFlag)
{
	CameraFile *file;
	char * imageData;
	retval = gp_file_new(&file);
	retval = gp_camera_capture_preview(this->mCamera, file, this->mContext);
	if (retval != GP_OK)
	{
		Log::getInstance()->error(TAG, "gp_camera_capture_preview failed...");
		break;
	}
	else
	{
		retval = gp_file_get_data_and_size(file, (const char**)&imageData, &imageSize);
		if(retval != GP_OK)
		{
			Log::getInstance()->error(TAG, "gp_file_get_data_and_size failed...");
		}
		char *data = (char *) malloc(imageSize);
		memcpy(data, imageData, imageSize);
		gp_file_unref(file);

		if(this->isFinishedFrame)
		{
			this->isFinishedFrame = false;
			emit signalSendToVideoThread(data, imageSize);
		}
		else
		{
			delete data;
		}
	}
	QThread::msleep(20);
}

I wonder if i didn't use the api well, thus the application went wrong ? Thank you for your reply.