gphoto/libgphoto2 #92: 5D Mark iV support

cool-benson, 2016-09-24:
Hi I get to access the new 5D Mark iV, and I found it's working not very well. It seems that gphoto2 didn't close end the procedure correctly. For example, when I try gphoto --capture-image, the camera focus, than shutter snap, and than every thing just freeze. But --capture-preview works for the first time, when called the second time, it gives error "ptp operation not supported". Maybe I'm doing some thing wrong please advise. My configuration is raspberry pi/raspbian  + gphoto-updater. Thanks in advance, I can also help to write the code, but I will need some pointers.

following is lsusb -v output

> Bus 001 Device 008: ID 04a9:3281 Canon, Inc. 
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.10
  bDeviceClass            0 (Defined at Interface level)
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x04a9 Canon, Inc.
  idProduct          0x3281 
  bcdDevice            0.02
  iManufacturer           1 Canon Inc.
  iProduct                2 Canon Digital Camera
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           39
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xc0
      Self Powered
    MaxPower                2mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           3
      bInterfaceClass         6 Imaging
      bInterfaceSubClass      1 Still Image Capture
      bInterfaceProtocol      1 Picture Transfer Protocol (PIMA 15470)
      iInterface              0 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x02  EP 2 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval              10
Binary Object Store Descriptor:
  bLength                 5
  bDescriptorType        15
  wTotalLength           22
  bNumDeviceCaps          2
  USB 2.0 Extension Device Capability:
    bLength                 7
    bDescriptorType        16
    bDevCapabilityType      2
    bmAttributes   0x0000090e
      Link Power Management (LPM) Supported
  SuperSpeed USB Device Capability:
    bLength                10
    bDescriptorType        16
    bDevCapabilityType      3
    bmAttributes         0x00
    wSpeedsSupported   0x000e
      Device can operate at Full Speed (12Mbps)
      Device can operate at High Speed (480Mbps)
      Device can operate at SuperSpeed (5Gbps)
    bFunctionalitySupport   1
      Lowest fully-functional device speed is Full Speed (12Mbps)
    bU1DevExitLat          10 micro seconds
    bU2DevExitLat        2047 micro seconds
Device Status:     0x0001
  Self Powered

cool-benson, 2016-09-26:
gphoto2 --summary
Camera summary:                                                                
Manufacturer: Canon.Inc
Model: Canon EOS 5D Mark IV
  Version: 3-1.0.1
Vendor Extension ID: 0xb (1.0)

Capture Formats: JPEG
Display Formats: Association/Directory, Script, DPOF, MS AVI, MS Wave, JPEG, CRW, Unknown(b103), Unknown(bf02), Defined Type, Unknown(b104), Unknown(b105)

Device Capabilities:
	File Download, File Deletion, File Upload
	No Image Capture, No Open Capture, Canon EOS Capture, Canon EOS Shutter Button
	Canon Wifi support

Storage Devices Summary:

Device Property Summary:
Property 0xd402:(read only) (type=0xffff) 'Canon EOS 5D Mark IV'
Property 0xd407:(read only) (type=0x6) 1
Property 0xd406:(readwrite) (type=0xffff) 'Unknown Initiator'
Property 0xd303:(read only) (type=0x2) 1
Battery Level(0x5001):(read only) (type=0x2) Enumeration [100,0,75,0,50] value: 78% (78)

msmeissn, 2016-09-29:
can you try current git ? some enhancements were done in current git which have not been released

cool-benson, 2016-10-03:
I think the only difference I notice is when I use gphoto --summary, it shows a computer on the camera screen. -capture-and-download did fire the shot but program did not shutdown properly.
I installed the newest one, using gphoto-update -d, following is the version information. 
 
gphoto2 2.5.10.1

Copyright (c) 2000-2016 Lutz Mueller and others

gphoto2 comes with NO WARRANTY, to the extent permitted by law. You may
redistribute copies of gphoto2 under the terms of the GNU General Public
License. For more information about these matters, see the files named COPYING.

This version of gphoto2 is using the following software versions and options:
gphoto2         2.5.10.1       gcc, popt(m), exif, no cdk, no aa, no jpeg, no readline
libgphoto2      2.5.10.1       all camlibs, gcc, ltdl, EXIF
libgphoto2_port 0.12.0         gcc, ltdl, USB, serial without locking

--------------------------------------------------------------------------------------
--auto-detect
Model                          Port                                            
----------------------------------------------------------
Canon EOS 5D Mark IV           usb:001,009 
--------------------------------------------------------------------------------------
--summary
Camera summary:                                                                
Manufacturer: Canon.Inc
Model: Canon EOS 5D Mark IV
  Version: 3-1.0.1
  Serial Number: <deleted by me>
Vendor Extension ID: 0xb (1.0)

Capture Formats: JPEG
Display Formats: Association/Directory, Script, DPOF, MS AVI, MS Wave, JPEG, CRW, Unknown(b103), Unknown(bf02), Defined Type, Unknown(b104), Unknown(b105)

Device Capabilities:
	File Download, File Deletion, File Upload
	No Image Capture, No Open Capture, Canon EOS Capture, Canon EOS Shutter Button
	Canon Wifi support

Storage Devices Summary:
store_00010001:
	StorageDescription: CF
	VolumeLabel: None
	Storage Type: Removable RAM (memory card)
	Filesystemtype: Digital Camera Layout (DCIM)
	Access Capability: Read-Write
	Maximum Capability: 256020316160 (244160 MB)
	Free Space (Bytes): 234248208384 (223396 MB)
	Free Space (Images): -1

Device Property Summary:
Property 0xd402:(read only) (type=0xffff) 'Canon EOS 5D Mark IV'
Property 0xd407:(read only) (type=0x6) 1
Property 0xd406:(readwrite) (type=0xffff) 'Unknown Initiator'
Property 0xd303:(read only) (type=0x2) 1
Battery Level(0x5001):(read only) (type=0x2) Enumeration [100,0,75,0,50] value: 95% (95)

msmeissn, 2016-10-05:
what does "did not shutdown correctly" mean?

cool-benson, 2016-10-05:
Basically the program do take the picture, but the picture is not downloaded to computer, and the program halt until I press control+c. I've made a video to show the problem([link](https://1drv.ms/v/s!AoDR4RutsuCJ5wfzsm7dKpCxLfCz)) I've tried capture-image, capture-image-and-download, and capture-preview. Only the capture-preview worked. Please turn the sound on, so that you can hear the shutter sound.

cool-benson, 2016-10-05:
I have access to a 80D also, and the problem reoccurs. I think maybe we are dealing with same problem here with [#60](https://github.com/gphoto/libgphoto2/issues/60)

odedcomay, 2016-10-13:
Does not work for me either. It complains about an unsupported PTP operation:

LANG=C ./gphoto2/gphoto2 --debug --debug-logfile=summary-debug.txt --auto-detect --summary --version
gphoto2 2.5.10.1

Copyright (c) 2000-2016 Lutz Mueller and others

gphoto2 comes with NO WARRANTY, to the extent permitted by law. You may
redistribute copies of gphoto2 under the terms of the GNU General Public
License. For more information about these matters, see the files named COPYING.

This version of gphoto2 is using the following software versions and options:
gphoto2         2.5.10.1       gcc, popt(m), exif, no cdk, no aa, jpeg, readline
libgphoto2      2.5.10.1       all camlibs, gcc, ltdl, EXIF
libgphoto2_port 0.12.0         gcc, ltdl, USB, serial lockdev locking
Model                          Port                                            
----------------------------------------------------------
Canon EOS 5D Mark IV           usb:003,023     

*** Error ***              
PTP Operation Not Supported

*** Error ***              
An error occurred in the io-library ('Unsupported operation'): No error description available
*** Error (-6: 'Unsupported operation') ***       

[lsusb.txt](https://github.com/gphoto/libgphoto2/files/527160/lsusb.txt)
[summary-debug.txt](https://github.com/gphoto/libgphoto2/files/527175/summary-debug.txt)

elijahparker, 2016-10-13:
This is just a guess, but it seems many of the newer Canon cameras with WiFi require it to be disabled in order for the USB interface to work properly.  Has anyone tried with WiFi disabled on the camera?

cool-benson, 2016-10-13:
Just tried, WiFi disabled didn't work, same problem.

elijahparker, 2016-10-13:
Thanks for checking!  I'm going to need this to work with the Mark IV as well -- I might rent one soon to do some debugging if there aren't any other developers here with access to one.

cool-benson, 2016-10-13:
Actually I would like to help, I'll try capture usb packets, it would be great if you can give some pointers.

cool-benson, 2016-10-13:
this is the log after running gphoto2 --capture-image &>log.txt [log.txt](https://github.com/gphoto/libgphoto2/files/528290/log.txt)

msmeissn, 2016-10-14:
thanks for the trace.

1.192808 ptp                         (2): event 2: unknown EOS event c1a9

thats the key problem, a new event which we currently do not parse. I will see if i can work on this after the weekend

msmeissn, 2016-10-23:
what kind of picture did you take there? was it around 3668283 bytes long? JPEG or RAW?

msmeissn, 2016-10-23:
can you capture a debug trace of a capture and one of -L afterwards perhaps, so I can see the size and potentially other magic values?

msmeissn, 2016-10-23:
hmm, might be easy ... if you could test current GIT?

cool-benson, 2016-10-25:
Hi msmeissn, I'm happy to report -capture-image-and-download worked:-), for the first attempt. What happens now is that when I try capture-image-and-download for the first time it worked flawlessly, but on the second time it's not working any more.

cool-benson, 2016-10-25:
This is the log for first one, the successful one.
[log1.txt](https://github.com/gphoto/libgphoto2/files/551757/log1.txt)
This is the log for second one, the one that failed.
[log2.txt](https://github.com/gphoto/libgphoto2/files/551758/log2.txt)

This is the -L
pi@raspberrypi:~/Documents/gphoto $ gphoto2 -L
There is no file in folder '/'.                                                
There is no file in folder '/store_00010001'.
There is no file in folder '/store_00010001/DCIM'.
There are 15 files in folder '/store_00010001/DCIM/100EOS5D'.
#1     FD9A0233.JPG               rd  6054 KB image/jpeg
#2     FD9A0234.MOV               rd 2397276 KB application/x-unknown
#3     FD9A0235.MOV               rd 2265163 KB application/x-unknown
#4     FD9A0238.MOV               rd 927780 KB application/x-unknown
#5     FD9A0243.MOV               rd 1039230 KB application/x-unknown
#6     FD9A0244.MOV               rd 1199933 KB application/x-unknown
#7     FD9A0245.MOV               rd 737067 KB application/x-unknown
#8     FD9A0247.MOV               rd 1447956 KB application/x-unknown
#9     FD9A0248.MOV               rd 1174296 KB application/x-unknown
#10    FE0A0239.MOV               rd 918276 KB application/x-unknown
#11    FV2A0236.MOV               rd 302310 KB application/x-unknown
#12    FV2A0237.MOV               rd 552920 KB application/x-unknown
#13    JY4A0240.MOV               rd 1132730 KB application/x-unknown
#14    JY4A0241.MOV               rd 1921110 KB application/x-unknown
#15    JY4A0246.MOV               rd 1361671 KB application/x-unknown

btw I'm more interested in movie capture, and it didn't work.

stloeffler, 2016-10-26:
I see a similar improvement on the 80D which apparently suffers from similar problems, i.e. capture and download works with the latest patch.
Regarding the "only works once" problem: it seems to me that there is some problem with session management. It does not seem to be related to image capture, it also occurs when running `gphoto2 --summary` twice. To debug further, I wrote a small python script to test several different approaches. Here are my findings:
 * In a normal session (i.e. `0x1002 (Open session)`, maybe some commands, `0x1003 (Close session)`), after `0x1003 (Close session)` is sent, the camera seems to stop responding properly to any further request. Future `0x1002 (Open session)` commands (as well as others) fail with `0x2005 not supported` (even though it shouldn't do that, according to the PIMA specs). It does not matter whether these commands are sent from the same process/instance or a new one.
 * Waiting for the camera to switch to standy/off and reactivating it apparently resets everything to proper condition, as does switching the camera off and on again manually or unplugging and replugging the USB cable
 * Sending `0x1003 (Close session)` without opening a session first properly returns `0x2003 session not open` and normal operation continues.
 * Never sending `0x1003 (Close session)` seems to work. I.e., sending `0x1002 (Open session)` followed by `0x9114 (Canon set remote)` multiple times seems to work
 * Performing a device reset (like ptp_resetdevice()) instead of `0x1003 (Close session)` seems to work as well (though not performing reset after close session)

The last two points I also tested by modifying camera_exit() in camlibs/ptp2/library.c and both variants seem to work (at least) with `gphoto2 --summary` being run multiple times in a row.

msmeissn, 2016-10-26:
the second logfile is basically "failed to open session" right at the start, which is weird

msmeissn, 2016-10-26:
close session works for all other cameras, so i wonder why it suddenly stops with new EOS :(

stloeffler, 2016-10-30:
I agree that `close session` should work. However, I have done some investigation under Windows 8.1 (using Microsoft Message Analyzer) and I have never seen a `close session` being sent. When I connect the camera, EOSUtility automatically starts and sends `open session`. However, neither "closing" the EOSUtility (which just minimizes it to the tray or something) nor "quitting" the tray program actually send `close session`. Indeed, restarting the EOSUtility (without reconnecting the camera) just happily resumes where previous instances have left off, including incrementing the transaction number from the previous session (i.e., if the last transaction of the first instance was 100, the first transaction of the new one is 101). Apparently, the session is managed somewhere in the background by some system process (which I could not identify so far, unfortunately). 

If the camera switches off or is reconnected, everthing starts from the beginning with `open session` with the same session ID (0x01). Once a session is open, all instances seem to start with `set remote mode (0x9114) to 1`, followed by `set event mode (0x9115) to 1` and finish with `set event mode (0x9115) to 0` and `set remote mode (0x9114) to 0`

msmeissn, 2016-10-30:
the initialization we do the same.

windows usually opens a PTP device once and never closes it, the lifetime is managed by Windows.

In Linux, libgphoto2 and its frontends usually expect to open and close the device multiple times. (exception is gvfs which tries to open it once and keep it open.)

axxel, 2016-11-28:
I'd like to confirm that a trivial change like adding the line
`		if (!params->deviceinfo.Model || strcmp(params->deviceinfo.Model, "Canon EOS 5D Mark IV"))
`
in front of
`		        ptp_closesession (params);
`
in `camera_exit()` fixes this particular issue of reopening the EOS 5Dm4 with gphoto2.

I can easily imagine how the canon firmware developers might have overlooked this regression if their windows use case simply never closes the session :-/.

msmeissn, 2016-11-30:
I will probalby need to add a flag to the camera list which will avoid closing the session

msmeissn, 2016-12-04:
can you test this patch?

kentavv, 2016-12-04:
At least on the Canon 80D, below is the message that is generated by gphoto2. Also, the small top facing LCD screen on the 80D shows "Err 70" and the battery must be removed before the error message is cleared. Toggling power switch was not enough. On the plus side, there is an image saved, sometimes, but not always. Didn't seem to be a pattern to when it would succeed or not. Thank you

[kent@localhost gphoto2]$ gphoto2 --capture-image-and-download
                                                                               
*** Error ***              
PTP General Error
ERROR: Could not capture image.
ERROR: Could not capture.

*** Error ***              
PTP I/O Error
[kent@localhost gphoto2]$

msmeissn, 2016-12-04:
The Err 70 would happen if you run --set-config capturetarget=XXX ?

kentavv, 2016-12-04:
Sorry to be dense, but do you mean as in the following?

[kent@Unknown gphoto]$ gphoto2 --set-config capturetarget=XXX --capture-image-and-download
                                                                               
*** Error ***              
PTP General Error
ERROR: Could not capture image.
ERROR: Could not capture.

msmeissn, 2016-12-04:
I meant, there is a bug in 2.5.11 and 2.5.10 which will cause Err 70 when you do --set-config capturetarget=xxx ...  It is fixed in current GIT master.

if you just and only do --capture-image-and-download   ... does it continously work?

kentavv, 2016-12-04:
My original message today is after doing a 'git clone' on both libgphoto2 and gphoto2. I verified the libgphoto2 has PTP_DONT_CLOSE_SESSION update for the 80D. The error messages were from the newly compiled versions. I have no other versions on my machine. Let me know specific commands and I would be glad to do further testing. Best wishes, Kent

axxel, 2016-12-05:
Works with the EOS 5D Mark IV (as expected).

cpoetter, 2016-12-05:
Also fixes the just-one-time-shot problem on the EOS 80D. @msmeissn Thanks a lot for solving this issue!

cpoetter, 2016-12-05:
But the Err 70 when I run

$ gphoto2 --set-config iso=8000

is still there. Just removing the battery fixes it.

msmeissn, 2019-03-09:
I think we meanwhile fixed this problem in 2.5.17.

if there are any issues left, please open a new ticket :)

srednax, 2020-02-18:
I'm trying to use this with my 5D IV. No luck. --auto-detect and --abilities outputs what is expected, but no luck capturing an image, or downloading one shooting tethered.

[anders@elias anders]$ env LANG=C gphoto2 --port usb: --capture-tethered --debug
0.000007 main                        (2): ALWAYS INCLUDE THE FOLLOWING LINES WHEN SENDING DEBUG MESSAGES TO THE MAILING LIST:
0.000032 main                        (2): gphoto2 2.5.23
0.000037 main                        (2): gphoto2 has been compiled with the following options:
0.000040 main                        (2):  + gcc (C compiler used)
0.000046 main                        (2):  + popt (mandatory, for handling command-line parameters)
0.000051 main                        (2):  + exif (for displaying EXIF information)
0.000059 main                        (2):  + no cdk (for accessing configuration options)
0.000065 main                        (2):  + no aa (for displaying live previews)
0.000070 main                        (2):  + jpeg (for displaying live previews in JPEG format)
0.000074 main                        (2):  + readline (for easy navigation in the shell)
0.000082 main                        (2): libgphoto2 2.5.23
0.000086 main                        (2): libgphoto2 has been compiled with the following options:
0.000090 main                        (2):  + all camlibs
0.000094 main                        (2):  + gcc (C compiler used)
0.000098 main                        (2):  + ltdl (for portable loading of camlibs)
0.000103 main                        (2):  + EXIF (for special handling of EXIF files)
0.000110 main                        (2): libgphoto2_port 0.12.0
0.000115 main                        (2): libgphoto2_port has been compiled with the following options:
0.000119 main                        (2):  + iolibs: disk ptpip serial usb1 usbdiskdirect usbscsi
0.000123 main                        (2):  + gcc (C compiler used)
0.000127 main                        (2):  + ltdl (for portable loading of iolibs)
0.000131 main                        (2):  + USB (libusb1, for USB cameras)
0.000135 main                        (2):  + serial (for serial cameras)
0.000139 main                        (2):  + no resmgr (serial port access and locking)
0.000143 main                        (2):  + no ttylock (serial port locking)
0.000147 main                        (2):  + no lockdev (serial port locking)
0.000152 main                        (2): CAMLIBS env var not set, using compile-time default instead
0.000156 main                        (2): IOLIBS env var not set, using compile-time default instead
0.000160 main                        (2): invoked with following arguments:
0.000164 main                        (2):   --port
0.000168 main                        (2):   usb:
0.000172 main                        (2):   --capture-tethered
0.000176 main                        (2):   --debug
0.000190 main                        (2): Processing 'port' option ('usb:')...
0.000199 gp_port_info_list_load      (2): Using ltdl to load io-drivers from '/usr/lib/libgphoto2_port/0.12.0'...
0.000250 foreach_func                (2): Called for filename '/usr/lib/libgphoto2_port/0.12.0/disk'.
0.000362 gp_port_library_list        (2): found fstab fsname UUID=9a152f28-7684-4ec5-920d-087cc4ad133f
0.000375 gp_port_library_list        (2): found fstab fsname UUID=6D05-20A2
0.000387 gp_port_library_list        (2): found fstab fsname UUID=94ba31d6-5c9e-47d3-821d-b3cbe297fa04
0.000396 gp_port_library_list        (2): found fstab fsname UUID=57f6d221-9a02-4e21-a16c-556e5a39857f
0.000479 gp_port_library_list        (2): found mtab fsname proc
0.000485 gp_port_library_list        (2): found mtab fsname sys
0.000506 gp_port_library_list        (2): found mtab fsname dev
0.000518 gp_port_library_list        (2): found mtab fsname run
0.000527 gp_port_library_list        (2): found mtab fsname efivarfs
0.000537 gp_port_library_list        (2): found mtab fsname /dev/sda1
0.000543 gp_port_library_list        (2): found mtab fsname securityfs
0.000546 gp_port_library_list        (2): found mtab fsname tmpfs
0.000555 gp_port_library_list        (2): found mtab fsname devpts
0.000558 gp_port_library_list        (2): found mtab fsname tmpfs
0.000568 gp_port_library_list        (2): found mtab fsname cgroup2
0.000572 gp_port_library_list        (2): found mtab fsname cgroup
0.000576 gp_port_library_list        (2): found mtab fsname pstore
0.000583 gp_port_library_list        (2): found mtab fsname none
0.000590 gp_port_library_list        (2): found mtab fsname cgroup
0.000619 gp_port_library_list        (2): found mtab fsname cgroup
0.000623 gp_port_library_list        (2): found mtab fsname cgroup
0.000627 gp_port_library_list        (2): found mtab fsname cgroup
0.000631 gp_port_library_list        (2): found mtab fsname cgroup
0.000635 gp_port_library_list        (2): found mtab fsname cgroup
0.000638 gp_port_library_list        (2): found mtab fsname cgroup
0.000642 gp_port_library_list        (2): found mtab fsname cgroup
0.000645 gp_port_library_list        (2): found mtab fsname cgroup
0.000649 gp_port_library_list        (2): found mtab fsname cgroup
0.000652 gp_port_library_list        (2): found mtab fsname cgroup
0.000656 gp_port_library_list        (2): found mtab fsname systemd-1
0.000660 gp_port_library_list        (2): found mtab fsname hugetlbfs
[... 538 lines cut ...]
20.100360 gp_context_error            (0): error 0x02ff

*** Error ***              
[... cut ...]
60.100861 gp_port_write [gphoto2-port.c:406](0): Writing 16 = 0x10 bytes to port failed: Timeout reading from or writing to the port (-10)
60.100866 ptp_usb_sendreq [usb.c:109] (0): PTP_OC 0x9114 sending req failed: Timeout reading from or writing to the port (-10)
60.100873 camera_prepare_canon_eos_capture [config.c:388](0): 'ptp_canon_eos_setremotemode(params, 1)' failed: PTP Timeout (0x02fa)
60.100907 ptp_usb_sendreq             (2): Sending PTP_OC 0x911d (PTP_OC_CANON_EOS_KeepDeviceOn) request...
60.100917 gp_port_write               (3): Writing 12 = 0xc bytes to port...

^C
Cancelling...
^C

srednax, 2020-02-19:
Updated firmware to 1.2.1, still does not work.

srednax, 2020-02-19:
With updated firmware:
[anders@elias tmp]$ gphoto2 --port usb: --capture-tethered
Venter på hændelser fra kamera. Tryk Ctrl-C for at afbryde.                    
UNKNOWN PTP Property d116 changed
UNKNOWN PTP Property d105 changed
UNKNOWN PTP Property d138 changed
UNKNOWN PTP Property d102 changed
UNKNOWN PTP Property d101 changed
UNKNOWN PTP Property d103 changed
UNKNOWN PTP Property d106 changed
UNKNOWN PTP Property d108 changed
UNKNOWN PTP Property d107 changed
UNKNOWN PTP Property d109 changed
UNKNOWN PTP Property d11b changed
UNKNOWN PTP Property d104 changed
UNKNOWN PTP Property d10c changed
UNKNOWN PTP Property d10b changed
UNKNOWN PTP Property d10e changed
UNKNOWN PTP Property d10d changed
UNKNOWN PTP Property d1d9 changed
UNKNOWN PTP Property d1d5 changed
UNKNOWN PTP Property d1c1 changed
UNKNOWN PTP Property d10f changed
UNKNOWN PTP Property d114 changed
UNKNOWN PTP Property d1b7 changed
^C
Annullerer...
UNKNOWN PTP Property d110 changed

msmeissn, 2020-03-14:
the initial directory read took very long ... is this a very full SD card? Can you use a empty one?

srednax, 2020-03-15:
This was a probably empty, newly formatted CF card. But I can try with a SD. 

lørdag den 14. marts 2020 18.52.55 CET skrev Marcus Meissner:
> the initial directory read took very long ... is this a very full SD card?
> Can you use a empty one?

msmeissn, 2020-03-15:
Hmm, I was thinking the initial read took too long. there is some weirdness there during inuitial directory read definitely.

odeleongt, 2020-05-17:
Hi. I have been trying this too with a Canon 5D Mark IV on firmware version 1.1.2, wifi disabled, auto standby disabled.

The issue was just as described originally when I was using `libgphoto2 2.5.9`, with `gphoto2 --capture-image` resulting in the camera hanging. At least in my case, it seems that the camera does not even write the file. When you try to turn it off, it shows a message "Recording 1 of x files" (you can try manually taking pictures with the camera after the gphoto call, so x can be more than 1). Using `gphoto2 --capture-preview` worked just fine.

I just compiled `libgphoto2 2.5.24` and the issue with `--capture-image` and `--capture-image-and-save` is gone. Horray. Now `--capture-tethered` works too.

Debug trace of for the issue in 2.5.9 attached: [gphoto2-canon-5d4-image-debug.txt](https://github.com/gphoto/libgphoto2/files/4639807/gphoto2-canon-5d4-image-debug.txt)

odeleongt, 2020-05-17:
Now, I also have another issue that seems related to the original solution you provided above.

I am trying to capture video using `gphoto2 --capture-movie` (and piping through ffmpeg to v4l2loopback to use as "webcam"...). Thing is, video is accessible just fine (from `/dev/video1` in my system), but only for a few seconds. Usually less than ~1000 frames are transmitted before the feed hangs.

When I just dump the stream to `--stdout` (without piping to ffmpeg and the loopback device), after the feed halts and gphoto2 stops the console would be filled with `c62;` repeating over and over.

The debug trace in this case, since we got hundreds of frames, is about 300 MB, but I include attached a copy where I removed most of the hexdumps (about 4M lines of them) but left the first few thousand lines (up to the `[hexdump]` text I added in line 18150), and also everything after what looked to be the last correct hexdump: [gphoto2-2.5.9_canon-5d4_capture-movie_debug.txt](https://github.com/gphoto/libgphoto2/files/4639822/gphoto2-2.5.9_canon-5d4_capture-movie_debug.txt)

This trace was using `libgphoto2 2.5.9`, but the problem persists using `libgphoto2 2.5.24`.