gphoto/libgphoto2 #93: PTP I/O Error

lluk12, 2016-09-26:
A few weeks ago, I was testing my custom software with a Sony A5000 camera using libgphoto2 library, and it would take photos correctly.

Today, I tried to use the same software and I can't manage to take any photo, it returns an unspecified error, and when trying to take a capture with gphoto2 I get:
```
$ gphoto2 --capture-image
                                                                               
*** Error ***              
PTP I/O error

*** Error ***              
An error occurred in the io-library ('Unspecified error'): The supplied vendor or product id (0x0,0x0) is not valid.
ERROR: Could not capture image.
ERROR: Could not capture.

*** Error ***              
PTP I/O error

*** Error ***              
An error occurred in the io-library ('Unspecified error'): The supplied vendor or product id (0x0,0x0) is not valid.
```

When I try to print the camera summary, I just get a response the first time I call the function, after that, I just get errors. This is the summary I get the first time:
```
Manufacturer: Sony Corporation
Model: ILCE-5000
  Version: 1.0
  Serial Number: 00000000000000003282653003953039
Vendor Extension ID: 0x11 (1.0)
Vendor Extension Description: Sony PTP Extensions

Capture Formats: 
Display Formats: JPEG, Unknown(b301), Unknown(b101)

Device Capabilities:
	File Download, No File Deletion, No File Upload
	No Image Capture, No Open Capture, No vendor specific capture
```

And the gphoto2 debug which may help:
```
$ gphoto2 --capture-image --debug
0.000019 main(2): ALWAYS INCLUDE THE FOLLOWING LINES WHEN SENDING DEBUG MESSAGES TO THE MAILING LIST:
0.000037 main(2): gphoto2 2.5.3
0.000041 main(2): gphoto2 has been compiled with the following options:
0.000043 main(2):  + gcc (C compiler used)
0.000046 main(2):  + popt (mandatory, for handling command-line parameters)
0.000048 main(2):  + exif (for displaying EXIF information)
0.000050 main(2):  + cdk (for accessing configuration options)
0.000052 main(2):  + aa (for displaying live previews)
0.000053 main(2):  + jpeg (for displaying live previews in JPEG format)
0.000055 main(2):  + readline (for easy navigation in the shell)
0.000058 main(2): libgphoto2 2.5.3.1
0.000061 main(2): libgphoto2 has been compiled with the following options:
0.000063 main(2):  + all camlibs
0.000065 main(2):  + gcc (C compiler used)
0.000069 main(2):  + ltdl (for portable loading of camlibs)
0.000071 main(2):  + EXIF (for special handling of EXIF files)
0.000079 main(2): libgphoto2_port 0.10.0
0.000084 main(2): libgphoto2_port has been compiled with the following options:
0.000088 main(2):  + gcc (C compiler used)
0.000091 main(2):  + ltdl (for portable loading of camlibs)
0.000095 main(2):  + USB (libusb1, for USB cameras)
0.000099 main(2):  + serial (for serial cameras)
0.000102 main(2):  + no resmgr (serial port access and locking)
0.000106 main(2):  + no ttylock (serial port locking)
0.000110 main(2):  + no lockdev (serial port locking)
0.000114 main(2): CAMLIBS env var not set, using compile-time default instead
0.000118 main(2): IOLIBS env var not set, using compile-time default instead
0.000138 setting/gphoto2-setting.c(2): Creating $HOME/.gphoto
0.000176 setting/gphoto2-setting.c(2): Loading settings from file "/home/ubuntuk/.gphoto/settings"
[... 368 lines cut ...]
0.033766 ptp2/usb_sendreq(2): request code 0x1002 sending req result -35
0.033781 ptp2/camera_init(0): ptp_opensession returns 2ff
0.033783 ptp2/device_reset_request(2): sending reset
0.033785 gphoto2-port(2): Writing message (request=0x66 value=0x0 index=0x0 size=0=0x0)...
0.033788 gphoto2-port(3): No hexdump (NULL buffer)
0.033891 context(0): PTP I/O error

*** Error ***              
PTP I/O error
0.033900 gphoto2-port(2): Closing port...
0.033958 context(0): An error occurred in the io-library ('Unspecified error'): The supplied vendor or product id (0x0,0x0) is not valid.

*** Error ***              
An error occurred in the io-library ('Unspecified error'): The supplied vendor or product id (0x0,0x0) is not valid.
0.034169 gp-camera(2): Freeing camera...
0.034172 gphoto2-port(2): Freeing port...
0.034175 gphoto2-port(2): Closing port...
0.034288 gphoto2-filesystem(2): resetting filesystem
0.034291 libgphoto2/gphoto2-filesys.c(2): Clearing fscache LRU list...
0.034293 libgphoto2/gphoto2-filesys.c(2): fscache LRU list already empty
0.034295 gphoto2-filesystem(2): Internally deleting all folders from '/'...
0.034296 gphoto2-filesystem(2): Lookup folder '/'...
0.034299 gphoto2-filesystem(2): Found! / is 0x10c1320
0.034301 gphoto2-filesystem(2): Recurse delete folder 0x10c1320//
```

msmeissn, 2016-10-11:
For libgphoto2 and sony cameras please update to a newer libgphoto2 version.

Sony DSLR support was added in libgphoto2 2.5.4.

check if libusb1 support is built in...

sfPlayer1, 2017-02-14:
I'm also seeing PTP I/O errors with a Sony device, in my case the A6000. The capture preview command causes the camera link to hang, other commands like summary don't respond any more until USB is temporarily disconnected.

The gphoto2 and libgphoto2 binaries were built from yesterday's git master, platform is armv7. I have no idea whether the device is properly set up, the USB connection setting is "PC Remote".

```
$ gphoto2 --capture-preview --debug
0.000691 main                        (2): ALWAYS INCLUDE THE FOLLOWING LINES WHEN SENDING DEBUG MESSAGES TO THE MAILING LIST:
0.000805 main                        (2): gphoto2 2.5.11.1
0.000868 main                        (2): gphoto2 has been compiled with the following options:
0.000896 main                        (2):  + gcc (C compiler used)
0.000918 main                        (2):  + popt (mandatory, for handling command-line parameters)
0.000939 main                        (2):  + exif (for displaying EXIF information)
0.000960 main                        (2):  + cdk (for accessing configuration options)
0.000980 main                        (2):  + aa (for displaying live previews)
0.000999 main                        (2):  + jpeg (for displaying live previews in JPEG format)
0.001024 main                        (2):  + readline (for easy navigation in the shell)
0.001075 main                        (2): libgphoto2 2.5.12.1
0.001109 main                        (2): libgphoto2 has been compiled with the following options:
0.001131 main                        (2):  + all camlibs
0.001149 main                        (2):  + gcc (C compiler used)
0.001168 main                        (2):  + ltdl (for portable loading of camlibs)
0.001188 main                        (2):  + EXIF (for special handling of EXIF files)
0.001214 main                        (2): libgphoto2_port 0.12.0
0.001247 main                        (2): libgphoto2_port has been compiled with the following options:
0.001272 main                        (2):  + gcc (C compiler used)
0.001292 main                        (2):  + ltdl (for portable loading of camlibs)
0.001316 main                        (2):  + USB (libusb0, for USB cameras)
0.001337 main                        (2):  + serial (for serial cameras)
0.001356 main                        (2):  + no resmgr (serial port access and locking)
0.001376 main                        (2):  + no ttylock (serial port locking)
0.001396 main                        (2):  + no lockdev (serial port locking)
0.001417 main                        (2): CAMLIBS env var not set, using compile-time default instead
0.001437 main                        (2): IOLIBS env var not set, using compile-time default instead
0.001457 main                        (2): invoked with following arguments:
0.001476 main                        (2):   --capture-preview
0.001495 main                        (2):   --debug
0.001635 load_settings               (2): Creating gphoto config directory ('/home/pi/.gphoto')
0.001854 load_settings               (2): Loading settings from file '/home/pi/.gphoto/settings'.
0.002139 main                        (2): The user has not specified both a model and a port. Try to figure them out.
0.002251 gp_port_info_list_load      (2): Using ltdl to load io-drivers from '/usr/local/lib/libgphoto2_port/0.12.0'...
0.002819 foreach_func                (2): Called for filename '/usr/local/lib/libgphoto2_port/0.12.0/disk'.
0.003839 gp_port_library_list        (2): found fstab fsname proc
0.003897 gp_port_library_list        (2): found fstab fsname /dev/mmcblk0p1
0.003988 gp_port_library_list        (2): found fstab fsname /dev/mmcblk0p2
0.004050 gp_port_library_list        (2): found fstab fsname //192.168.0.9/motion-upload
0.004423 gp_port_library_list        (2): found mtab fsname /dev/root
0.004482 gp_port_library_list        (2): found mtab fsname devtmpfs
0.004522 gp_port_library_list        (2): found mtab fsname sysfs
0.004557 gp_port_library_list        (2): found mtab fsname proc
0.004589 gp_port_library_list        (2): found mtab fsname tmpfs
0.004660 gp_port_library_list        (2): found mtab fsname devpts
0.004697 gp_port_library_list        (2): found mtab fsname tmpfs
0.004760 gp_port_library_list        (2): found mtab fsname tmpfs
0.004830 gp_port_library_list        (2): found mtab fsname tmpfs
0.004917 gp_port_library_list        (2): found mtab fsname cgroup
0.004952 gp_port_library_list        (2): found mtab fsname cgroup
0.004982 gp_port_library_list        (2): found mtab fsname cgroup
0.005015 gp_port_library_list        (2): found mtab fsname cgroup
0.005045 gp_port_library_list        (2): found mtab fsname cgroup
0.005178 gp_port_library_list        (2): found mtab fsname cgroup
0.005213 gp_port_library_list        (2): found mtab fsname cgroup
0.005247 gp_port_library_list        (2): found mtab fsname systemd-1
0.005277 gp_port_library_list        (2): found mtab fsname debugfs
0.005351 gp_port_library_list        (2): found mtab fsname mqueue
0.005385 gp_port_library_list        (2): found mtab fsname fusectl
0.005420 gp_port_library_list        (2): found mtab fsname configfs
0.005504 gp_port_library_list        (2): found mtab fsname /dev/mmcblk0p1
0.005573 gp_port_library_list        (2): found mtab fsname tmpfs
0.005647 gp_port_library_list        (2): found mtab fsname gvfsd-fuse
0.005834 foreach_func                (2): Loaded '' ('^disk:') from '/usr/local/lib/libgphoto2_port/0.12.0/disk'.
[... cut ...]
80.197184 gp_filesystem_reset         (2): resetting filesystem
80.197441 gp_filesystem_lru_clear     (2): Clearing fscache LRU list...
80.197681 gp_filesystem_lru_clear     (2): fscache LRU list already empty
80.197924 delete_all_folders          (2): Internally deleting all folders from '/'...
80.198501 lookup_folder               (2): Lookup folder '/'...
80.199060 lookup_folder               (2): Found! / is 0xb4f3c0
80.199613 recurse_delete_folder       (2): Recurse delete folder 0xb4f3c0//
```

sfPlayer1, 2017-02-14:
It's quite likely that the camera firmware doesn't implement the functionality over USB, commercial offerings don't support it either. See https://www.phaseone.com/en/Products/Software/Capture-One-for-Sony/Supported-Cameras.aspx for example.

msmeissn, 2017-02-14:
--capture-preview only works for some Sony SLT models, the more expensive ones I think.

On others it will hang.

elijahparker, 2017-02-14:
The A6300 & A6500 support capture preview, but not the A6000.  All the
current models seem to support it, just not the prior generation.

msmeissn, 2017-02-14:
I now added the PTP_CAP_PREVIEW flag to the A6300. I have no seperate USB id for the A6500 yet.

DavidVik, 2017-06-15:
I have been trying to get my Canon EOS M3 working with gphoto2 ...

The camera is auto-detected, it can list-files, and successfully did gphoto2 --set-config capturetarget=1
But I cannot successfully trigger capture. I get the following:

david@david-MS-7817 ~ $ gphoto2 --trigger-capture
                                                                               
*** Error ***              
Canon EOS M Full-Press failed (0x2019: PTP Device Busy)
ERROR: Could not trigger capture.
*** Error (-110: 'I/O in progress') ***       

For debugging messages, please use the --debug option.
Debugging messages may help finding a solution to your problem.
If you intend to send any error or debug messages to the gphoto
developer mailing list <gphoto-devel@lists.sourceforge.net>, please run
gphoto2 as follows:

    env LANG=C gphoto2 --debug --debug-logfile=my-logfile.txt --trigger-capture

Please make sure there is sufficient quoting around the arguments.

So I ran the debug command and here is the DEBUG file :
  
[my-logfile.txt](https://github.com/gphoto/libgphoto2/files/1078773/my-logfile.txt)

msmeissn, 2017-06-15:
does regular --capture-image work better?

DavidVik, 2017-06-15:
Thank you for your speedy response. I actually hadn't tried that 
*[**does regular --capture-image work better?]*  yet. But now have done 
so. It also fails:

*david@david-MS-7817 ~ $ gphoto2 --capture-image*

*** Error ***
Canon EOS M Full-Press failed (0x2019: PTP Device Busy)
ERROR: Could not capture image.
ERROR: Could not capture.
*** Error (-110: 'I/O in progress') ***

For debugging messages, please use the --debug option.
Debugging messages may help finding a solution to your problem.
If you intend to send any error or debug messages to the gphoto
developer mailing list <gphoto-devel@lists.sourceforge.net>, please run
gphoto2 as follows:

     env LANG=C gphoto2 --debug --debug-logfile=my-logfile.txt 
--capture-image

Please make sure there is sufficient quoting around the arguments.

DavidVik, 2017-06-15:
I have installed Entangle [see https://entangle-photo.org/] which is based on libgphoto. And I am able to successfully capture with my EOS M3 ...
But of course I would like to have the gphoto2 working so i could do more than Entangle can do (I dont think you could do time lapse and more challenging camera projects with it).

msmeissn, 2017-06-15:
weird enough, entangle uses libgphoto2... is this with preview enabled? perhaps try a --capture-preview in front?

DavidVik, 2017-06-16:
In Entangle, yes, this was with preview enabled. But it also works 
[captures image] in Entangle /without/ preview enabled.

However [in Entangle] when I tried changing camera settings, for example 
ISO, then I get error message:

Returning to GPHOTO2

I tried the command line in terminal : gphoto2 --capture-preview

But it still gives an error:

david@david-MS-7817 ~ $ gphoto2 --capture-preview
*** Error (-1: 'Unspecified error') ***

For debugging messages, please use the --debug option.
Debugging messages may help finding a solution to your problem.
If you intend to send any error or debug messages to the gphoto
developer mailing list <gphoto-devel@lists.sourceforge.net>, please run
gphoto2 as follows:

     env LANG=C gphoto2 --debug --debug-logfile=my-logfile.txt 
--capture-preview

Please make sure there is sufficient quoting around the arguments.

DavidVik, 2017-06-16:
Thank you for your speedy response. I actually had not tried that yet [does regular --capture-image work better?]. But now have done so, and it also fails to work:

david@david-MS-7817 ~ $ gphoto2 --capture-image
                                                                              
*** Error ***             
Canon EOS M Full-Press failed (0x2019: PTP Device Busy)
ERROR: Could not capture image.
ERROR: Could not capture.
*** Error (-110: 'I/O in progress') ***      

For debugging messages, please use the --debug option.
Debugging messages may help finding a solution to your problem.
If you intend to send any error or debug messages to the gphoto
developer mailing list <gphoto-devel@lists.sourceforge.net>, please run
gphoto2 as follows:

    env LANG=C gphoto2 --debug --debug-logfile=my-logfile.txt --capture-image

Please make sure there is sufficient quoting around the arguments.

DavidVik, 2017-06-17:
Hello Marcus thank you for your efforts to help me.

Last night I tried again. I notice that the Camera is not picked up and 
mounted on Thunar file viewer/manager, unlike the day before ... and so 
I tried  the gphoto2 commands,

and it worked on:

-- capture-preview

and

--trigger capture  .... but still gave error when I tried --capture-image

Also gave error on --set-config shutterspeed=16 ...(etc)

But at least it is partly working now ...

I am not sure why ...

msmeissn, 2018-07-21:
we made quite some improvements in the last year. 

the bugreport is a bit confusing, if issues still persist with 2.5.18 please open a new ticket.

closed 2018-07-21 by msmeissn

trougnouf, 2022-10-10:
The issue persists with 2.5.28 (Sony A7C); --capture-image works until it doesn't (then I need to physically reset the connection), but --trigger-capture always works.