gphoto/libgphoto2 #94: Cannot take long exposures on Nikon D5100 camera with latest gphoto2

jonbelanger, 2016-10-01:
I just compiled libgphoto2 and gphoto2, but the problem persists.

The issue is basically this:

root@jon-desktop:~# gphoto2 --auto-detect --set-config=/main/settings/capturetarget=0
Model                          Port                                            
----------------------------------------------------------
Nikon DSC D5100 (PTP mode)     usb:001,010     
root@jon-desktop:~# gphoto2 --auto-detect --get-config=/main/settings/capturetarget
Model                          Port                                            
----------------------------------------------------------
Nikon DSC D5100 (PTP mode)     usb:001,010     
Label: Capture Target
Type: RADIO
Current: Internal RAM
Choice: 0 Internal RAM
Choice: 1 Memory card
root@jon-desktop:~# env LANG=C gphoto2 --auto-detect --debug --debug-logfile=gphoto2-latest-internalram.log --capture-image-and-download
Model                          Port                                            
----------------------------------------------------------
Nikon DSC D5100 (PTP mode)     usb:001,010     
ERROR: Could not capture image.
ERROR: Could not capture.
New file is in location /capt0000.nef on the camera                            
File capt0000.nef exists. Overwrite? [y|n] n
Specify new filename? [y|n] y
Enter new filename: capt0000-2.nef
Saving file as capt0000-2.nef
Deleting file /capt0000.nef on the camera

This is with shutterspeed set to option 51 (30 seconds) (set using stock Ubuntu MATE gphoto2, but the setting persisted into using the latest compiled versions):

# gphoto2 --auto-detect --set-config=/main/capturesettings/shutterspeed=51
Model                          Port                                            
----------------------------------------------------------
Nikon DSC D5100 (PTP mode)     usb:001,009

There is an extremely long pause between the shutter closing and the image downloading.  When it does finally download the two ERROR messages appear in the terminal along with the New file messages.  The file does actually download eventually, but these timeouts are causing problems with other software that relies on gphoto2.

jonbelanger, 2016-10-01:
Logs for capturetarget=0 and capturetarget=1 are here:

https://drive.google.com/open?id=0B5ztTI1IvqPfZDFSM1lYdTZaZFE

jonbelanger, 2016-10-03:
Here's a more concise description of the problem from the debug logs.

Please note with the previously attached debug logs I was pointing to the default repo libgphoto2 (2.5.9), but I fixed that as listed below gphoto2 now points to (2.5.10 in /usr/local), but the problem still persists.

I believe this is going to be a problem for long exposures over all cameras with the ptp2 driver.  I'm told the canon driver "just works".

Also, there seems to be a problem with it reporting the "Property not found" error, but I can confirm from the camera display that the shutter speed is changing after each command.

Thanks for your attention.

0.000570 main                        (2): ALWAYS INCLUDE THE FOLLOWING LINES WHEN SENDING DEBUG MESSAGES TO THE MAILING LIST:
0.000724 main                        (2): gphoto2 2.5.10
0.000764 main                        (2): gphoto2 has been compiled with the following options:
0.000793 main                        (2):  + gcc (C compiler used)
0.000819 main                        (2):  + popt (mandatory, for handling command-line parameters)
0.000844 main                        (2):  + exif (for displaying EXIF information)
0.000869 main                        (2):  + no cdk (for accessing configuration options)
0.000893 main                        (2):  + no aa (for displaying live previews)
0.000918 main                        (2):  + no jpeg (for displaying live previews in JPEG format)
0.000942 main                        (2):  + no readline (for easy navigation in the shell)
0.001002 main                        (2): libgphoto2 2.5.10
0.001040 main                        (2): libgphoto2 has been compiled with the following options:
0.001066 main                        (2):  + INCOMPLETE CAMLIB SET (canon ptp2)
0.001091 main                        (2):  + gcc (C compiler used)
0.001116 main                        (2):  + ltdl (for portable loading of camlibs)
0.001140 main                        (2):  + EXIF (for special handling of EXIF files)
0.001172 main                        (2): libgphoto2_port 0.12.0
0.001206 main                        (2): libgphoto2_port has been compiled with the following options:
0.001232 main                        (2):  + gcc (C compiler used)
0.001257 main                        (2):  + ltdl (for portable loading of camlibs)
0.001283 main                        (2):  + USB (libusb1, for USB cameras)
0.001308 main                        (2):  + serial (for serial cameras)
0.001332 main                        (2):  + no resmgr (serial port access and locking)
0.001356 main                        (2):  + no ttylock (serial port locking)
0.001381 main                        (2):  + no lockdev (serial port locking)
0.001406 main                        (2): CAMLIBS env var not set, using compile-time default instead
0.001430 main                        (2): IOLIBS env var not set, using compile-time default instead
0.001454 main                        (2): invoked with following arguments:
0.001479 main                        (2):   --capture-image-and-download
0.001503 main                        (2):   --force-overwrite
0.001528 main                        (2):   --debug
0.001552 main                        (2):   --debug-logfile=fastcapture.log
0.001746 load_settings               (2): Creating gphoto config directory ('/root/.gphoto')
0.001890 load_settings               (2): Loading settings from file '/root/.gphoto/settings'.
0.002184 main                        (2): The user has not specified both a model and a port. Try to figure them out.
0.002244 gp_port_info_list_load      (2): Using ltdl to load io-drivers from '/usr/local/lib/libgphoto2_port/0.12.0'...
0.002569 foreach_func                (2): Called for filename '/usr/local/lib/libgphoto2_port/0.12.0/disk'.
0.003542 gp_port_library_list        (2): found fstab fsname proc
0.003598 gp_port_library_list        (2): found fstab fsname /dev/mmcblk0p2
0.003670 gp_port_library_list        (2): found fstab fsname /dev/mmcblk0p1
0.004065 gp_port_library_list        (2): found mtab fsname /dev/root
0.004135 gp_port_library_list        (2): found mtab fsname devtmpfs
0.004177 gp_port_library_list        (2): found mtab fsname sysfs
0.004213 gp_port_library_list        (2): found mtab fsname proc
0.004248 gp_port_library_list        (2): found mtab fsname tmpfs
0.004322 gp_port_library_list        (2): found mtab fsname devpts
0.004361 gp_port_library_list        (2): found mtab fsname tmpfs
0.004427 gp_port_library_list        (2): found mtab fsname tmpfs
0.004497 gp_port_library_list        (2): found mtab fsname tmpfs
0.004580 gp_port_library_list        (2): found mtab fsname cgroup
0.004618 gp_port_library_list        (2): found mtab fsname cgroup
0.004693 gp_port_library_list        (2): found mtab fsname cgroup
0.004729 gp_port_library_list        (2): found mtab fsname cgroup
0.004764 gp_port_library_list        (2): found mtab fsname cgroup
0.004888 gp_port_library_list        (2): found mtab fsname cgroup
0.004927 gp_port_library_list        (2): found mtab fsname cgroup
0.004964 gp_port_library_list        (2): found mtab fsname systemd-1
0.004999 gp_port_library_list        (2): found mtab fsname debugfs
0.005080 gp_port_library_list        (2): found mtab fsname mqueue
0.005118 gp_port_library_list        (2): found mtab fsname tracefs
[... 541 lines cut ...]
[... cut ...]
151.868798 camera_wait_for_event       (2): waiting for events timeout 2993 ms
151.869746 camera_wait_for_event       (2): waiting for events timeout 2992 ms
155.620767 camera_wait_for_event       (2): waiting for events timeout 3000 ms
155.621919 camera_wait_for_event       (2): waiting for events timeout 2999 ms
155.622540 camera_wait_for_event       (2): waiting for events timeout 2999 ms
155.673876 camera_wait_for_event       (2): waiting for events timeout 47 ms

real	2m36.117s
user	0m3.270s
sys	0m0.590s

jonbelanger, 2016-10-07:
Disappointed with the lack of response on this issue.

msmeissn, 2016-10-07:
The first issue with the very long timeouts was a libgphoto2  2.5.9 problem. basically we only waited for 5 seconds after capture, instead of the intended 50 seconds. As the cameras was not finished, it tried retrieving an unknown object and that hung also with a timeout of 50 seconds.
As you see libgphoto2 2.5.10 fixed this.

i am now trying to undertsand your second issue ;)

msmeissn, 2016-10-07:
is the last call really done with libgphoto2 2.5.10? 

can you capture new debuglogfiles for those, the ones in google drive use libgphoto2 2.5.9.

jonbelanger, 2016-10-10:
What's the reason for waiting so long after exposures?  Please note the 2 minutes 37 seconds wait time for a 30 second exposure.

The full logs are here:

https://drive.google.com/open?id=0B5ztTI1IvqPfbm9ySGl2R1hiaTg

$ lsusb
Bus 001 Device 004: ID 04b0:0429 Nikon Corp. D5100
Bus 001 Device 003: ID 0424:ec00 Standard Microsystems Corp. SMSC9512/9514 Fast Ethernet Adapter
Bus 001 Device 002: ID 0424:9514 Standard Microsystems Corp. SMC9514 Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

$ which gphoto2
/usr/local/bin/gphoto2

$ ldd /usr/local/bin/gphoto2
	linux-vdso.so.1 =>  (0x7eed6000)
	/usr/lib/arm-linux-gnueabihf/libarmmem.so (0x76f1d000)
	libgphoto2.so.6 => /usr/local/lib/libgphoto2.so.6 (0x76e79000)
	libgphoto2_port.so.12 => /usr/local/lib/libgphoto2_port.so.12 (0x76e61000)
	libpthread.so.0 => /lib/arm-linux-gnueabihf/libpthread.so.0 (0x76e3d000)
	libexif.so.12 => /usr/lib/arm-linux-gnueabihf/libexif.so.12 (0x76e01000)
	libpopt.so.0 => /lib/arm-linux-gnueabihf/libpopt.so.0 (0x76de8000)
	libc.so.6 => /lib/arm-linux-gnueabihf/libc.so.6 (0x76cfb000)
	/lib/ld-linux-armhf.so.3 (0x54b51000)
	libltdl.so.7 => /usr/lib/arm-linux-gnueabihf/libltdl.so.7 (0x76ce5000)
	libm.so.6 => /lib/arm-linux-gnueabihf/libm.so.6 (0x76c6d000)
	libdl.so.2 => /lib/arm-linux-gnueabihf/libdl.so.2 (0x76c5a000)

$ /usr/local/bin/gphoto2 --auto-detect
Model                          Port                                            
----------------------------------------------------------
Nikon DSC D5100 (PTP mode)     usb:001,004     
jon@jon-desktop:~$ /usr/local/bin/gphoto2 --auto-detect --list-config | grep shutterspeed
/main/capturesettings/minimumshutterspeed
/main/capturesettings/shutterspeed
/main/capturesettings/shutterspeed2

$ /usr/local/bin/gphoto2 --auto-detect --set-config=/main/capturesettings/shutterspeed=51
Model                          Port                                            
----------------------------------------------------------
Nikon DSC D5100 (PTP mode)     usb:001,004     

*** Error ***              
Property '/main/capturesettings/shutterspeed' not found.
jon@jon-desktop:~$ /usr/local/bin/gphoto2 --auto-detect --get-config=/main/capturesettings/shutterspeed
Model                          Port                                            
----------------------------------------------------------
Nikon DSC D5100 (PTP mode)     usb:001,004     

*** Error ***              
Property '/main/capturesettings/shutterspeed' not found.
Label: Shutter Speed
Type: RADIO
Current: 30.0000s
Choice: 0 0.0002s
Choice: 1 0.0003s
Choice: 2 0.0004s
Choice: 3 0.0005s
Choice: 4 0.0006s
Choice: 5 0.0008s
Choice: 6 0.0010s
Choice: 7 0.0012s
Choice: 8 0.0015s
Choice: 9 0.0020s
Choice: 10 0.0025s
Choice: 11 0.0031s
Choice: 12 0.0040s
Choice: 13 0.0050s
Choice: 14 0.0062s
[... 55 lines cut ...]
0.156050 gp_port_set_timeout         (2): Setting port timeout to 500 milliseconds.
0.169959 gp_port_set_timeout         (2): Setting port timeout to 5000 milliseconds.
0.190210 gp_port_set_timeout         (2): Setting port timeout to 5000 milliseconds.
0.214078 gp_port_set_timeout         (2): Setting port timeout to 5000 milliseconds.
0.215901 gp_port_set_timeout         (2): Setting port timeout to 500 milliseconds.
0.216433 gp_port_set_timeout         (2): Setting port timeout to 500 milliseconds.
0.227947 gp_port_set_timeout         (2): Setting port timeout to 5000 milliseconds.
0.245402 gp_port_set_timeout         (2): Setting port timeout to 5000 milliseconds.
0.256541 gp_port_set_timeout         (2): Setting port timeout to 8000 milliseconds.
0.260269 gp_port_set_timeout         (2): Setting port timeout to 20000 milliseconds.
1.503250 gp_port_set_timeout         (2): Setting port timeout to 100000 milliseconds.
51.504265 gp_libusb1_read [libusb1.c:571](0): 'libusb_bulk_transfer (port->pl->dh, port->settings.usb.inep, (unsigned char*)bytes, size, &curread, port->timeout)' failed: Pipe error (-9)
151.504739 gp_libusb1_read [libusb1.c:571](0): 'libusb_bulk_transfer (port->pl->dh, port->settings.usb.inep, (unsigned char*)bytes, size, &curread, port->timeout)' failed: Operation timed out (-7)
151.505278 camera_wait_for_event       (2): waiting for events timeout 3000 ms
151.506457 camera_wait_for_event       (2): waiting for events timeout 2999 ms
151.507136 camera_wait_for_event       (2): waiting for events timeout 2999 ms
151.507757 camera_wait_for_event       (2): waiting for events timeout 2998 ms
151.508405 camera_wait_for_event       (2): waiting for events timeout 2997 ms
151.509037 camera_wait_for_event       (2): waiting for events timeout 2997 ms
151.509657 camera_wait_for_event       (2): waiting for events timeout 2996 ms
151.510272 camera_wait_for_event       (2): waiting for events timeout 2995 ms
151.510915 camera_wait_for_event       (2): waiting for events timeout 2995 ms
156.757971 camera_wait_for_event       (2): waiting for events timeout 3000 ms
156.758561 camera_wait_for_event       (2): waiting for events timeout 100 ms
156.758914 camera_wait_for_event       (2): waiting for events timeout 100 ms

jonbelanger, 2016-10-10:
Also just a note - I had to compile with libusb - 

  495  apt-get install libusb-1.0-0-dev
  496  apt-get install libusb-1.0-0

in order to get libgphoto2-2.5.10 to recognize the camera correctly.

msmeissn, 2016-10-10:
looking at the timing the image gets added after 60 seconds , here are the interrupts:

66.152622 _cb_irq                     (2): 0x489700 with status 0
66.152758 _cb_irq                     (3): interrupt (hexdump of 16 bytes)
0000  10 00 00 00 04 00 02 40-ff ff ff ff 2c 49 19 09  .......@....,I..

66.152817 _cb_irq                     (2): requeuing complete transfer 0x489700
66.248585 _cb_irq                     (2): 0x48ff08 with status 0
66.248641 _cb_irq                     (3): interrupt (hexdump of 16 bytes)
0000  10 00 00 00 04 00 0d 40-ff ff ff ff 07 01 00 00  .......@........

which is weird for a 30 seconds exposure. But as it around 2x30 seconds ... Is the camera taking a dark-reference picture right afterwards?

rlamarche, 2016-10-10:
Try do **disable** the in-camera noise reduction for long exposure and see if it is faster.
https://www.youtube.com/watch?v=YmVZPYxio_s

jonbelanger, 2016-10-10:
I only hear it clicking once, don't think dark reference frame is getting captured.  I'll try to disable NR.

jonbelanger, 2016-10-11:
disabling NR worked, no more delay now in 2.5.10.  Images are immediately downloaded after 30 seconds exposure.  Entire process takes about 55 seconds per frame.

msmeissn, 2016-10-21:
this means we probably need to extent the wait a bit to cover the NR delay. I am going to 60 seconds for now in git.

closed 2016-10-21 by msmeissn