gphoto/libgphoto2 #96: "No iolibs found" when using as a static lib on OSX

lab101, 2016-10-05:
Hey Gphoto2 people,

I'm on OSX 10.11.4 and build a static version of gPhoto2 with the package manager Brew.
The static lib works fine on my  dev machine but when I deploy the application I got an error when calling `retval = gp_camera_init(camera, cameracontext);`

```
No iolibs found in '/usr/local/Cellar/libgphoto2/2.5.10/lib/libgphoto2_port/0.12.0'
Retval: -4 Error loading a library
```

The /usr/local/Cellar/... path is where the lib was compiled on the dev machine.
So it's still referenced somewhere.

Can somebody point me in the right direction on how to fix this.
I'm not very experienced in compiling libs so could be a very a stupid mistake.

Thx in advance
Kris

msmeissn, 2016-10-11:
libgphoto2 kind of reuqires to dynamically load its drivers. A full static build is not easily possible  I think, libtool might allow this somehow, but i have never tried it.

lab101, 2016-10-11:
As I mentioned before I'm not very experienced in compiling libs so I probably continue to use it as dynamic lib.

closed 2016-10-11 by lab101

ndim, 2016-10-28:
Way back, I tried to build _iolibs_ and _camlibs_ into `libgphoto2_port.so` and `libgphoto2.so`, respectively. IIRC, we still wanted to support non-`libltdl` builds which made it quite difficult for not a lot of gain.

I see two possibilities to integrate _iolibs_ and _camlibs_ into the respective libs:

1. Either use `libtool`+`libltdl` and the same symbols we are using right now.
2. Change to a system to statically link _iolibs_ and _camlibs_ into the libraries similar to how the Linux kernel builds drivers into the kernel image.

Either of those two might accelerate `libgphoto2_port` and `libgphoto2` startup, which could be a better reason to look into this technical change than static linking is.