gphoto/libgphoto2 #97: Windows port - few questions

peterbud, 2016-10-05:
I have successfully compiled and built libgphoto2 and gphoto2 under Windows 10 using the excellent MSYS2 mingw64 environment. With my Nikon D5000 I started testing the basic functionalities. As a reference I used my dual booted Ubuntu environment where I had libgphoto2 2.5.3, and it was nicely working with my D5000, and I have created debug files from both environment to compare the behaviour

Many of the functions are working perfectly on Windows, however I face sume issues where I'd need some help, as I'm not familiar with the library.
**--auto-detect** works fine, and recognizes the camera
**--sumary** works also fine, but I see that in the windows build at the end of the debug file I got some strange messages (_close_async_interrupts and _cb_irq), which I don't have on Linux, and I believe are some sort of errors:
`
1.611582 ptp_usb_getresp             (2): Reading PTP_OC 0x1003 (Close session) response...
1.627206 gp_port_read                (3): Reading 1024 = 0x400 bytes from port...
1.627206 gp_port_read                (3): Read    12 = 0xc out of 1024 bytes from port: (hexdump of 12 bytes)
0000  0c 00 00 00 03 00 01 20-7d 00 00 00              ....... }...    

1.627206 gp_port_close               (2): Closing port...
1.642835 _close_async_interrupts     (2): canceling transfer 0:0000000002ceab10 (status 0)
1.642835 _close_async_interrupts     (2): canceling transfer 1:0000000002cec1d0 (status 0)
1.642835 _close_async_interrupts     (2): canceling transfer 2:0000000002cec2c0 (status 0)
1.642835 _close_async_interrupts     (2): canceling transfer 3:0000000002df3d80 (status 0)
1.642835 _close_async_interrupts     (2): canceling transfer 4:0000000002df3e70 (status 0)
1.642835 _close_async_interrupts     (2): canceling transfer 5:0000000002d25070 (status 0)
1.642835 _close_async_interrupts     (2): canceling transfer 6:0000000002d25160 (status 0)
1.642835 _close_async_interrupts     (2): canceling transfer 7:0000000002d25250 (status 0)
1.642835 _close_async_interrupts     (2): canceling transfer 8:0000000002cee980 (status 0)
1.642835 _close_async_interrupts     (2): canceling transfer 9:0000000002ceea70 (status 0)
1.642835 _cb_irq                     (2): 0000000002ceab10 with status 3
1.642835 _cb_irq                     (2): 0000000002cec1d0 with status 3
1.642835 _cb_irq                     (2): 0000000002cec2c0 with status 3
1.642835 _cb_irq                     (2): 0000000002df3d80 with status 3
1.642835 _cb_irq                     (2): 0000000002df3e70 with status 3
1.642835 _cb_irq                     (2): 0000000002d25070 with status 3
1.642835 _cb_irq                     (2): 0000000002d25160 with status 3
1.642835 _cb_irq                     (2): 0000000002d25250 with status 3
1.642835 _cb_irq                     (2): 0000000002cee980 with status 3
1.642835 _cb_irq                     (2): 0000000002ceea70 with status 3
1.642835 gp_filesystem_reset         (2): resetting filesystem
1.642835 gp_filesystem_lru_clear     (2): Clearing fscache LRU list...
1.642835 gp_filesystem_lru_clear     (2): fscache LRU list already empty
1.642835 delete_all_folders          (2): Internally deleting all folders from '/'...
1.642835 lookup_folder               (2): Lookup folder '/'...
1.642835 lookup_folder               (2): Found! / is 000000000093a1c0
1.642835 recurse_delete_folder       (2): Recurse delete folder 000000000093a1c0//
1.642835 gp_port_free                (2): Freeing port...
1.642835 gp_port_close               (2): Closing port...
1.642835 gp_filesystem_reset         (2): resetting filesystem
1.642835 gp_filesystem_lru_clear     (2): Clearing fscache LRU list...
1.642835 gp_filesystem_lru_clear     (2): fscache LRU list already empty
1.642835 delete_all_folders          (2): Internally deleting all folders from '/'...
1.642835 lookup_folder               (2): Lookup folder '/'...
1.642835 lookup_folder               (2): Found! / is 000000000093a1c0
1.642835 recurse_delete_folder       (2): Recurse delete folder 000000000093a1c0//`

 **--capture-image-and-download** works as well on Windows, it captures the imafe, downloads and creates the image on the computer, but fails on renaming the file. I assume there are still some file handles open, which prevents renaming the file. If I rename the file manually later, it is a pefrect JPG file, which I can open with any editor. Also in the debug log I see similar messages like above

**Question:** Is the above a known issue? If not, what is the best way to debug these errors? Any help would be appreciated.

Peter

msmeissn, 2016-10-06:
the async and cb_irq things are expected and wanted. They will show when libgphoto2 is built against libusb1.   
libusb0 does not have thgose, so your linux libgphoto2 is likjely built against libusb0 instead of libusb1.

peterbud, 2016-10-06:
Hi Marcus, thank you for your quick reply.

The ubuntu version of ligphoto2 seems to be built using lubusb1 as well, at least the log says:
0.000097 main(2): libgphoto2_port has been compiled with the following options:
0.000100 main(2):  + gcc (C compiler used)
0.000103 main(2):  + ltdl (for portable loading of camlibs)
0.000106 main(2):  + USB **(libusb1**, for USB cameras)
0.000109 main(2):  + serial (for serial cameras)
0.000112 main(2):  + no resmgr (serial port access and locking)
0.000115 main(2):  + no ttylock (serial port locking)
0.000118 main(2):  + no lockdev (serial port locking)

I have attached 2 files: one of the edited log of the ubuntu version and the second is the edited log of the windows version for the same command **capture-image-and-download** (I have removed the data transfer pieces)
[log-capture-and-download.ubuntu (short).txt](https://github.com/gphoto/libgphoto2/files/514574/log-capture-and-download.ubuntu.short.txt)
[log-capture-and-download.windows (short).txt](https://github.com/gphoto/libgphoto2/files/514573/log-capture-and-download.windows.short.txt)

The issue is that the Windows build gives me a **permission denied** error constantly, altough I have access to the folder.
New file is in location /capt0000.jpg on the camera
Saving file as a:\t0000.jpg
**rename: Permission denied**
Deleting file /capt0000.jpg on the camera

I have compared the two files in details, to pinpoint where the issue could be, but apparently there are different command used, maybe because the ubuntu version is 2.5.3 and the Windows port I have compiled is 2.5.10. I'm stuck as Iäm not familiar how I can debug the process properly. Any idea where should I look for fixing this? 

Thank you!

msmeissn, 2016-10-06:
Saving file as a:\t0000.jpg ... it seems to try to save it to a window style drive A: (floppy) filename ... 

it pretty much seems the filename handling during save here is broken.

you can try --filename "c:\\something\\foo.jpg" or similar?

peterbud, 2016-10-06:
Hi Marcus, 

I have specified the file name like this:
gphoto2.exe --capture-image-and-download --filename "D:\Peter\Documents\Development\build\gphoto2build\foo.jpg" --debug --debug-loglevel debug --debug-logfile capturelog.txt

In the terminal the same issue visible:
New file is in location /capt0000.jpg on the camera
Saving file as D:\Peter\Documents\Development\build\gphoto2build\foo.jpg
rename: Permission denied
Deleting file /capt0000.jpg on the camera

And here is the log:
[capturelog.txt](https://github.com/gphoto/libgphoto2/files/514868/capturelog.txt)

The intersting is that the file itself IS created locally with a temp name like "tmpfileiG2Kb7", just the code cannot rename to the specified name (as I believe its still being written to or locked for other reasons)

PS: I have also build a DEBUG version, but this surfaced a few issues in other parts of the code (like some undefined variables nothing serious :)) Tomorrow I can try to debug more.

msmeissn, 2016-10-07:
we have code that copies stuff when rename does not work, i added EPERM as indicator for needing that too now.

peterbud, 2016-10-10:
I have spent some time and I believe I have found the issue:
First of all the errno I get back is not EPERM, but EACCESS

But checking on EPERM or EACCESS as you suggested is not solving the problem as the subsequent attempt to open it read only fails as well, and this line returns an error:
`	in_fd = open(curname, O_RDONLY);`

As I debug it thoroughly, I see the following (and I'd need your confirmation whether I interpret it right)

1. The temp file is being created with a call to **mkstemp** in **function save_file_to_file**
`fd = mkstemp(tmpname);`
2. Then the captured image is saved into that file either by the **fd** method or by the **handler** method (BTW I don't know what is the difference between the two?)
3. Then a function **save_camera_file_to_file** is called which tries to rename the temp file which has been created at point 1. However the file has not been closed yet, and at least on Windows it throws the error we have seen. 
4. The temp file is closed **only after** the function **save_camera_file_to_file** finishes:
```
	res = save_camera_file_to_file (folder, filename, type, file, tmpfilename);
	if (ps && ps->fd) close (ps->fd);
	if (ps) free (ps);

```
Also it seems to me that if the **fd** method is being used earlier, the file is never closed explicitly in the function as ps is NULL.

What I suggest for resolution:
Close the temp file before calling **save_camera_file_to_file** something like this:
```
	//close the temp file as you cannot rename it later
	if (ps && ps->fd) close (ps->fd);
	free (ps);
	ps = NULL;
	if (fd) close (fd);

```

With that changes this function is working nicely for me.

peterbud, 2016-10-15:
Marcus, let's close this issue here, as it is related to gphoto2, and I beleive I have found the solution for this, it will come with a gphoto2 PR.

closed 2016-10-15 by peterbud

msmeissn, 2016-10-18:
okay, thanks ! :)

bob10818, 2017-10-14:
I've used gphoto2 successfully for a couple of years on Raspberry Pi's, but now I need a Windows version, and it's a bit beyond me to build from scratch. Is there a plug-and-play version anywhere for Windows 10, or a documented build process?
Much obliged.

peterbud, 2017-10-14:
No standalone installer, but you can use the MSYS2 version of gphoto2 on Windows 10 64-bit. [Download](http://www.msys2.org/) and install MSYS2, and then you can install `gphoto2 `by:
`pacman -S mingw-w64-x86_64-gphoto2`

After that you can use either from bash (environment variable set by pacman automatically) or from a command prompt, assuming you set the right environment variables by hand.

bob10818, 2017-10-14:
Thanks very much for your prompt reply. I'm having a few problems, but I'll
see what I can work out tomorrow.

In the MSYS shell, it just returns to the prompt without appearing to do
anything, while in the DOS cmd shell, it fails to start with error
0xc000007b

Regards
Bob

peterbud, 2017-10-15:
With `MSYS2` you get 3 shells: `MSYS2`, `MINGW32`, `MINGW64`. You need to launch the `MINGW64 `shell, and use `gphoto2` from there, as this is a native Win64 application.

anotherche, 2017-12-09:
Thank you, Peter, for the possibility to build gphoto2 and libgphoto2 for Windows.
I have a couple of questions about the usage.
gphoto2 works perfect from mingw64 shell. But is it possible to use it from standard cmd? The program itself is working from cmd but --auto-detect shows no camera attached (gphoto2 --auto-detect launched in migw64 terminal shows my Canon EOS 400D at the same time).
As well gphoto2 --list-ports started in cmd window shows empty list (while a list of 7 connections is shown in migw64 terminal). I tried various usb drivers (libusb-win32, WinUSB, no driver) - gphoto2 launched in mingw64 terminal always sees the camera, but never sees it when launched from cmd window. What can be done to use gphoto2 on systems without mingw/msys2 installed?

Another question is about binding libgphoto2 to java. I currently develop java based GUI for time lapse capture using jna based binding to libgphoto2. It works perfect in Linux. But it seems I cannot bind it correct in Windows. The best result is reached when I put all native dll's to a .jar file on the class path. The library (lingphoto2-6.dll) is found (although after renaming it to gphoto2.dll) and probably loaded. But then unsatisfied link appears because some necessary module cannot be found. It seems that it is some of dependencies, but I cannot understand which one. 
So the question is: is it possible in principle to use jna with these dll's built with mingw64, and what is the correct procedure?

peterbud, 2017-12-09:
Hi @anotherche , yes the version which you used under from the mingw64 shell is working from the standard cmd, but in order to make that work, you need to make sure that the shell finds the dependent libraries (and `libgphoto2 `has a number of those). You either add to the `PATH` the `mingw64\bin `folder, or even better distribute/copy the dependent libraries with your gphoto2 installation. You can get a list of those libraries by using Dependency Walker.
Also, I'd suggest to check how we are distributing `libgphoto2` with `darktable`.

With the java binding, unfortunately I cannot help, as I don't know java that much.

anotherche, 2017-12-10:
Thank you again. I managed to run gphoto2 from cmd line finally after specifying necessary paths to the dependencies and after setting two env variables, IOLIBS and CAMLIBS, following corresponding locations of dll's for cameras and ports.  
But I have found some problems with the windows port of gphoto2. My camera (Canon EOS 400D) captures only the first image in response to --F n -I m --capture-image-and-download.  The next capture in the series produces no shutter release and finishes with "Sorry, your camera does not support generic capture". Single image capture (no -F specified) goes without any problem. When -debug is asked the camera does no capture at all with or without -F option specified (again that "Sorry, your camera does not support generic capture").
[my-logfile.txt](https://github.com/gphoto/libgphoto2/files/1545415/my-logfile.txt)

msmeissn, 2017-12-12:
Please do not use 2.5.15, it has a serious bug which you are triggering. upgrade to 2.5.16.

peterbud, 2017-12-12:
I might have missed that release... I'll update the mingw repository to 2.5.16

anotherche, 2017-12-13:
I confirm that 2.5.16 works without these errors under Windows. Compiled in msys2 with the same patches by Peter.

define-private-public, 2017-12-25:
Hi.

I've been working with a toy C# wrapper/binding for libgphoto2.  So far things have been going well on Linux using Mono, and I assume things should be as smooth when I test it on OS X.  I want to test this wrapper on Windows as well (and hopefully under the .NET runtime).  Are there some build steps available for this library on Windows?

peterbud, 2017-12-25:
Start with the [build script](https://github.com/Alexpux/MINGW-packages/blob/master/mingw-w64-libgphoto2/PKGBUILD), it also lists all the direct dependencies.

francescojordan, 2018-02-21:
Hi peterbud! 
Thank so much for your contribute. I have a question, can i use your library to run the "--get config shuttercounter " command?
i'm trying to build a shutter counter for windows but i have not understood if i can use you lib.

Thank you so much for your help!

best regards!

Francesco

peterbud, 2018-02-21:
@francescojordan  - I have never tried that paticular setting, but it should work as the windows binaries have no specific limitations compared to other platforms.

If you build a software using libgphoto on Windows bear in mind that normal default camera drivers on Windows unfortunately do not have libusb interface, therefore ligphoto cannot use cameras with those drivers. You need to install (by hand) drivers which do support libusb interface.

You can read more about that here:
https://github.com/libusb/libusb/wiki/Windows

And here is a software which makes easy to install such drivers:
https://github.com/pbatard/libwdi/wiki/Zadig

define-private-public, 2018-03-06:
Sorry to keep on bumping a closed issue, but I was able to get `libgphoto2` to build under windows using Cygwin with pretty minimal effort.  I haven't tested if the DLL works with MSVC (and using my C# bindings it failed, probably due to a 32/64 bit mismatch issue).

If I can get it working, would you be willing to merge in a `HOWTO-WINDOWS` doc?

faceang, 2018-06-26:
mingw64 gphoto2: Error (-53: 'Could not claim the USB device')

blakerutledge, 2018-10-25:
Hello @peterbud! 

First off, thank you so much for porting to Windows! I have been using `gphoto2` for years but now am working on a larger application that has a Windows 10 requirement, and you guys seem to have done it but something isn't working for me, following all of the steps above. Any insight or help at all would be super appreciated!! (and apologies for also bumping a closed issue...)

- installed `MSYS2`
- installed gphoto2 and libgphoto2 via `pacman -S mingw-w64-x86_64-gphoto2`, everything seems successful
- In the `MINGW64` shell, I try and use gphoto2, and even with the simplest `gphoto2 --version` I get:

```
    C:/msys64/mingw64/bin/gphoto2.exe: error while loading shared libraries: libreadline7.dll: 
    cannot open shared object file: No such file or directory
```
In the `mingw64\bin` folder, `libreadline7.dll` does exists, so it seems gphoto2 is simply looking for it elsewhere. I've tried adding `C:\msys64\mingw64\bin` to the PATH env vars, but this doesn's solve the issue (does allow me to get similar results in a `CMD` shell so that's something). Same results on two machines. Might something have broken with dependencies or a new version of gphoto2? I'm totally new to MSYS2 so I could be missing something simple... 

Thanks a million, mate!

peterbud, 2018-10-25:
Hi @blakerutledge 
Needless to say it's working here...

```
$ gphoto2 --version
gphoto2 2

Copyright (c) 2000-2018 Lutz Mueller and others

gphoto2 comes with NO WARRANTY, to the extent permitted by law. You may
redistribute copies of gphoto2 under the terms of the GNU General Public
License. For more information about these matters, see the files named COPYING.

This version of gphoto2 is using the following software versions and options:
gphoto2         2.5.17         x86_64-w64-mingw32-gcc, popt(m), exif, no cdk, no aa, jpeg, readline
libgphoto2      2.5.19         all camlibs, x86_64-w64-mingw32-gcc, ltdl, EXIF
libgphoto2_port 0.12.0         iolibs: disk ptpip usb1, x86_64-w64-mingw32-gcc, ltdl, USB, no serial

```

Question: after you have installed MSYS2, have you updated the MSYS2 system  by opening the MSYS2 shell, and typing `pacman -Syu` ?

blakerutledge, 2018-10-25:
Thanks for the reply @peterbud! Very glad to hear it still builds on your end -- Yes, I did that step, and then it asked me to close the shell and run `pacman -Su` to complete the update. I'm on a 64bit machine as well, and am currently reading through MSYS installation troubleshooting to see if there's more I can try.

define-private-public, 2018-10-25:
I was able to build libgphoto2 using MSYS2, but I wasn't able to get an actual .dll file.  Is there something else that I was missing from the compiler options.

blakerutledge, 2018-10-25:
When I install `MSYS2`  into the default `C:\msys64`, open automatically after install, and run `pacman -Syu`, i get this:
```
:: Proceed with installation? [Y/n] y
:: Retrieving packages...
 filesystem-2018.08-1-...    43.7 KiB  42.7M/s 00:00 [###########################] 100%
 msys2-runtime-2.11.1-...     2.5 MiB  1362K/s 00:02 [###########################] 100%
 mintty-1~2.9.1-1-x86_64    288.3 KiB  1737K/s 00:00 [###########################] 100%
 pacman-mirrors-201806...    17.1 KiB  8.37M/s 00:00 [###########################] 100%
 pacman-5.1.1-3-x86_64       12.8 MiB  2.30M/s 00:06 [###########################] 100%
(5/5) checking keys in keyring                       [###########################] 100%
(5/5) checking package integrity                     [###########################] 100%
(5/5) loading package files                          [###########################] 100%
(5/5) checking for file conflicts                    [###########################] 100%
(5/5) checking available disk space                  [###########################] 100%
warning: could not get file information for home/
warning: could not get file information for opt/
:: Processing package changes...
(1/5) upgrading filesystem                           [###########################] 100%
(2/5) upgrading msys2-runtime                        [###########################] 100%
(3/5) upgrading mintty                               [###########################] 100%
(4/5) upgrading pacman-mirrors                       [###########################] 100%
(5/5) upgrading pacman                               [###########################] 100%
warning: terminate MSYS2 without returning to shell and check for updates again
warning: for example close your terminal window instead of calling exit
```

Are those two warnings for `home/` and `opt/` a clue?  Those directories are empty on my file system. 

At this point, I close the msys64 shell, (processes are running, close anyway...), open a new one via the Start Menu, and according to the install website, I then run `pacman -Su`. This all looks successful, but get one warning: `warning: could not get file information for autorebasebase1st.bat`, but after the update this file exists in `C:\msys64`.

Then I install gphoto & libgphoto with `pacman -S mingw-w64-x86_64-gphoto2`. Verify they are installed with `pacman -Ss gphoto2`, and still get this, but the .dll does in fact exist in `C:\msys64\mingw64\bin`:

```
$ gphoto2 --version
C:/msys64/mingw64/bin/gphoto2.exe: error while loading shared libraries: 
libreadline7.dll: cannot open shared object file: No such file or directory
```

blakerutledge, 2018-10-25:
@peterbud Now here's something -- I copy/pasted `libreadline7.dll` into various folders to see where it was looking for the dll. Found msys wanted it in `C:\msys64\usr\bin` instead of `C:\msys64\mingw64\bin`. So I got a new error, that `libpopt-0.dll` was missing, copied it over. Continued this process a few times, and got this (which is the same error I get if I add `C:\msys64\mingw64\bin` to my user PATH env var, which makes some amount of sense):

```
$ gphoto2 --version
C:/msys64/mingw64/bin/gphoto2.exe: error while loading shared libraries: ?: 
cannot open shared object file: No such file or directory
```

Does anything about this jump out to you that I could investigate? Not seeing any `?` filenames when I search. Again thank you immensely for your time, if you have any to spare!

peterbud, 2018-10-26:
@blakerutledge : which SHELL you are using: MSYS2 or MINGW32 or MINGW64?
You need to use MSYS2 shell to run pacman. 
You need to use MINGW64 shell to run software which are MINGW64 packages

I believe you tried to launch gphoto2 from an MSYS2 shell or you have a wrong PATH. Can you share what do you see when you enter `echo $MSYSTEM`? What do you see when you enter `echo $PATH`?

Reading MSYS2 [wiki](https://github.com/msys2/msys2/wiki) could help, especially [this part](https://github.com/msys2/msys2/wiki/MSYS2-introduction)

blakerutledge, 2018-10-26:
Hey @peterbud 
Yea, been reading the wiki up and down -- anything with `pacman` I use the MSYS2 shell, and am trying to launch `gphoto2` with MINGW64 shell. Thinking something is up with the PATH.
Yes, in the MINGW64 shell, `echo $MSYSTEM` yeilds `MINGW64`, makes sense. In the same shell, `echo $PATH` yeilds:
```
/mingw64/bin:/usr/local/bin:/usr/bin:/bin:/c/Windows/System32:/c/Windows:/c/Windows/System32/Wbem:/c/Windows/System32/WindowsPowerShell/v1.0/
```

I see that the first entry directs it to the binaries in mingw64 where `gphoto.exe` and its dependencies exist. But MINGW64 cant find the packages there. For some reason, I get past the missing `libreadline` etc and on to the missing `?` package with `C:\msys64\mingw64\bin` in my user PATH, even without an env var that is `MSYS2_PATH_TYPE` as `inherit`. Seems contrary to my understanding of the wiki..? Could it be conflicting with something else on my PC?

peterbud, 2018-10-27:
That is really weird. So the PATH and the SHELL is OK. Next step: could you run `ldd /mingw64/bin/gphoto2`

For me it gives the following:

```
$ ldd /mingw64/bin/gphoto2
        ntdll.dll => /c/WINDOWS/SYSTEM32/ntdll.dll (0x7ffbbd810000)
        KERNEL32.DLL => /c/WINDOWS/System32/KERNEL32.DLL (0x7ffbbce70000)
        KERNELBASE.dll => /c/WINDOWS/System32/KERNELBASE.dll (0x7ffbb9d50000)
        ADVAPI32.dll => /c/WINDOWS/System32/ADVAPI32.dll (0x7ffbbcf40000)
        msvcrt.dll => /c/WINDOWS/System32/msvcrt.dll (0x7ffbbc3a0000)
        sechost.dll => /c/WINDOWS/System32/sechost.dll (0x7ffbbc490000)
        RPCRT4.dll => /c/WINDOWS/System32/RPCRT4.dll (0x7ffbbca90000)
        libgphoto2-6.dll => /mingw64/bin/libgphoto2-6.dll (0x62280000)
        libgphoto2_port-12.dll => /mingw64/bin/libgphoto2_port-12.dll (0x66600000)
        libexif-12.dll => /mingw64/bin/libexif-12.dll (0x6b440000)
        libwinpthread-1.dll => /mingw64/bin/libwinpthread-1.dll (0x64940000)
        SHELL32.dll => /c/WINDOWS/System32/SHELL32.dll (0x7ffbbad20000)
        libpopt-0.dll => /mingw64/bin/libpopt-0.dll (0x633c0000)
        libreadline7.dll => /mingw64/bin/libreadline7.dll (0x6e780000)
        libintl-8.dll => /mingw64/bin/libintl-8.dll (0x61cc0000)
        cfgmgr32.dll => /c/WINDOWS/System32/cfgmgr32.dll (0x7ffbba9d0000)
        USER32.dll => /c/WINDOWS/System32/USER32.dll (0x7ffbbcbf0000)
        libltdl-7.dll => /mingw64/bin/libltdl-7.dll (0x6d480000)
        win32u.dll => /c/WINDOWS/System32/win32u.dll (0x7ffbba9b0000)
        libsystre-0.dll => /mingw64/bin/libsystre-0.dll (0x6bcc0000)
        ucrtbase.dll => /c/WINDOWS/System32/ucrtbase.dll (0x7ffbb9ba0000)
        GDI32.dll => /c/WINDOWS/System32/GDI32.dll (0x7ffbbc460000)
        libtermcap-0.dll => /mingw64/bin/libtermcap-0.dll (0x6ac40000)
        shcore.dll => /c/WINDOWS/System32/shcore.dll (0x7ffbbd0a0000)
        libiconv-2.dll => /mingw64/bin/libiconv-2.dll (0x66000000)
        gdi32full.dll => /c/WINDOWS/System32/gdi32full.dll (0x7ffbb9a00000)
        combase.dll => /c/WINDOWS/System32/combase.dll (0x7ffbbc760000)
        libtre-5.dll => /mingw64/bin/libtre-5.dll (0x63bc0000)
        msvcp_win.dll => /c/WINDOWS/System32/msvcp_win.dll (0x7ffbb98e0000)
        bcryptPrimitives.dll => /c/WINDOWS/System32/bcryptPrimitives.dll (0x7ffbb9980000)
        windows.storage.dll => /c/WINDOWS/System32/windows.storage.dll (0x7ffbba1d0000)
        profapi.dll => /c/WINDOWS/System32/profapi.dll (0x7ffbb9890000)
        powrprof.dll => /c/WINDOWS/System32/powrprof.dll (0x7ffbb9810000)
        shlwapi.dll => /c/WINDOWS/System32/shlwapi.dll (0x7ffbbcd90000)
        kernel.appcore.dll => /c/WINDOWS/System32/kernel.appcore.dll (0x7ffbb98c0000)
        cryptsp.dll => /c/WINDOWS/System32/cryptsp.dll (0x7ffbbaa20000)

```
And the next question: How do you start your terminal? Have you edited any files, for example `/etc/profile` or `msys2_shell.cmd`?

blakerutledge, 2018-10-31:
@peterbud Sorry for the delay, started a new gig and haven't been near my Windows setup.  When I run `ldd /mingw64/bin/gphoto2` I get different responses... literally running these commands 10 seconds after one another. First try I got:
```
ntdll.dll => /c/WINDOWS/SYSTEM32/ntdll.dll (0x7ff855ad0000)
KERNEL32.DLL => /c/WINDOWS/System32/KERNEL32.DLL (0x7ff853cb0000)
KERNELBASE.dll => /c/WINDOWS/System32/KERNELBASE.dll (0x7ff852040000)
ADVAPI32.dll => /c/WINDOWS/System32/ADVAPI32.dll (0x7ff855310000)
msvcrt.dll => /c/WINDOWS/System32/msvcrt.dll (0x7ff853880000)
sechost.dll => /c/WINDOWS/System32/sechost.dll (0x7ff853e40000)
RPCRT4.dll => /c/WINDOWS/System32/RPCRT4.dll (0x7ff8539e0000)
libexif-12.dll => /mingw64/bin/libexif-12.dll (0x6b440000)
libwinpthread-1.dll => /mingw64/bin/libwinpthread-1.dll (0x64940000)
libgphoto2-6.dll => /mingw64/bin/libgphoto2-6.dll (0x62280000)
libgphoto2_port-12.dll => /mingw64/bin/libgphoto2_port-12.dll (0x66600000)
SHELL32.dll => /c/WINDOWS/System32/SHELL32.dll (0x7ff853ea0000)
libpopt-0.dll => /mingw64/bin/libpopt-0.dll (0x633c0000)
libreadline7.dll => /mingw64/bin/libreadline7.dll (0x6e780000)
cfgmgr32.dll => /c/WINDOWS/System32/cfgmgr32.dll (0x7ff852f10000)
libintl-8.dll => /mingw64/bin/libintl-8.dll (0x61cc0000)
USER32.dll => /c/WINDOWS/System32/USER32.dll (0x7ff8553c0000)
ucrtbase.dll => /c/WINDOWS/System32/ucrtbase.dll (0x7ff852d60000)
win32u.dll => /c/WINDOWS/System32/win32u.dll (0x7ff852ab0000)
libsystre-0.dll => /mingw64/bin/libsystre-0.dll (0x6bcc0000)
GDI32.dll => /c/WINDOWS/System32/GDI32.dll (0x7ff8537e0000)
shcore.dll => /c/WINDOWS/System32/shcore.dll (0x7ff855700000)
combase.dll => /c/WINDOWS/System32/combase.dll (0x7ff8534b0000)
gdi32full.dll => /c/WINDOWS/System32/gdi32full.dll (0x7ff851ea0000)
msvcp_win.dll => /c/WINDOWS/System32/msvcp_win.dll (0x7ff852cc0000)
bcryptPrimitives.dll => /c/WINDOWS/System32/bcryptPrimitives.dll (0x7ff8522c0000)
windows.storage.dll => /c/WINDOWS/System32/windows.storage.dll (0x7ff8523a0000)
shlwapi.dll => /c/WINDOWS/System32/shlwapi.dll (0x7ff853810000)
kernel.appcore.dll => /c/WINDOWS/System32/kernel.appcore.dll (0x7ff851e60000)
profapi.dll => /c/WINDOWS/System32/profapi.dll (0x7ff851e40000)
powrprof.dll => /c/WINDOWS/System32/powrprof.dll (0x7ff851de0000)
FLTLIB.DLL => /c/WINDOWS/System32/FLTLIB.DLL (0x7ff851e30000)
```

Tried again a minute later and get seemingly all the same entries, plus the `??? => ???` one.
```
ntdll.dll => /c/WINDOWS/SYSTEM32/ntdll.dll (0x7ff855ad0000)
KERNEL32.DLL => /c/WINDOWS/System32/KERNEL32.DLL (0x7ff853cb0000)
KERNELBASE.dll => /c/WINDOWS/System32/KERNELBASE.dll (0x7ff852040000)
ADVAPI32.dll => /c/WINDOWS/System32/ADVAPI32.dll (0x7ff855310000)
msvcrt.dll => /c/WINDOWS/System32/msvcrt.dll (0x7ff853880000)
sechost.dll => /c/WINDOWS/System32/sechost.dll (0x7ff853e40000)
RPCRT4.dll => /c/WINDOWS/System32/RPCRT4.dll (0x7ff8539e0000)
libwinpthread-1.dll => /mingw64/bin/libwinpthread-1.dll (0x64940000)
libexif-12.dll => /mingw64/bin/libexif-12.dll (0x6b440000)
libgphoto2-6.dll => /mingw64/bin/libgphoto2-6.dll (0x62280000)
libgphoto2_port-12.dll => /mingw64/bin/libgphoto2_port-12.dll (0x66600000)
SHELL32.dll => /c/WINDOWS/System32/SHELL32.dll (0x7ff853ea0000)
libpopt-0.dll => /mingw64/bin/libpopt-0.dll (0x633c0000)
libreadline7.dll => /mingw64/bin/libreadline7.dll (0x6e780000)
cfgmgr32.dll => /c/WINDOWS/System32/cfgmgr32.dll (0x7ff852f10000)
USER32.dll => /c/WINDOWS/System32/USER32.dll (0x7ff8553c0000)
libintl-8.dll => /mingw64/bin/libintl-8.dll (0x61cc0000)
??? => ??? (0x150000)
ucrtbase.dll => /c/WINDOWS/System32/ucrtbase.dll (0x7ff852d60000)
win32u.dll => /c/WINDOWS/System32/win32u.dll (0x7ff852ab0000)
libsystre-0.dll => /mingw64/bin/libsystre-0.dll (0x6bcc0000)
shcore.dll => /c/WINDOWS/System32/shcore.dll (0x7ff855700000)
GDI32.dll => /c/WINDOWS/System32/GDI32.dll (0x7ff8537e0000)
combase.dll => /c/WINDOWS/System32/combase.dll (0x7ff8534b0000)
gdi32full.dll => /c/WINDOWS/System32/gdi32full.dll (0x7ff851ea0000)
msvcp_win.dll => /c/WINDOWS/System32/msvcp_win.dll (0x7ff852cc0000)
bcryptPrimitives.dll => /c/WINDOWS/System32/bcryptPrimitives.dll (0x7ff8522c0000)
windows.storage.dll => /c/WINDOWS/System32/windows.storage.dll (0x7ff8523a0000)
shlwapi.dll => /c/WINDOWS/System32/shlwapi.dll (0x7ff853810000)
kernel.appcore.dll => /c/WINDOWS/System32/kernel.appcore.dll (0x7ff851e60000)
profapi.dll => /c/WINDOWS/System32/profapi.dll (0x7ff851e40000)
powrprof.dll => /c/WINDOWS/System32/powrprof.dll (0x7ff851de0000)
FLTLIB.DLL => /c/WINDOWS/System32/FLTLIB.DLL (0x7ff851e30000)
```

Finally, I've tried the entire process a few times from fresh MSYS2 installs, with various ways of opening the shells... Originally as the website instructions seem to say, I open via the Start Menu: search and open `MSYS2 MSYS` to install via pacman, and `MSYS2 MinGW 64-bit` to run any tests / try to use gphoto. After poking around, Ive tried running `msys2.exe` and `mingw64.exe` directly in `C:\msys64`, and lastly tried running `C:\msys64\usr\bin\mintty.exe` and selecting which shell via the gui there. Not noticing any differences with this. 

Does any of this reveal anything off to you about my system? Ill try just about anything. I have a dev machine that I could wipe with a fresh Windows install.... again -- thank you immensely for your time!!

peterbud, 2018-11-02:
Gees - I have the same symptoms. I have never seen anything like that, computers should be a deterministic machines. Let me ask some guys who are more knowledgeable.

peterbud, 2018-11-02:
OK, looks like we should use `ntldd`.

You can install `ntldd`:
` pacman -S mingw-w64-x86_64-ntldd-git`

And then:
```
$ ntldd /mingw64/bin/gphoto2
        ADVAPI32.dll => C:\WINDOWS\SYSTEM32\ADVAPI32.dll (0x0000000000d80000)
        KERNEL32.dll => C:\WINDOWS\SYSTEM32\KERNEL32.dll (0x0000000000e30000)
        msvcrt.dll => C:\WINDOWS\SYSTEM32\msvcrt.dll (0x0000000000e30000)
        libwinpthread-1.dll => C:\msys64\mingw64\bin\libwinpthread-1.dll (0x0000000000020000)
        libexif-12.dll => C:\msys64\mingw64\bin\libexif-12.dll (0x0000000000c30000)
        libgphoto2-6.dll => C:\msys64\mingw64\bin\libgphoto2-6.dll (0x00000000001d0000)
        libgphoto2_port-12.dll => C:\msys64\mingw64\bin\libgphoto2_port-12.dll (0x0000000000c30000)
        libpopt-0.dll => C:\msys64\mingw64\bin\libpopt-0.dll (0x00000000001d0000)
        libreadline7.dll => C:\msys64\mingw64\bin\libreadline7.dll (0x0000000000c30000)

```

blakerutledge, 2018-12-02:
Hello again @peterbud! I'm finally back able to work on this again.

Really odd.. So `ntldd` gives me seemingly equivalent results aside from memory addresses (assuming that's what those are haha..)
```
$ ntldd /mingw64/bin/gphoto2
        ADVAPI32.dll => C:\WINDOWS\SYSTEM32\ADVAPI32.dll (0x0000000000af0000)
        KERNEL32.dll => C:\WINDOWS\SYSTEM32\KERNEL32.dll (0x0000000000e20000)
        msvcrt.dll => C:\WINDOWS\SYSTEM32\msvcrt.dll (0x0000000000e20000)
        libwinpthread-1.dll => C:\msys64\mingw64\bin\libwinpthread-1.dll (0x0000000000020000)
        libexif-12.dll => C:\msys64\mingw64\bin\libexif-12.dll (0x0000000000af0000)
        libgphoto2-6.dll => C:\msys64\mingw64\bin\libgphoto2-6.dll (0x00000000001d0000)
        libgphoto2_port-12.dll => C:\msys64\mingw64\bin\libgphoto2_port-12.dll (0x0000000000af0000)
        libpopt-0.dll => C:\msys64\mingw64\bin\libpopt-0.dll (0x00000000001d0000)
        libreadline7.dll => C:\msys64\mingw64\bin\libreadline7.dll (0x0000000000af0000)
```

On the side I am looking into using the Canon SDK for Windows to interface with my cameras, but there are a number of reasons why gphoto2 would be preferable. Is there anything left that you know of that I can try testing?

blakerutledge, 2018-12-12:
@peterbud All good -- I think I've found an alternate solution, and I'm sorry to leave gphoto behind but windows is a larger requirement. Thanks so much for helping me think this through, and especially thank you for putting the windows port together! It's a great contribution.

define-private-public, 2018-12-29:
How is `darktable` packaging/building it's `libgphoto2` dep?

As a side note, what is `gdlib` used for in `libgphoto2`?  I've been able to put together a CMake file that downloads all of the necessary deps from source for libgphoto2, and compiles them under an MSYS2 environment.  So far it seems to be working, but I still need to do some testing; I'm mainly doing this so I can distribute the native DLLs alongside `libgphoto2Sharp`/`SharpCamera`.

peterbud, 2018-12-29:
For packaging libgphoto2 into darktable see these places:
https://github.com/darktable-org/darktable/blob/master/cmake/windows-macros.cmake#L153
https://github.com/darktable-org/darktable/blob/master/cmake/windows-macros.cmake#L160

And in the installer how to add environment variables:
https://github.com/darktable-org/darktable/blob/b4493219e0cdb1bc2bcb5a5e3c00b74e45c02d0e/cmake/modules/NSIS.template.in#L717

msmeissn, 2018-12-29:
libgd is used for converting images, only used in the pictureframe drivers currently.

define-private-public, 2018-12-29:
Pictureframe?  Is that one of those devices that is a flat screen and rotates though a series of photos?

My main concern is for working with DSLRs (tethering mainly), but I really don't want to distribute a DLL that is missing any important features.  Adding in the build commands for libGD it's easy, but it has it's in slew of external deps, and that's for supporting multiple image formats.

msmeissn, 2018-12-29:
correct, so libgd is likely not required for your usecase.

Fred3D-tech, 2020-07-10:
tried all the solutions given here..  downloaded MSYS, then used pacman -S mingw-w64-x86_64-gphoto2..
It looks like you need to configure manually some paths but could not get any information about that.. what I see is that using gphoto2 in a command prompt give this result:
'gphoto2.exe' is not recognized as an internal or external command,
operable program or batch file.
please make a serious turorial ! 
...