gphoto/libgphoto2 #101: gp_camera_file_read vs. gp_camera_file_get

peterbud, 2016-10-15:
I was using the sample program sample-capture and sample-trigger-capture, and found the following on my Windows port:
- call to function **gp_camera_file_get** used in sample-capture WORKS, captures the image with both way withoout any problem
- call to function **gp_camera_file_read** used in sample-trigger-capture with the same camrea ad setting FAILS. As far as I can debug, gp_camera_file_read calls gp_filesystem_read_file, which then calls read_file_func. This one impleneted in ptp2/libary.c, and has a define for **folder_to_storage**, which expects that the folder paramter should have the name like "**/store_xxxxxxxxx/**"

Question: is this really a necessary check? Why then the gp_camera_file_get works perfectly?

Sorry if the question is stupid, I have started getting familiar with the library only recently.

romainreignier, 2018-12-26:
I have faced the same issue.
I wanted to avoid the *unnecessary* write/delete operation on host filesystem.
But I have figured out that the function `gp_camera_file_read` is never used in the `libgphoto2` and `gphoto2`.

romainreignier, 2018-12-26:
In my case, for a Sony PTP camera, the `path->folder` value is overridden in the capture function here:
https://github.com/gphoto/libgphoto2/blob/bc181c8ae3d7eaf175455697678e3b24ae1cdfcd/camlibs/ptp2/library.c#L4215

So the check in `folder_to_storage` cannot be valid.

msmeissn, 2018-12-27:
Hmm. The gp_camera_file_read does not cover this case, it should check the filesystem backed cached storage here. This is missing. the _read function is not complete :/

But gp_camera_file_get() does not touch the host filesystem, unless you use gp_file_new_from_fd() or so, so you can still use that.

romainreignier, 2018-12-27:
Oh! I've missed the point that I can initialize a `CameraFile` from memory with `gp_file_new()` and not `gp_file_new_from_fd()`.
Thanks.
Until now, I was creating a temporary file with `mkstemp()` and then delete it with `unlink()`.