gphoto/libgphoto2 #102: Sony SLT-A58 preview implementation

Creatux, 2016-10-18:
Hello to all.

i'm trying to get the preview mode of my SLT-A58 (plugged into a debian 8 Laptop) and entangle to work.
But i noticed that the preview mode for the A58 isn't implemented...
so i would like to a request the implementation of said function.

if there are any informations you need, i will try to deliver those to you.

greetings Creatux

msmeissn, 2016-10-23:
The windows app I had does not have preview mode for the SLT A58 ... have you seen an app that can show the preview for it?

Creatux, 2016-10-24:
I don't use windows anymore... 
so i can only assume that the official remote control app from sony has preview functions???

As mentioned above i'm a Linux-user.  But you can check for yourself;)
open up the sony website, click on support and search for the SLT A58.
then try the remote control tool.

Regards, Creatux

P.s.

The preview mode is also more commonly referred to as Tethered shooting.
a good website to look at is tethertools. ;)

msmeissn, 2016-10-24:
The windows application I used did not show the preview... i also currently see no uninvestigated opcodes, but that does not mean much.

I seem to have used the tool you mention from http://support.d-imaging.sony.co.jp/imsoft/Win/rcc/us.html with my Sony SLT A58

Tethering would usually any form of remote control over cable, the wording is not clearly defined by people.

msmeissn, 2020-03-01:
sadly preview is not possible for older Sony Alpha like the A58

closed 2020-03-01 by msmeissn