gphoto/libgphoto2 #103: Mac Port installation of libgphoto2 fails

kaelte, 2016-10-26:
I am not sure whether this shall be reported here or to the Mac Port Developers.

The error message suggests this place:

```
:info:build aox/library.c:214:10: warning: equality comparison with extraneous parentheses [-Wparentheses-equality]
:info:build                 if ((w == 320)) {
:info:build                      ~~^~~~~~
```

## Platform

iMac (Retina 5K, 27-inch, Late 2014)

```
$ xcodebuild -version

Xcode 8.0
Build version 8A218a
```

## Building

```
$ sudo port install libgphoto2 1> libgphoto2_2016-10-26.log 2>libgphoto2_2016-10-26.err

$ cat libgphoto2_2016-10-26.log

--->  Computing dependencies for libgphoto2
--->  Fetching distfiles for libgphoto2
--->  Verifying checksums for libgphoto2
--->  Extracting libgphoto2
--->  Configuring libgphoto2
--->  Building libgphoto2
Please see the log file for port libgphoto2 for details:
    /opt/local/var/macports/logs/_opt_local_var_macports_sources_rsync.macports.org_release_tarballs_ports_devel_libgphoto2/libgphoto2/main.log
To report a bug, follow the instructions in the guide:
    http://guide.macports.org/#project.tickets

$ cat libgphoto2_2016-10-26.err

Error: org.macports.build for port libgphoto2 returned: command execution failed
Error: Processing of port libgphoto2 failed

$ tail -68 /opt/local/var/macports/logs/_opt_local_var_macports_sources_rsync.macports.org_release_tarballs_ports_devel_libgphoto2/libgphoto2/main.log | head -16
:info:build libtool: compile:  /usr/bin/clang -DHAVE_CONFIG_H -I. -I.. -D_GPHOTO2_INTERNAL_CODE -DLOCALEDIR=\"/opt/local/share/locale\" -DCAMLIBS=\"/opt/local/lib/libgphoto2/2.5.10\" -I.. -I.. -I../libgphoto2_port -I../libgphoto2 -I../libgphoto2 -I/opt/local/include -pipe -Os -arch x86_64 -Wall -Wmissing-declarations -Wmissing-prototypes -MT agfa-cl20/agfa_cl20.lo -MD -MP -MF agfa-cl20/.deps/agfa_cl20.Tpo -c agfa-cl20/agfa_cl20.c  -fno-common -DPIC -o agfa-cl20/.libs/agfa_cl20.o
:info:build libtool: compile:  /usr/bin/clang -DHAVE_CONFIG_H -I. -I.. -D_GPHOTO2_INTERNAL_CODE -DLOCALEDIR=\"/opt/local/share/locale\" -DCAMLIBS=\"/opt/local/lib/libgphoto2/2.5.10\" -I.. -I.. -I../libgphoto2_port -I../libgphoto2 -I../libgphoto2 -I/opt/local/include -I/opt/local/include -pipe -Os -arch x86_64 -Wall -Wmissing-declarations -Wmissing-prototypes -MT ax203/ax203_la-library.lo -MD -MP -MF ax203/.deps/ax203_la-library.Tpo -c ax203/library.c  -fno-common -DPIC -o ax203/.libs/ax203_la-library.o
:info:build libtool: compile:  /usr/bin/clang -DHAVE_CONFIG_H -I. -I.. -D_GPHOTO2_INTERNAL_CODE -DLOCALEDIR=\"/opt/local/share/locale\" -DCAMLIBS=\"/opt/local/lib/libgphoto2/2.5.10\" -I.. -I.. -I../libgphoto2_port -I../libgphoto2 -I../libgphoto2 -I/opt/local/include -pipe -Os -arch x86_64 -Wall -Wmissing-declarations -Wmissing-prototypes -MT aox/library.lo -MD -MP -MF aox/.deps/library.Tpo -c aox/library.c  -fno-common -DPIC -o aox/.libs/library.o
:info:build libtool: compile:  /usr/bin/clang -DHAVE_CONFIG_H -I. -I.. -D_GPHOTO2_INTERNAL_CODE -DLOCALEDIR=\"/opt/local/share/locale\" -DCAMLIBS=\"/opt/local/lib/libgphoto2/2.5.10\" -I.. -I.. -I../libgphoto2_port -I../libgphoto2 -I../libgphoto2 -I/opt/local/include -I/opt/local/include -pipe -Os -arch x86_64 -Wall -Wmissing-declarations -Wmissing-prototypes -MT ax203/ax203_la-ax203_decode_yuv.lo -MD -MP -MF ax203/.deps/ax203_la-ax203_decode_yuv.Tpo -c ax203/ax203_decode_yuv.c  -fno-common -DPIC -o ax203/.libs/ax203_la-ax203_decode_yuv.o
:info:build libtool: compile:  /usr/bin/clang -DHAVE_CONFIG_H -I. -I.. -D_GPHOTO2_INTERNAL_CODE -DLOCALEDIR=\"/opt/local/share/locale\" -DCAMLIBS=\"/opt/local/lib/libgphoto2/2.5.10\" -I.. -I.. -I../libgphoto2_port -I../libgphoto2 -I../libgphoto2 -I/opt/local/include -pipe -Os -arch x86_64 -Wall -Wmissing-declarations -Wmissing-prototypes -MT aox/aox.lo -MD -MP -MF aox/.deps/aox.Tpo -c aox/aox.c  -fno-common -DPIC -o aox/.libs/aox.o
:info:build aox/library.c:214:10: warning: equality comparison with extraneous parentheses [-Wparentheses-equality]
:info:build                 if ((w == 320)) {
:info:build                      ~~^~~~~~
:info:build aox/library.c:214:10: note: remove extraneous parentheses around the comparison to silence this warning
:info:build                 if ((w == 320)) {
:info:build                     ~  ^     ~
:info:build aox/library.c:214:10: note: use '=' to turn this equality comparison into an assignment
:info:build                 if ((w == 320)) {
:info:build                        ^~
:info:build                        =
:info:build aox/library.c:220:10: warning: equality comparison with extraneous parentheses [-Wparentheses-equality]
```

kaelte, 2016-10-26:
I hope I can now attach main.zip.

[Uploading main.zip…]()

kaelte, 2016-10-26:
Using Safari I can apparently attach a file.
However, upload does not work with IceCat.
[main.zip](https://github.com/gphoto/libgphoto2/files/554111/main.zip)

msmeissn, 2016-10-26:
the problem is actually here:

:info:build In file included from ax203/ax203.c:38:
:info:build ax203:info:build In file included from ax203/ax203.c:38:
:info:build ax203/jpeg_memsrcdest.h:4:1: error: conflicting types for 'jpeg_mem_src'
:info:build jpeg_mem_src (j_decompress_ptr cinfo, unsigned char * buffer,
/jpeg_memsrcdest.h:4:1: error: conflicting types for 'jpeg_mem_src'
:info:build jpeg_mem_src (j_decompress_ptr cinfo, unsigned char * buffer,

and it is fixed by commit 0df14db3eda1b780a27b1c54bc8f251d15afebbc which will be in the next release.

closed 2016-11-20 by msmeissn