gphoto/libgphoto2 #105: Camera doesnt keep capture mode.

AlexisPotvin, 2016-11-01:
I have a small problem with one of my camera. It is a parrot sequioa. 
The problem is that it don't keep the capture mode. It always change it back to single shot. 
The problem occur both with trigger capture and capture image.

Houw could i resolve that problem ? thanks for your help!

msmeissn, 2016-11-07:
umm. i never heard of this camera before. 

can you attach --debug --debug-logfile=camera.log   outout?

AlexisPotvin, 2016-11-24:
Sorry for the delay i didn't had my hands on the camera. 
[camera.zip](https://github.com/gphoto/libgphoto2/files/611672/camera.zip)

msmeissn, 2016-11-29:
can you do this with the command you try? 

e.g.

gphoto2 --summary --debug --debug-logfile=camera.log

AlexisPotvin, 2016-12-13:
There you go!
[camera.txt](https://github.com/gphoto/libgphoto2/files/649009/camera.txt)

msmeissn, 2016-12-22:
nice. a completelz new camera vendor and model.

can you also attach:

gphoto2 --summary
gphoto2 --list-all-config

output?