gphoto/libgphoto2 #106: recording video on Sony a5100

averigo, 2016-11-02:
Hello,
Thank you very much, gphoto2 --capture-image-and-download --filename = image.jpg works fine on sony a5100 camera. 
Will you add video recording?

Andrei

msmeissn, 2016-11-02:
On Wed, Nov 02, 2016 at 01:05:39PM -0700, averigo wrote:
> Hello,
> Thank you very much, gphoto2 --capture-image-and-download --filename = image.jpg works fine on sony a5100 camera. 
> Will you add video recording?

There is the movie config option in PC Control mode:

gphoto2 --set-config movie=1 --wait-event=10s --set-config movie=0

would record a 10 second movie.

It can however not be retrieved in the PC Control mode currently,
only in the regular access mode.

Ciao, Marcus

tiepviet, 2016-12-14:
I have a problem. I want to take capture movie on Theta S over USB using gphoto2. Is it possible? I read gphoto2 library but it seems not support "capture movie". Please help me. Sorry for my bad English. Thank you so much.

msmeissn, 2019-02-10:
it currently is not possible to our knowledge. If the windows software allows it, we can recheck it.

closed 2019-02-10 by msmeissn