gphoto/libgphoto2 #107: Unable put file in camera flash card.

dhamp, 2016-11-09:


dhamp, 2016-11-09:
some logs^ https://gist.github.com/dhamp/aa8f49f4f489ba9ec035b05c3323f0fc

msmeissn, 2016-11-09:
the upload ability detection is a bit flaky. cameras rarely support upload, and it does not seem to be there in this case.

dhamp, 2016-11-13:
If it is a can not be determined by ptp2 protocol, then maybe it's worth to try to upload a test file in the camera, and only in case of success in the output display gphoto2 --summary that have upload support.
Although it is to some extent "crutch" and it only applies to gphoto2, but not libghoto2.
In any case, it seems to me, it can not be solved by libghoto2 library.

msmeissn, 2016-11-20:
i would prefer --summary not doing any active stuff :/ i could basically disable the "Upload" capable information for all cameras, as they rarely can do it