gphoto/libgphoto2 #108: Floating point issue with GP_WIDGET_RANGE widgets

nmrmak, 2016-11-12:
Hi,

There seems to be a problem with setting range widgets such as flashexposurecompensation on Nikon D7000.

For example, it is not possible to run something like:
gp_widget_set_value(widget, -7.0/3);
when evstep is 1/2, because:
-7.0/3 equals -2.333333254
step equals 0.3333333433

So it would work with -2.3333333433, but that's not what -7.0/3 evaluates to. I could wrap this setting in an elaborate function that would deal with steps, but that just doesn't seem like an elegant way to use the library.

Is there a better way to deal with floats in libgphoto2?

Thanks in advance,
Nebojsa

msmeissn, 2016-11-20:
hmm. the value you pass in via gp_widget_set_value is not checked internally against the range/step settings.  the multiplication *6.0 might result in -13.9999999 later  on though instead of -14

where do you actually see troubles?

nmrmak, 2017-01-15:
The problem can be reproduced by trying to set evstep to minus 7 thirds programatically, like so:
gp_widget_set_value(widget, -7.0/3)