gphoto/libgphoto2 #111: set-config capturetarget broken on EOS (at least 5Dm4)

axxel, 2016-11-24:
Calling _set-config capturetarget_ on an EOS 5Dm4 causes the camera to go into a "ERR 70" state where you have to power cycle the body to have it work again. I found the problem but I did find out when this broke. The reason why the canon 'crashes' is, because it gets send a SetDevicePropValueEx ptp command with a property id of 0x0. I have not found out whether this has always been the case or has been introduced somewhere between 2.5.8 (last version I have been working with) and 2.5.11 as I have only access to a 5Dm4 body at the moment.

The root cause is a non-meaningful handling of the "virtual" capturetarget property. In `_set_config()` the `cursub->putfunc` pointer points to `_put_CaptureTarget()` which sends a proper SetDevicePropValueEx command to the camera and returns GP_OK, which is immediately followed by another `ptp_canon_eos_setdevicepropvalue` call inside `_set_config()`, only this time the propid is the "virtual" id 0x0. This makes no sense. It only gets through the check for valid _canon_props_ inside `ptp_canon_eos_setdevicepropvalue` because the canon did indeed enumerate a property with id 0x0 (for whatever reason):

`0.192452 ptp                         (2): event 88: EOS prop 0000 info record, datasize is 16`

The question is: where is the most appropriate place to make sure this "virtual" propid gets not send to the camera and why has this worked before (at least with other canon bodies).

msmeissn, 2016-11-25:
The non_EOS code handled this correctly, I adjusted it for the EOS code too.

Must have happened during the single config rewrite of this function

msmeissn, 2016-11-25:
(please test the above commit or current git and report back ;)

closed 2016-11-25 by msmeissn

axxel, 2016-11-28:
passt :)