gphoto/libgphoto2 #116: Patch: Nikon AW130

kristapsdz, 2016-12-13:
This patch gets my Nikon AW130 recognised.  I'm not sure if the BROKEN_CAP still applies; I haven't tried it without.  (Should I?)

[library.c.diff.txt](https://github.com/gphoto/libgphoto2/files/650312/library.c.diff.txt)

msmeissn, 2016-12-14:
can you please truy it without?

kristapsdz, 2016-12-14:
Sure---is there a particular set of operations that can bring out whatever behaviour the BROKEN_CAP is meant to solve?

msmeissn, 2016-12-15:
--capture-image might not work if it is not set ... can you test both SDRAM and CARD capturetargets?

kristapsdz, 2016-12-15:
The BROKEN_CAP is required: --capture-image fails without it.  I'll look into the capturetargets later!  (This is all from libgphoto2 and gphoto2 in GitHub master on OpenBSD 6.0 using their default package configuration.)

closed 2016-12-17 by msmeissn