gphoto/libgphoto2 #117: Image doesn't save on SD card using --capture-image on Sony A5000

HanYangZhao, 2016-12-14:
Here's the output

➜  ~ gphoto2 --capture-image
New file is in location /capt0000.jpg on the camera
New file is in location /capt0000.arw on the camera
➜  ~ gphoto2 --list-files
There is no file in folder '/'.

I checked with the SD on the computer and there isn't any images there

--capture-image-and-download works

It is because the A5000 doesn't support the feature?
Thanks

elijahparker, 2016-12-15:
This is a Sony limitation -- saving to the camera's card is disabled while connected via USB

msmeissn, 2016-12-17:
thats what is also my understanding. it is currently a camera limitation.
(if there is some hidden setting we do not yet this might be the case, but i would assume it is not possible)

closed 2016-12-17 by msmeissn