gphoto/libgphoto2 #121: Can't set burstnumber on Nikon D500 Support

suu----, 2016-12-21:
I want to burst capture on D500 but can't set burstnumber.
I set burstnumber , but burstnumber was not changed. 
I'm ready to test. I'm on macOS El Captain.
[d500-list-all-config.txt](https://gist.github.com/suu----/2301252b159c473cfeaa9944b83b75d0)
[D500_lsusb.txt](https://gist.github.com/suu----/b5f246d25d3c2500f8a9df828beeeb75)
[set-config_burstnumber.txt](https://gist.github.com/suu----/0a6baf17f98c9a37ba032303e718626c)
[d500_summary.txt](https://gist.github.com/suu----/d05cedefb03d10d00dd4bcad176cc6b7)

rlamarche, 2016-12-21:
Hi,

Try to set the capturemode to burst. Here a extract from your d500-list-all-config.txt :

/main/capturesettings/capturemode
Label: Still Capture Mode
Type: RADIO
Current: Single Shot
Choice: 0 Single Shot
Choice: 1 Burst
Choice: 2 Continuous Low Speed
Choice: 3 Timer
Choice: 4 Mirror Up
Choice: 5 Quiet Release
Choice: 6 Unknown value 8018

suu----, 2016-12-21:
I set capturemode for Burst, but can't set burstnumber.
`sudo gphoto2 --get-config capturemode  
Label: Still Capture Mode                                                      
Type: RADIO
Current: Burst
Choice: 0 Single Shot
Choice: 1 Burst
Choice: 2 Continuous Low Speed
Choice: 3 Timer
Choice: 4 Mirror Up
Choice: 5 Quiet Release
Choice: 6 Unknown value 8018
[] ~
% sudo gphoto2 --get-config capturemode
Label: Still Capture Mode                                                      
Type: RADIO
Current: Continuous Low Speed
Choice: 0 Single Shot
Choice: 1 Burst
Choice: 2 Continuous Low Speed
Choice: 3 Timer
Choice: 4 Mirror Up
Choice: 5 Quiet Release
Choice: 6 Unknown value 8018
[] ~
% gphoto2 --set-config burstnumber=5            
[] ~
% gphoto2 --get-config burstnumber              
Label: Burst Number                                                            
Type: RANGE
Current: 1
Bottom: 1
Top: 200
Step: 1
`

rlamarche, 2016-12-21:
Ok I'll test this evening with my Nikon D800.

I did succeed to make burst capture with it. The process was :
* I set the capture mode to burst
* I set the burst number to 5 (for example)
* I launch 1 (one) capture, and the camera does 5 captures

suu----, 2016-12-21:
Thank you! The process was correct.
I can set burstnumber and burst capture on D3300. So I think it is unique problem of D500

rlamarche, 2016-12-21:
ok it's good to know !

I'm curious about the Nikon D500 remoting features so don't hesitate to ask if any other problem.
About the burstnumber, have you downloaded the D500 SDK document ? There is certainly some interesting informations in it.

I'll take a look in it

rlamarche, 2016-12-21:
What's the current mode of your camera ?

Try to manually set the selector on continous shooting mode, because between each gphoto2 command I suspect the camera to restore the settings.

I've just read this in the sdk :

Single frame | InvalidOnly | one image can be captured.

rlamarche, 2016-12-21:
When you run a gphoto2 command, a session with the camera is open, configuration set, and closed.

On next session open, the state of the camera is reflected to the mechanical state of the buttons.
I saw this with movie / photo liveview mode. I can set it with gphoto, but between each session, the camera restore the button's value.

So you might need to set multiple values with a single commandline with mant "--set-config " in it.

suu----, 2016-12-21:
```
gphoto2 --set-config burstnumber=2 --get-config burstnumber --capture-image
Label: Burst Number                                                            
Type: RANGE
Current: 2
Bottom: 1
Top: 200
Step: 1
```
I can burst capture on D500 by this command. Great!!! Thank you!!!

closed 2016-12-21 by suu----