gphoto/libgphoto2 #123: Support for live-view on Sony a6300/a7s/a7ii and newer models

xarple, 2016-12-28:
Hi, is there any way to make libgphoto2 supports for live-view functions on Sony a6300 or other newer models? I saw that these models are able to send live-view informations via usb becauses [RCCDroid](http://rccdroidpro-en.glensk.com) and [Capture One](https://www.phaseone.com/) are already have this functions. Thanks.

msmeissn, 2016-12-28:
So far we do not know if the Sonys can do that, we have not seen any indication of it. The Windows App itself does not support it last I checked.

rccdroidpro: Preview of taken pictures on an Android Display (Semi-Lifeview)  ... so not real live view
camera one: i dont see sony cameras in their compatibility list

msmeissn, 2016-12-28:
I see:

Ist Liveview möglich ?

NEU:  Liveview für A7S, A7II and A77IIN
Für andere Modelle ist es nicht möglich. Das ist kein Manko der App, sondern eine Designlücke in der Sony Firmware.

Thiswould mean these models might have it. Would need to find the opcodes

rlamarche, 2016-12-28:
There is a lot of interesting informations in the FAQ :

> In the meantime some of the Sony cameras have the option to transfer liveview data over USB, so that our app is able to show it on display (real liveview). This function is documented for A7S, A7II, A77II and A6300. We assume, that all newer models will have this option as well, but for now (status April 08. 2016) we do not really know, because we do not own A7SII. You are invited to install RCCDRoidPro and make a test. Please let us (and other interested users) know, if it works.

The author talked about a documentation. It would be interesting if we could get it...

rlamarche, 2016-12-28:
More infos here : http://briansmith.com/capture-one-pro-for-sony-adds-live-view-tethering-for-sony/

rlamarche, 2016-12-28:
And the complete list of Sony Cameras having liveview feature is here : https://www.phaseone.com/en/Products/Software/Capture-One-for-Sony/Supported-Cameras.aspx

Knowing that, I'm now interesting in buying a Sony Camera ;)

elijahparker, 2017-01-07:
I've got LV for Sony figured out -- I'll post the details on Monday and put together a pull request

msmeissn, 2017-01-11:
todays git now has sony live view support thanks to @elijahparker

closed 2017-01-11 by msmeissn

572017317, 2017-01-14:
Excuse me, Is there any way you can modify the SONY camera application installed on a web page recognition machine model with camera information?I want to export other SONY camera application, but I only have a SONY micro single, can be done by modifying the drivers of SONY web page? Hope to have the idea of tell me what method can achieve     
Can contact me by email, my email is      572017317@qq.com

msmeissn, 2017-01-15:
This software is not the sony official software, but an opensource project unrelated to SONY. it is not clear what you need.

572017317, 2017-01-16:
My camera is A6000 now, but I think in the website to download A7 or other type of APP, whether can be done by changing the driver?