gphoto/libgphoto2 #124: ISO 25 on Sony a6000

abrodkin, 2017-01-02:
I'm seeing quite a strange issue while playing with Sony a6000 camera.
The first strange thing is the camera reports support of ISO 25 even though there's no way to either manually set it or find any mentions of support for such a low ISO value in any camera's specs. Still see https://github.com/gphoto/libgphoto2/blob/master/camlibs/ptp2/cameras/sony-a6000.txt#L84:
```
/main/imgsettings/iso
Label: ISO Speed
Type: RADIO
Current: Auto ISO
Choice: 0 Auto ISO
Choice: 1 25
```

Now if I try to use that value for capturing image gphoto gets stuck doing a lot of data transfers, see my log of `gphoto2 --set-config /main/imgsettings/iso=25 --capture-image --debug` here http://pastebin.com/q845W2Yv

abrodkin, 2017-01-02:
That's funny: on the other end there's also some fake value which is 102400, see https://github.com/gphoto/libgphoto2/blob/master/camlibs/ptp2/cameras/sony-a6000.txt#L110

And behavior seems to be exactly the same.

msmeissn, 2017-01-02:
The list is actually what the camera reports to us.... can you attach

gphoto2 --summary

output too?

The setting of values is tricky, we can basically just send "1 down" or "1 up" commands and hope for the best.

elijahparker, 2017-01-02:
I've had this issue too -- for the A7II and others I've tested, a large range of ISO is given regardless of what the camera supports.  I don't remember this being an issue when I was parsing data from an A7 with my own embedded PTP library, but I could be wrong.  Here's the code I was using to parse the chunk that Sony sends for its settings: https://github.com/timelapseplus/TimelapsePlus-Firmware/blob/master/src/PTP.cpp#L2555-L2681

msmeissn, 2017-01-03:
I could reproduce on my Sony and fixed the code to abort in such cases.

msmeissn, 2017-01-03:
so ... it will be still be offered, but setting now leads to a clean exit and no endless loop.

closed 2017-01-03 by msmeissn

abrodkin, 2017-01-03:
Thanks for fixing this!