gphoto/libgphoto2 #127: Recording video with multiple cameras on Sony Theta S

tiepviet, 2017-01-10:
I can `start` recording video mode with gphoto2 on Sony Theta S. But I can't `stop` recording with gphoto2 command. Please help me. Sorry for my problem.

msmeissn, 2017-01-17:
how do you try to start and stop it?

--set-config movie=0 

should work?

tiepviet, 2017-01-18:
Thank you for reply my issue. I used below config: 
`--set-config movie=1` : start record (worked true)
`--set-config movie=0` : stop record (not work with transaction error). 
I'm working on Ricoh Theta S. Do I must to modify libgphoto2 to work with this camera?

msmeissn, 2017-01-19:
there needs to be a fix apparently. a debuglogfile would help, not sure when I find time to try it on my Theta

tiepviet, 2017-01-26:
@msmeissn Sorry for my delay. Thank your support. This is my log file after I tried to capture with ` gphoto2 --set-config movie=1 --wait-event=10s --set-config movie=0`
[capture-movie.txt](https://github.com/gphoto/libgphoto2/files/731529/capture-movie.txt)

msmeissn, 2017-01-27:
(please test this patch or current git ... it should fix it)

closed 2017-01-27 by msmeissn

tiepviet, 2017-01-28:
Thank you very much. Have a nice day!