gphoto/libgphoto2 #130: Canon exposure compensation RADIO settings for 1/2 ev step

nmrmak, 2017-01-16:
Hello again,

When I set my Canon EOS 650D to use 1/2-stop exposure level increments via Custom Functions from the menu, I get some weird values:

 gphoto2 --get-config=exposurecompensation
Label: Exposure Compensation
Type: RADIO
Current: 0
Choice: 0 -5
Choice: 1 Unknown value 00dc
Choice: 2 -4
Choice: 3 Unknown value 00e4
Choice: 4 -3
Choice: 5 -2.5
Choice: 6 -2
Choice: 7 -1.5
Choice: 8 -1.0
Choice: 9 -0.5
Choice: 10 0
Choice: 11 0.5
Choice: 12 1.0
Choice: 13 1.5
Choice: 14 2
Choice: 15 2.5
Choice: 16 3
Choice: 17 Unknown value 001c
Choice: 18 4
Choice: 19 Unknown value 0024
Choice: 20 5

nmrmak, 2017-01-16:
Created pull request for this at: https://github.com/gphoto/libgphoto2/pull/131

Please be gentle and point out if I did something wrong, because this is the first time I'm contributing to libgphoto2 and to any project on github.

msmeissn, 2017-01-16:
looks good, I merged it :)

closed 2017-01-16 by msmeissn

sinardet, 2024-09-14:
Hi everyone,
I'm totaly new and would like to use the same command gphoto2 --get-config exposurecompensation .

But when I try I've got only that below:
Label: Exposure Compensation
Readonly: 0
Type: RADIO
Current: 0
Choice: 0 0
END
I've got all the rest of commands like :gphoto2 --get-config iso or aperture etc they are working perfectly.
Please let me know what can I do to fix it thanks