gphoto/libgphoto2 #132: iOS 10.2 PTP Invalid Storage ID Error (-1: 'Unspecified error')

jritzerfeld, 2017-01-22:
Trying to access an iPhone running iOS 10.2 using openSUSE Leap 42.2 and KDE via kio_kamera does not work anymore while it worked using Leap 42.1 IIRC. I already updated libimobiledevice to master and libgphoto to 2.5.12 from the graphics repository. Using gphoto2 directly does not help, too. Sometimes it works and at other times it does not:
```
jan@karl:~> env LANG=C gphoto2 --debug --debug-logfile=my-logfile.txt --list-folders
There is 1 folder in folder '/'.                                               
 - store_00010001

*** Error ***              
PTP Invalid Storage ID
*** Error (-1: 'Unspecified error') ***  
```
Is there any other sensitive or private data in the debug logfile  that I should remove before attaching it here, besides the serial number?

msmeissn, 2017-01-22:
the serial number perhaps. you can also mail me at marcus@jet.franken.de if you care

jritzerfeld, 2017-01-23:
Thank you for your prompt response. I found and replaced the serial number by 'X's. I also uploaded a logfile of a successful "--list-folders" execution.
[my-logfile-failure.txt](https://github.com/gphoto/libgphoto2/files/722931/my-logfile-failure.txt)
[my-logfile-success.txt](https://github.com/gphoto/libgphoto2/files/722930/my-logfile-success.txt)

msmeissn, 2017-01-23:
on the failure log the camera ports 0x00000000 as storage id (its invalid according to the standard). on success it is 0x00010001 
it is not clear why it returns 0x0000000 in the failure case

jritzerfeld, 2017-01-25:
Could this be a problem in libimobiledevice regarding 10.2, then?

msmeissn, 2017-01-27:
Well, libimobiledevice is not used by libgphoto2 at all  / is unrelated.

I have currently no good idea what to do :/

jritzerfeld, 2017-01-27:
Okay. My idea was that libgphoto2 (indirectly) depends on libimobiledevice, effectively usbmuxd, for trusting the device. That is, if you do not pair your iOS device with your host, libgphoto2 will only see the "store_feedface" folder that does not contain any photo you took while the device was unlocked. Only a paired and unlocked iOS device will give access to the real folder named "store_xxxxxxxx", while xxxxxxx looks like the storage id you mentioned. Thus, I guessed that this trust might be temporarily lost so that the storage id reverts to 00000000.

ObsidianParis, 2017-02-08:
Hello everyone,
I'm experiencing the exact same problem. **iOS 10.2.1** on a **iPhone5**.

I usually have the problem when I try to import photos using **Shotwell** (currently on a **Fedora 22**). Since I usually dump them very unfrequently, I always face a new problem at the time I do…

My distribution was using version **2.5.7** of libgphoto2. I recompiled from the Git repository the last release, **2.5.12**, with no more results. I also have upgraded the same way my **imobiledevice** suite, which enabled me to reassociate my device a correct way and mount it properly with nautilus, but did not help neither (and even made the things worse during a while, for unknown reasons).

Last time, I reported my observations at the end of this bug : https://bugzilla.redhat.com/show_bug.cgi?id=1178256

It seems that periodically, Apple introduces (deliberately or not) a new bug that affects PTP transfers. Last time, the pictures was stored in a subdirectory under DCIM/ which name was changing each time the connection was reset (check out the link above). At this time, I temporarily patched Shotwell so it would keep the camera object alive to maintain it open through the different download phasis.

Today, I can't access my file using **gphoto2 --port=usb --shell** : either I get an error code, or the directory is claimed to be empty, while it's not. The error code is either -1 ("undefined") or -107 ("Directory Not Found"). Sometimes, the number of the directory increases. Example:

```
gphoto2: {/home/test/Git/libgphoto2} /> cd store_00010001/ 

*** Error ***              
PTP Invalid Storage ID
*** Error (-107: 'Directory not found') ***
gphoto2: {/home/test/Git/libgphoto2} /> ls
store_00010001/     
gphoto2: {/home/test/Git/libgphoto2} /> 
dom 17:03:51 libgphoto2 $ LANG=C gphoto2 --port=usb --shell
gphoto2: {/home/test/Git/libgphoto2} /> ls                                      
store_00010002/    
```

The store code is expressed in hexadecimal (I raised it up to 001000a). This incrementation doesn't seem to be the origin of the problem. Sometimes it does, sometimes not, and when I try to guess the next value, I still can't get inside the directory.

```
$ LANG=C gphoto2 --port=usb --shell
gphoto2: {/home/test/Git/libgphoto2} /> ls
store_00010002/
gphoto2: {/home/test/Git/libgphoto2} /> cd store_00010002/
*** Error (-107: 'Directory not found') ***
gphoto2: {/home/test/Git/libgphoto2} /> ls
store_00010002/
gphoto2: {/home/test/Git/libgphoto2} /> cd store_00010002/
*** Error (-107: 'Directory not found') ***
gphoto2: {/home/test/Git/libgphoto2} /> ls
store_00010002/
```

I'm not quite used with the PTP protocol, but I'd willing to contribute/write some code/make some tests it you're OK to drive us a little bit.

msmeissn, 2017-02-08:
there is breakage in the ios 10.2 regarding PTP support I think. 

the traces I looked like I see inconsistent storage 0x0000000 (bad) reported . 

I pretty much think this is a iOS fuckup :(

ObsidianParis, 2017-02-08:
This is what I thought about at first, too. But when I read the jritzerfeld's comment, could it be indirectly due to a lack of proper pairing/trusting from imobiledevice (even thought it's unrelated to gphoto2, I suppose it remains necessary to gain access to the repository, but I'm no iOS expert). Could it also be a problem of conflicting access between these libraries ?

Whatever the cause, do you think we could reasonably work around this ?

ObsidianParis, 2017-03-05:
Hello there,
Any news regarding this issue ?

I've performed some further checks on my side, and I wrote my own little C++ based PTP communicator over libusb-1 to try to communicate with my device and learn more about PTP. I've also found online a precise description of the first version of the protocol (2000), but couldn't find anything about 1.1 (2008) nor 2013.

My program currently get the device informations (GetDeviceInfo), opens a session (OpenSession), retrieve the storage IDs (GetStorageIDs), then closes the session (CloseSession), but don't go any further yet.

Here's what I found:

- Each time one opens a session, then asks for GetStorageIDs, the device first returns **0xfeedface** as a storage ID, then returns a correct ID the next times one asks for it within this particular session;
- When the session is closed, the storage IDs have a lifetime of **3 seconds**: if one reopens a session within this period, one gets the same storage IDs. Otherwise, one get 0xfeedface again and then, the storage ID is **incremented**: 0x00010002 instead of 0x00010001;
- This phenomena seems to be linked to session openings. Opening/Closing the usb link itself doesn't seem to matter;
- The ID number seems to be incremented each time "feedface" shows up. If one gets three times this value, then gets a valid ID, its number would directly have jumped to 0x00010004.

I can't tell if "feedface" is especially linked to the 2011 exploit or if it's just a common magic number used by Apple over time.

Please find attached a tarball that contains 5 log files.
- Two of them opens 10 sessions each containing one ID query. Sessions are respectively spaced with 2 seconds and 5 seconds delays ;
- Three of them perform 10 ID queries within a single session. The queries are spaced with 2 seconds, 5 seconds and 20 seconds delays.

I've masked my device serial number on purpose.
[logs.tar.gz](https://github.com/gphoto/libgphoto2/files/819030/logs.tar.gz)

msmeissn, 2017-03-05:
Hmm.

Mostly we run single session gphoto2, and the previous logfiles never showed 0xfeedface ... 

But it gives me an idea of what a problem might be.

If a PTP session is opened _twice_, once by gvfs of GNOME (and kept open) and then by gphoto2 commandline, we could confuse the device.

Is the iPhone gvfs mount working?

Does gphoto2 -L --wait-event=1s -L   report a StorageInfoChanged event? (for the 0xfeedface -> 0x00010001 -> 0x00010002 transition)?

ObsidianParis, 2017-03-05:
Thanks again for your prompt answer.
The GNOME mount is operational:

```
dom 11:05:00 ~ $ ps aux | grep gv[f]s 
gdm       1606  0.0  0.1 322220  7000 tty1     Sl+  10:18   0:00 /usr/libexec/gvfsd
gdm       1610  0.0  0.1 305436  6864 tty1     Sl+  10:18   0:00 /usr/libexec/gvfsd-fuse /run/user/42/gvfs -f -o big_writes
gdm       1689  0.0  0.2 414312  8984 tty1     Sl+  10:18   0:00 /usr/libexec/gvfs-udisks2-volume-monitor
gdm       1701  0.0  0.1 325600  6832 tty1     Sl+  10:18   0:00 /usr/libexec/gvfs-mtp-volume-monitor
gdm       1705  0.0  0.1 314928  6328 tty1     Sl+  10:18   0:00 /usr/libexec/gvfs-goa-volume-monitor
gdm       1717  0.0  0.2 428368  9748 tty1     Sl+  10:18   0:00 /usr/libexec/gvfs-afc-volume-monitor
gdm       1722  0.0  0.1 334824  7492 tty1     Sl+  10:18   0:00 **/usr/libexec/gvfs-gphoto2-volume-monitor**
dom       2106  0.0  0.1 322324  7100 tty2     Sl+  10:25   0:00 /usr/libexec/gvfsd
dom       2110  0.0  0.1 305436  6860 tty2     Sl+  10:25   0:00 /usr/libexec/gvfsd-fuse /run/user/1000/gvfs -f -o big_writes
dom       2272  0.0  0.2 414312 11128 tty2     Sl+  10:25   0:00 /usr/libexec/gvfs-udisks2-volume-monitor
dom       2283  0.0  0.1 325600  6728 tty2     Sl+  10:25   0:00 /usr/libexec/gvfs-mtp-volume-monitor
dom       2290  0.0  0.1 314928  6424 tty2     Sl+  10:25   0:00 /usr/libexec/gvfs-goa-volume-monitor
dom       2307  0.0  0.2 428360 10112 tty2     Sl+  10:25   0:00 /usr/libexec/gvfs-afc-volume-monitor
dom       2314  0.0  0.1 334828  7004 tty2     Sl+  10:25   0:00 /usr/libexec/gvfs-gphoto2-volume-monitor
dom       2540  0.0  0.3 585476 12604 tty2     Sl+  10:26   0:00 /usr/libexec/gvfsd-afc --spawner :1.6 /org/gtk/gvfs/exec_spaw/0
dom       2542  0.0  0.2 570140 10152 tty2     Sl+  10:26   0:00 /usr/libexec/gvfsd-afc --spawner :1.6 /org/gtk/gvfs/exec_spaw/1
dom       2850  0.0  0.1 252168  7216 tty2     Sl+  10:27   0:00 /usr/libexec/gvfsd-metadata
```

The command you asked does show a transition, written down in the attached log (the first call is in french, sorry about this :) I've added LANG=C to the next ones). Also, I indeed still have ~ 1900 pictures pending in my device…

I tried to stop all gvfs programs before but it didn't help. I also tried to run it « outside a gnome environment », killing usbmuxd and such for the same reasons, but I still need my device to be paired with the computer to have an access to it.

I also have reset my softwares to the default version of my distribution. I now will compile the latest releases. It didn't solve the problem, as explained above, but it could give some more explicit messages.
[log1.txt](https://github.com/gphoto/libgphoto2/files/819240/log1.txt)

jritzerfeld, 2017-03-05:
@ObsidianParis Many thanks for your effort! Besides the 0xfeedface id, the 3 seconds timeout would explain the change of storage ids that I am observing, too.

Regarding the 0xfeedface id: This id seems to correspond to the store_feedface folder I mentionend in  [comment of 27 Jan](https://github.com/gphoto/libgphoto2/issues/132#issuecomment-275791345)
Here, it only appeared if the iPhone was locked. Then you cannot access your photos anyway. Thus, I simply discarded all logs with 0xfeedface because these attempts would have never worked, even with Windows or an old iOS.

The output of the command command @msmeissn gave to us while unlocking my iPhone:
`There is no file in folder '/store_feedface'.`
`UNKNOWN PTP Event 4005, Param1 feedface`
`UNKNOWN PTP Event c001, Param1 00000000`
`UNKNOWN PTP Event 4004, Param1 0001xxxx`
`There is no file in folder '/store_feedface'.`
Unlocking my iPhone changes the folder name according to the id of the latest event but gphoto2 tries to list the old folder.

Surprisingly, using the command on an unlocked iPhone still shows the same events, including 0xfeedface. However, there is still something wrong with the folder name:
`There is no file in folder '/'.`
`[...] PTP Invalid Storage ID [...]`
`UNKNOWN PTP Event 4005, Param1 feedface`
`UNKNOWN PTP Event c001, Param1 00000000`
`UNKNOWN PTP Event 4004, Param1 0001001f`
`There is no file in folder '/'.`
`There is no file in folder '/store_00010001'.`
Again, the gphoto2 tries to list the wrong folder.

So, it really looks line the iPhone camera is always locked and unlocks automatically on request for a few seconds (see above) only if the iPhone is unlocked and the computer explicitly trusted for photo access. And while this permission check happens the camera reports an invalid storage id. Then, everything works fine as long as the timeout is not reached. That is, the first "gphoto2 -L" fails because of the invalid storage id. If you are fast enough and execute "gphoto2 -L" again within the 3 seconds timeout, it will work. And you can repeat this forever as long as you wait less than 3 seconds between these commands.

ObsidianParis, 2017-03-05:
@jritzerfeld Thanks a lot for your appreciation. Regarding store_feedace, the standard states that a persistent storage is represented by a 32-bit ID, having the 16 uppermost bits specifying a physical storage and the 16 lowest a logical storage inside it. Thus the 0x00010001. Nevertheless, any 32-bit number remains valid and if iOS chooses to return a magic number for any reason, we get a directory named that way.

For the rest, I get the same event numbers than you do, but the order sometimes differ. Seems that the unknown event 4004 gives us the last valid ID in date.

I've also tried with an Apple iPad 3 running iOS 9.3.5, and no problem occurs.

@msmeissn Here's another log with the same command results, using last stable versions of both gphoto2 and libgphoto2 (compiled from the Git repository).
[log2.txt](https://github.com/gphoto/libgphoto2/files/819323/log2.txt)

msmeissn, 2017-03-05:
Thanks for these traces. this means there is dynamic storage addition/removal even in 1 running session.

this will confuse gvfs too and not just plain gphoto2. I need to write code to handle that :/

ObsidianParis, 2017-03-05:
I performed once again some more controls according to your last indications. I switched to a WindowMaker session to get rid of GNOME for a while, but Fedora/SystemD deeply intricates all these subsystems in. I killed anyway every gvfs-* process and tried to unmount whatever was lying under /run/user/1000/*.

Seems that all symptoms remains the same: three events incoming while opening a session, always the same. More important: they don't seem to be triggered by a third part subsystem on host side. Instead, some 3-second timer seems to be set by iOS at startup time or when a session is closed, and the events are sent to notify the client that something was updated at this occasion.

I also tried to set the parameters to --wait-event=180s to see if something odd happens after a while, but nothing showed up.

Please check out this last attached log. As always, it works when querying the device within a 3 second grace delay, and fails with the observed events after that.

[log3.txt](https://github.com/gphoto/libgphoto2/files/819604/log3.txt)

We're willing to contribute, too. If we can write a bit of code to back you up, we'd be pleased.

msmeissn, 2017-03-05:
I added some code in current git to help.

basically dynamic addition / removal of storage is now done, but only handled if you call "--wait-event" 
so use some startup work of like:

gphoto2 --wait-event=2s -L    

depending on the time needed. Replace -L by -P for downloading.

(event should be checked in other functions too, but it is a bit trickier)

jritzerfeld, 2017-03-05:
@msmeissn Many thanks for your help! I have "cherry-picked" your commit into the 2.5.12 source and now "gphoto2 --wait-event=2s -L" sees the change of the storage id and, thus, there is no "invalid storage id" error anymore. Unfortunately, there are no files in the folder of the correct storage id. Hmm...

ObsidianParis, 2017-03-06:
Thanks for working on it at night ! :-)
Unfortunately, it doesn't work so well for me :-( Nothing really new in fact. I just compiled the latest commit ( 12b7c2223432599671a47f1c735c55f1826589be ), and here's what I get:

dom 00:46:30 bin $ **./gphoto2 --version**
```
gphoto2 2.5.11

Copyright (c) 2000-2016 Lutz Mueller and others

gphoto2 comes with NO WARRANTY, to the extent permitted by law. You may
redistribute copies of gphoto2 under the terms of the GNU General Public
License. For more information about these matters, see the files named COPYING.

This version of gphoto2 is using the following software versions and options:
gphoto2         **2.5.11**         gcc, popt(m), exif, no cdk, no aa, jpeg, readline
libgphoto2      **2.5.12.1**       all camlibs, gcc, ltdl, EXIF
libgphoto2_port **0.12.0**         gcc, ltdl, USB, serial without locking
```

dom 00:46:31 bin $ **./gphoto2 --wait-event=15s -L**
```
Waiting for 15 seconds for events from camera. Press Ctrl-C to abort.          
UNKNOWN PTP Event 4005, Param1 feedface
UNKNOWN PTP Event c001, Param1 00000000
UNKNOWN PTP Event 4004, Param1 00010009
There is no file in folder '/'.
There is no file in folder '/store_00010009'.
```

dom 00:47:05 bin $ **./gphoto2 --wait-event=15s -L**
```
Waiting for 15 seconds for events from camera. Press Ctrl-C to abort.          
UNKNOWN PTP Event 4005, Param1 feedface
UNKNOWN PTP Event c001, Param1 00000000
UNKNOWN PTP Event 4004, Param1 0001000a
There is no file in folder '/'.
There is no file in folder '/store_0001000a'.
```

dom 00:47:42 bin $ **./gphoto2 --wait-event=3s -L**
```
Waiting for 3 seconds for events from camera. Press Ctrl-C to abort.           
UNKNOWN PTP Event 4005, Param1 feedface
UNKNOWN PTP Event c001, Param1 00000000
UNKNOWN PTP Event 4004, Param1 0001000b
There is no file in folder '/'.
There is no file in folder '/store_0001000b'.
```

dom 00:47:57 bin $ **./gphoto2 --wait-event=2s -L**
```
Waiting for 2 seconds for events from camera. Press Ctrl-C to abort.           
UNKNOWN PTP Event 4005, Param1 feedface
There is no file in folder '/'.

*** Error ***              
PTP Invalid Storage ID
*** Error (-1: 'Unspecified error') ***       

For debugging messages, please use the --debug option.
Debugging messages may help finding a solution to your problem.
If you intend to send any error or debug messages to the gphoto
developer mailing list <gphoto-devel@lists.sourceforge.net>, please run
gphoto2 as follows:

    env LANG=C gphoto2 --debug --debug-logfile=my-logfile.txt --wait-event=2s -L

Please make sure there is sufficient quoting around the arguments.
```

dom 00:48:07 bin $ **./gphoto2 --wait-event=2s -L**
```
Waiting for 2 seconds for events from camera. Press Ctrl-C to abort.           
UNKNOWN PTP Event 4004, Param1 0001000c
UNKNOWN PTP Event 4005, Param1 feedface
There is no file in folder '/'.

*** Error ***              
PTP Invalid Storage ID
*** Error (-1: 'Unspecified error') ***       

For debugging messages, please use the --debug option.
Debugging messages may help finding a solution to your problem.
If you intend to send any error or debug messages to the gphoto
developer mailing list <gphoto-devel@lists.sourceforge.net>, please run
gphoto2 as follows:

    env LANG=C gphoto2 --debug --debug-logfile=my-logfile.txt --wait-event=2s -L

Please make sure there is sufficient quoting around the arguments.
```

I also tried with debug files:

dom 01:01:20 bin $ env LANG=C gphoto2 --debug --debug-logfile=my-logfile.txt --wait-event=2s -L
```
Waiting for 2 seconds for events from camera. Press Ctrl-C to abort.           
UNKNOWN PTP Event 4005, Param1 feedface
There is no file in folder '/'.

*** Error ***              
PTP Invalid Storage ID
*** Error (-1: 'Unspecified error') ***       
```

dom 01:01:39 bin $ env LANG=C gphoto2 --debug --debug-logfile=my-logfile2.txt --wait-event=10s -L
```
Waiting for 10 seconds for events from camera. Press Ctrl-C to abort.          
UNKNOWN PTP Event 4005, Param1 feedface
UNKNOWN PTP Event c001, Param1 00000000
UNKNOWN PTP Event 4004, Param1 00010010
There is no file in folder '/'.
There is no file in folder '/store_00010010'.
```

Seems that the last two events are now coming a little later than before, but I can't tell if it's due to your queue implementation or not. Here's the debug logs:

[my-logfile.txt](https://github.com/gphoto/libgphoto2/files/819904/my-logfile.txt)
[my-logfile2.txt](https://github.com/gphoto/libgphoto2/files/819903/my-logfile2.txt)

ObsidianParis, 2017-03-08:
Good evening everybody,

I finally wrote a stinky, terrible hack that applies on release **2.5.12** ( 1719ba76c7ad5b934f113fcd6e433999fe99b92e ). Here's the patch:

[ios102.diff.txt](https://github.com/gphoto/libgphoto2/files/825681/ios102.diff.txt)

This modifies the function **camera_init()** in « camlibs/ptp2/library.c » and looks for the Apple USB vendor id and waits for early events. There's a timeout of 3 seconds, which is reset each time an event is received. It also retries three times to retrieve storage ids if ever we get a feedface ID.

This causes delays that can raise up to 10 seconds each time the camera is initialized by upper dependencies, which includes object creation and destruction from higher level languages (such as vala, for Shotwell) and makes the applications seem to freeze for a while.

Also, this absolutely doesn't care about the nature of the events itself. It just absorbs early events we have observed and that should not occur at this time.

We definitely don't want to merge this as is, of course, but this prevents the connection from failing, which is necessary for applications that are not compiled yet to handle this issue, and this — finally — reenabled myself to import my thousands pictures from Shotwell.

Just a few more remarks:

- the storage ID still evolve from a session to the next one, and I was afraid this could prevent applications from recognizing the files inside as being the same ones, but Shotwell seems to correctly handle this;
- I suppose Apple cannot afford its devices to present such a bug on hosts that wouldn't have been updated. Maybe there's a way to "register" to the device to prevent this;
- iOS 10.3 is going to be released shortly. There's a chance for this bug to disappear by itself, since the previous one (described at https://github.com/gphoto/libgphoto2/issues/132#issuecomment-278378354 ) has finaly gone. But it's very unlikely, especially when legitimate bugs (such as battery lifetime problems) are not fixed among successive releases;

msmeissn, 2017-03-09:
question is ... does your list the files on that storage?

i also hope ios 10.3 fixes it :/

ObsidianParis, 2017-03-09:
Good evening,

@msmeissn 
Yes it does:
```
$ LANG=C gphoto2 -L
There is no file in folder '/'.                                                
There is no file in folder '/store_00010002'.
There is no file in folder '/store_00010002/DCIM'.
There are 2 files in folder '/store_00010002/DCIM/100APPLE'.
#1     IMG_0603.JPG               rd  1284 KB 2448x3264 image/jpeg
#2     IMG_0604.JPG               rd  2058 KB 2448x3264 image/jpeg
There are 106 files in folder '/store_00010002/DCIM/101APPLE'.
#3     IMG_1202.JPG               rd  1350 KB 2448x3264 image/jpeg
#4     IMG_1228.JPG               rd  1355 KB 2448x3264 image/jpeg
#5     IMG_1290.JPG               rd  2762 KB 2448x3264 image/jpeg
[...]
#106   IMG_1904.JPG               rd  2083 KB 3264x2448 image/jpeg
#107   IMG_1905.JPG               rd  1919 KB 3264x2448 image/jpeg
#108   IMG_1906.JPG               rd  1421 KB 3264x2448 image/jpeg
```
… and it also re-enabled me to use the --shell and delete my files from there.

But it's no surprise because all this hack really does is waiting for the early events to occur at camera initialization time, before starting real processing. It simply assumes that once the C001/4004 Add/4005 Remove events have come, the new storage ID remains stable until next initialization (or next session at least). This is what we have observed so far.

My main concern here is that I don't exactly know yet how libgphoto2 handles events, and we'll probably need to spawn a thread to process them if we are to expect them to come at any time.

As regards a possible fix in next release of iOS, I remain pessimist, except if this change breaks compatibility with too many strategic commercial softwares. However, I'm almost sure there's a way to "register" in a way or another and pretend we're an older version of iTunes or such.

msmeissn, 2017-03-10:
I added a more streamlined fix to current GIT. It should be doing the same basically.  Can youy try it?

ObsidianParis, 2017-03-10:
Hello,
Here's what I get:

```
dom 12:00:09 ~ $ LANG=C gphoto2 -L
There is no file in folder '/'.                                                
There is no file in folder '/store_00010002'.
dom 12:00:10 ~ $ LANG=C gphoto2 -L
There is no file in folder '/'.                                                
There is no file in folder '/store_00010002'.
There is no file in folder '/store_00010002/DCIM'.
There are 2 files in folder '/store_00010002/DCIM/100APPLE'.
#1     IMG_0603.JPG               rd  1284 KB 2448x3264 image/jpeg
#2     IMG_0604.JPG               rd  2058 KB 2448x3264 image/jpeg
There are 106 files in folder '/store_00010002/DCIM/101APPLE'.
#3     IMG_1202.JPG               rd  1350 KB 2448x3264 image/jpeg
#4     IMG_1228.JPG               rd  1355 KB 2448x3264 image/jpeg
#5     IMG_1290.JPG               rd  2762 KB 2448x3264 image/jpeg
[...]
#106   IMG_1904.JPG               rd  2083 KB 3264x2448 image/jpeg
#107   IMG_1905.JPG               rd  1919 KB 3264x2448 image/jpeg
#108   IMG_1906.JPG               rd  1421 KB 3264x2448 image/jpeg
dom 12:00:31 ~ $ 
```

At first try, the program always terminates after the second "no file in folder …" as stated above. If immediately relaunched, it takes each time **20 seconds** to list the files, but it does. 

And as usual, we redo from start if relaunched out of the 3-second grace delay, and suffer twenty seconds at each try within the delay, no matter the times we've already launched it before.

Thanks.

msmeissn, 2017-03-10:
can you post a debug trace of both? (the emoty one and the 20 second delay one)?

ObsidianParis, 2017-03-10:
Sure (I could have thought about it by myself, sorry about this):

[log-20170310-1.txt](https://github.com/gphoto/libgphoto2/files/833940/log-20170310-1.txt)
[log-20170310-2.txt](https://github.com/gphoto/libgphoto2/files/833941/log-20170310-2.txt)

msmeissn, 2017-03-10:
ok, i had some thinkos in my patch that caused both . I think ...  can you check current git again?

ObsidianParis, 2017-03-10:
Much better. Thanks!
The behaviour is now the same excepted the file are now listed almost instantly, no longer after twenty seconds.

The rest remains as is: failing at first try, success each time we retry within the three next seconds, id increment otherwise.

ObsidianParis, 2017-03-10:
[log-20170310-3.txt](https://github.com/gphoto/libgphoto2/files/834123/log-20170310-3.txt)
[log-20170310-4.txt](https://github.com/gphoto/libgphoto2/files/834124/log-20170310-4.txt)

msmeissn, 2017-03-10:
ok, one ore try? ;)

ObsidianParis, 2017-03-10:
That's perfect !

[log-20170310-5.txt](https://github.com/gphoto/libgphoto2/files/834483/log-20170310-5.txt)
[log-20170310-6.txt](https://github.com/gphoto/libgphoto2/files/834482/log-20170310-6.txt)

Roughly half a second at first try or out of grace delay, immediate otherwise.

The storage ID increases the same way than above (due to iOS) but it doesn't seem to be a problem anymore. Early tests with Shotwell seem to be fine, too: thumbnails are correctly and quickly retrieved. I haven't tried to import new photos yet nor deleting them. If this causes problem for any reason, I'll post here again.

I'll also make a report when iOS 10.3 has arrived. I'll check both this last version and the latest stable release to see if this bug remains or not. Let's hope the new release won't introduce _another_ bug. :-\

Thanks you for leaning on this problem. I hope this won't cause any regression to other cameras. Keep on asking for another tests if needed.

Cheers.

jritzerfeld, 2017-03-12:
Thank you both for your excellent work!
For openSUSE, I created a patched libgphoto2-2.5.12 package in home:jritzerfeld:branches:graphics for two other Leap 42.2 users who complained in the openSUSE mailing lists.

Pointedstick, 2017-03-21:
Thanks Jan! Any idea when this will be available for Tumbleweed users?

jritzerfeld, 2017-03-21:
Sorry, Marcus should know. However, you could test [home:jritzerfeld:branches:graphics openSUSE_Tumblewee](http://download.opensuse.org/repositories/home:/jritzerfeld:/branches:/graphics/openSUSE_Tumbleweed/), too.

msmeissn, 2017-03-22:
I took a current git snapshot and submitted it to Factory/Tumbleweed. will appear there in the next days.

ObsidianParis, 2017-03-30:
Just a word to say that I'm STILL waiting for iOS 10.3 to be available to my iPhone 5, so I can write a review. :-( This is a known issue from Apple (another one). Some users have their device still said to be up to date with 10.2.1, even when trying to explicitly fetch it through the Settings menu. I've read somewhere that Apple recommends to get it from iTunes on a regular computer, but mine is running Linux. So…

jritzerfeld, 2017-03-30:
I installed iOS 10.3 on my iPhone and I can still access, view and copy my photos with the fixed version from the openSUSE graphics repo!

ObsidianParis, 2017-04-06:
Finally, the **10.3.1** release has reached my device, which means that I directly jumped from 10.2.1 to 10.3.1, skipping 10.3. Looks like my device belongs to the last 32 bits CPU generation (Apple A6) and this soon will start to cause problems. In the mean time, I also switched from Fedora 22 to 25 (with a lot of troubles).

Be that as it may, it works !

To be more precise, the bug still exists in this 10.3.1 version, and is still correctly handled by the last versions of the library. In other word, it was not a waste of time to write something for it.

However, I sometime experienced some conflicts with either _gvfsd-gphoto2_ or _gvfs-gphoto2-volume-monitor_, which was hogging the PTP usb interface. I'll do some more tests to see if it's persistent problem or simply an intermittent bug due to the last upgrades.

Thanks for everything ! This would really have been a pain to wait half a year to dump pictures again. :-)

ObsidianParis, 2017-04-22:
Hello Marcus, hello everyone.

Just a quick question : I'd like to have this last release pushed in Fedora 25 (the distrib I'm currently using). libgphoto2 is packaged by it, last version currently being 2.5.12, built on 12th feb, 2017. Is there already any established workflow/protocol or are you disconnected from the Fedora's maintenance team ? I ask the question because I was about to send them a mail but I couldn't find any document describing the appropriate procedure...

Also, I can see in the Git repository that you sometimes use annotated tag, sometimes lightweight ones. Is there any special policy regarding this ?

Thanks once again for your work.

msmeissn, 2017-04-22:
I currently have no contacts to the Fedora libgphoto2 packager. But they will probably read bugreports. not sure.

huh. I am not very familar with git tags, I might be doing it wrong. Can you tell me the differing ones?

ObsidianParis, 2017-04-22:
Sure,

As regards Fedora, I now think I'll send a mail to the last reported maintainers according to the package's changelog. That's fine that way. I think this particular version would be appreciated because Fedora relies on Gnome by default, which detects iPhones, opens them in Nautilus, and automatically suggests Shotwell to get pictures, which relies on libgphoto2. This means that this last Apple's prank breaks the whole user experience under this kind of distribution…

As regards git tags, there's only four "lightweight" tags (out of 37). They are : libgphoto2-2_5_9-release, libgphoto2-2_5_10-release, libgphoto2-2_5_12-release and libgphoto2-2_5_13-release.

No problem neither about this. I simply noticed them because "tig" (git pager) highlights annotated tags and I was wondering if it was related to decisions of spreading or not a given release.

Thanks for taking the time to answer.

JR1994, 2017-04-28:
Could I use the fixed version from OpenSUSE on Debian?

msmeissn, 2017-04-28:
you can build it yourself on debian, or ask the debian maintainers to update ? 

there is also the gphoto2-updater.sh script you can use https://github.com/gonzalo/gphoto2-updater

parasense, 2017-04-29:
May I ask when we might see the next release of libgphoto2? So long as the fix provided here will land in the next official release, I'm willing to carry the patch in Fedora, and update the distro package.

ObsidianParis, 2017-04-30:
I should let Marcus answer here, but there's a recent release tagged **2.5.13** ( a632f2f56462ebe3afa599657147405e7adce9bc ) from 8th of April, 2017. This release covers the bug described hereby.

The current Fedora 25 package is told to provide version 2.5.12 and to be built the 12th of February.

Thanks for caring.

parasense, 2017-04-30:
Here is a build of 2.5.13 for Fedora 25:
http://koji.fedoraproject.org/koji/buildinfo?buildID=885008

and likewise the upcoming 26:
http://koji.fedoraproject.org/koji/buildinfo?buildID=884902

I would appreciate testing, and thanks in advance.

msmeissn, 2017-04-30:
yes, see above. please report problems if any

ObsidianParis, 2017-05-01:
Hi everyone,

Looks fine at first try : with **iOS 10.3.1** on a **iPhone 5**, using **Shotwell**, I can't get no picture with the current 2.5.12 release, while everything works as expected with the last 2.5.13 packages. Thumbnails are retrieved, and I'm then able to import the pictures themselves and remove them.

I sometime get a message "Unable to communicate with the USB device : error -53" at first try but I can't tell if it's related or not to this. When it happens, I'm able to get the pictures anyway by selecting my phone a second time in the left panel.

Also, calling **gphoto2 -L --port=usb:** gives me the list of my pictures at every time. As always, the storage ID remains the same when it's called within the 3-second window, and increments otherwise. It also works with previous versions (**iOS 9.3.5** on an iPad 3). Both Shotwell and gphoto2 works as expected, with no storage ID incrementation.

I unfortunately don't have any other regular PTP camera around to check for eventual regressions.

Some minor bugs seems to occur, but used to do before too : plug or unplug an Apple device seem to cause gnome-shell to freeze during a bunch of seconds and sometimes, especially when the device was not priorly paired, something hog the USB port and prevent applications from properly communicating with it.

My guess is that we should keep an eye on this bug entry, because it's likely to change again in the future. The previous bug described in my [comment above](https://github.com/gphoto/libgphoto2/issues/132#issuecomment-278378354) (randomly chosen subdirectory names at each call) has eventually disappeared, then we had this one two years later. If the observed behaviour here change slightly, it could break the library again.

Thanks to everyone for your work.

msmeissn, 2017-05-01:
the "comment above" was this problem fixed here.

so, lets hope this solves this

closed 2017-05-01 by msmeissn

ObsidianParis, 2017-05-01:
Hem, not quite: I was talking about the bugzilla entry I was refering to and that links to the previous problem that occurred two years ago, that is a changing subdirectory name UNDER the storage folder. This entry is different (spurious events, then deletion/reinsertion of the storage).

But for now, it works. Let's hope it will last.

JR1994, 2017-05-02:
I ran the gphoto2-updater.sh and then connected my phone. Debian is not mounting the iPhone as a USB device even though I see it recognized in dmesg as a USB device.

austinkelleher, 2017-07-20:
Running the the`gphoto2-updater.sh` (https://github.com/gonzalo/gphoto2-updater) fixed the issue for me too! Thanks @msmeissn!

arekm, 2018-03-28:
Not working for me. libgphoto2 built today from git, gphoto 2.5.15:

$  env LANG=C gphoto2 --debug --debug-logfile=my-logfile.txt --wait-event=2s -L
Waiting for 2 seconds for events from camera. Press Ctrl-C to abort.           
There is no file in folder '/'.

*** Error ***              
PTP Invalid Storage ID
*** Error (-1: 'Unspecified error') ***       

[my-logfile.txt](https://github.com/gphoto/libgphoto2/files/1855427/my-logfile.txt)

msmeissn, 2018-03-28:
@arekm  it looks like you did not unlock the phone during the 9 seconds of startup.

arekm, 2018-03-28:
Hm, retested. Connected phone, unlocked, started gphoto, got question to allow access on my phone, allowed and then gphoto came with the same error.

Started gphoto again, same error, log attached.
[my-logfile.txt](https://github.com/gphoto/libgphoto2/files/1855987/my-logfile.txt)

msmeissn, 2018-03-28:
hmm, it shows the removal of the invalid storageid "0xfeedface", but no addition of the actual storage. :/

ObsidianParis, 2018-04-08:
Hey everybody,
I just received the notification. Is this issue fixed or is it a request for working on another volontary bug ? I haven't downloaded my pics for a while a my 32-bits iPhone 5 no longer receives new iOS updates anyway.

Cheers.