gphoto/libgphoto2 #133: Fuji X-T2, USB ID 04cb:02cd

jbreiden, 2017-01-23:
This is the Fuji X-T2 camera. I'll look into USB communication details next.

$ lsusb | grep Fuji
Bus 004 Device 004: ID 04cb:02cd Fuji Photo Film Co., Ltd 

```
--- a/camlibs/ptp2/library.c
+++ b/camlibs/ptp2/library.c
@@ -1978,6 +1978,7 @@ static struct {
        {"Fuji:Fujifilm X-E2",                  0x04cb, 0x02b5, 0},
        /* Vladimir K <enewsletters@inbox.ru> */
        {"Fuji:Fujifilm X-T1",                  0x04cb, 0x02bf, 0},
+        {"Fuji:Fujifilm X-T2",                 0x04cb, 0x02cd, 0},
        /* https://github.com/gphoto/libgphoto2/issues/32 */
        {"Fuji:Fujifilm X-T10",                 0x04cb, 0x02c8, 0},
 
```

jbreiden, 2017-01-23:
```
0.066445 print_debug_deviceinfo      (2): Device info:
0.066448 print_debug_deviceinfo      (2): Manufacturer: FUJIFILM
0.066451 print_debug_deviceinfo      (2):   Model: X-T2
0.066453 print_debug_deviceinfo      (2):   device version: 1.00
0.066455 print_debug_deviceinfo      (2):   serial number: '5935333130321611296B3020112361'
0.066457 print_debug_deviceinfo      (2): Vendor extension ID: 0x0000000e
0.066459 print_debug_deviceinfo      (2): Vendor extension version: 100
0.066462 print_debug_deviceinfo      (2): Vendor extension description: fujifilm.co.jp: 1.0; 
0.066464 print_debug_deviceinfo      (2): Functional Mode: 0x0000
0.066466 print_debug_deviceinfo      (2): PTP Standard Version: 100
0.066468 print_debug_deviceinfo      (2): Supported operations:
0.066473 print_debug_deviceinfo      (2):   0x1001 (Get device info)
0.066476 print_debug_deviceinfo      (2):   0x1002 (Open session)
0.066479 print_debug_deviceinfo      (2):   0x1003 (Close session)
0.066482 print_debug_deviceinfo      (2):   0x1004 (Get storage IDs)
0.066484 print_debug_deviceinfo      (2):   0x1005 (Get storage info)
0.066487 print_debug_deviceinfo      (2):   0x1006 (Get number of objects)
0.066489 print_debug_deviceinfo      (2):   0x1007 (Get object handles)
0.066492 print_debug_deviceinfo      (2):   0x1008 (Get object info)
0.066494 print_debug_deviceinfo      (2):   0x1009 (Get object)
0.066497 print_debug_deviceinfo      (2):   0x100a (Get thumbnail)
0.066500 print_debug_deviceinfo      (2):   0x100b (Delete object)
0.066502 print_debug_deviceinfo      (2):   0x100c (Send object info)
0.066505 print_debug_deviceinfo      (2):   0x100d (Send object)
0.066507 print_debug_deviceinfo      (2):   0x100f (Format storage)
0.066510 print_debug_deviceinfo      (2):   0x1014 (Get device property description)
0.066513 print_debug_deviceinfo      (2):   0x1015 (Get device property value)
0.066516 print_debug_deviceinfo      (2):   0x1016 (Set device property value)
0.066518 print_debug_deviceinfo      (2):   0x101b (Get partial object)
0.066521 print_debug_deviceinfo      (2):   0x900c (Unknown VendorExtensionID)
0.066524 print_debug_deviceinfo      (2):   0x900d (Unknown VendorExtensionID)
0.066526 print_debug_deviceinfo      (2):   0x901d (Unknown VendorExtensionID)
0.066529 print_debug_deviceinfo      (2):   0x9801 (Unknown VendorExtensionID)
0.066532 print_debug_deviceinfo      (2):   0x9802 (Unknown VendorExtensionID)
0.066534 print_debug_deviceinfo      (2):   0x9803 (Unknown VendorExtensionID)
0.066537 print_debug_deviceinfo      (2):   0x9805 (Unknown VendorExtensionID)
0.066539 print_debug_deviceinfo      (2): Events Supported:
0.066541 print_debug_deviceinfo      (2):   0x4002
0.066542 print_debug_deviceinfo      (2):   0x4003
0.066544 print_debug_deviceinfo      (2):   0x4004
0.066546 print_debug_deviceinfo      (2):   0x4005
0.066547 print_debug_deviceinfo      (2):   0x4006
0.066549 print_debug_deviceinfo      (2):   0x4008
0.066551 print_debug_deviceinfo      (2):   0x4009
0.066552 print_debug_deviceinfo      (2): Device Properties Supported:
0.066554 print_debug_deviceinfo      (2):   0x5001
0.066556 print_debug_deviceinfo      (2):   0xd303
0.066558 print_debug_deviceinfo      (2):   0xd406
0.066559 print_debug_deviceinfo      (2):   0xd407
```

msmeissn, 2017-01-24:
So far there is no indication that it has capture opcodes.
The only unknown opcodes are 900c, 900d, 901d
(The 0x98xx opcodes are generic MTP commands.)
There is no event for capturecomplete.
The 0xd303 and 0xd4xx properties are the standard MTP properties.

While capture might be hiding behind the 0x90xx codes or be hidden from the deviceinfo altogether, this seems unlikely.

jbreiden, 2017-01-24:
Capture opcodes were added in a firmware revision released Nov 2016. I will updating the list, today as soon as I flash the firmware.

jbreiden, 2017-01-24:
To my surprise, the newer firmware (which supports tethering) has identical opcodes. I'll will shortly have my hands on a Windows computer and will endeavor to monitor USB communication during tethering. If there is a preferred mechanism for USB sniffing on Windows, please let me know.

```
0.034602 print_debug_deviceinfo      (2): Device info:
0.034605 print_debug_deviceinfo      (2): Manufacturer: FUJIFILM
0.034607 print_debug_deviceinfo      (2):   Model: X-T2
0.034609 print_debug_deviceinfo      (2):   device version: 1.10
0.034611 print_debug_deviceinfo      (2):   serial number: '5935333130321611296B3020112361'
0.034613 print_debug_deviceinfo      (2): Vendor extension ID: 0x0000000e
0.034614 print_debug_deviceinfo      (2): Vendor extension version: 100
0.034617 print_debug_deviceinfo      (2): Vendor extension description: fujifilm.co.jp: 1.0; 
0.034619 print_debug_deviceinfo      (2): Functional Mode: 0x0000
0.034621 print_debug_deviceinfo      (2): PTP Standard Version: 100
0.034622 print_debug_deviceinfo      (2): Supported operations:
0.034627 print_debug_deviceinfo      (2):   0x1001 (Get device info)
0.034629 print_debug_deviceinfo      (2):   0x1002 (Open session)
0.034631 print_debug_deviceinfo      (2):   0x1003 (Close session)
0.034634 print_debug_deviceinfo      (2):   0x1004 (Get storage IDs)
0.034636 print_debug_deviceinfo      (2):   0x1005 (Get storage info)
0.034638 print_debug_deviceinfo      (2):   0x1006 (Get number of objects)
0.034640 print_debug_deviceinfo      (2):   0x1007 (Get object handles)
0.034643 print_debug_deviceinfo      (2):   0x1008 (Get object info)
0.034645 print_debug_deviceinfo      (2):   0x1009 (Get object)
0.034647 print_debug_deviceinfo      (2):   0x100a (Get thumbnail)
0.034649 print_debug_deviceinfo      (2):   0x100b (Delete object)
0.034651 print_debug_deviceinfo      (2):   0x100c (Send object info)
0.034654 print_debug_deviceinfo      (2):   0x100d (Send object)
0.034656 print_debug_deviceinfo      (2):   0x100f (Format storage)
0.034658 print_debug_deviceinfo      (2):   0x1014 (Get device property description)
0.034661 print_debug_deviceinfo      (2):   0x1015 (Get device property value)
0.034663 print_debug_deviceinfo      (2):   0x1016 (Set device property value)
0.034665 print_debug_deviceinfo      (2):   0x101b (Get partial object)
0.034668 print_debug_deviceinfo      (2):   0x900c (Unknown VendorExtensionID)
0.034670 print_debug_deviceinfo      (2):   0x900d (Unknown VendorExtensionID)
0.034672 print_debug_deviceinfo      (2):   0x901d (Unknown VendorExtensionID)
0.034674 print_debug_deviceinfo      (2):   0x9801 (Unknown VendorExtensionID)
0.034677 print_debug_deviceinfo      (2):   0x9802 (Unknown VendorExtensionID)
0.034679 print_debug_deviceinfo      (2):   0x9803 (Unknown VendorExtensionID)
0.034681 print_debug_deviceinfo      (2):   0x9805 (Unknown VendorExtensionID)
0.034683 print_debug_deviceinfo      (2): Events Supported:
0.034684 print_debug_deviceinfo      (2):   0x4002
0.034686 print_debug_deviceinfo      (2):   0x4003
0.034687 print_debug_deviceinfo      (2):   0x4004
0.034689 print_debug_deviceinfo      (2):   0x4005
0.034690 print_debug_deviceinfo      (2):   0x4006
0.034692 print_debug_deviceinfo      (2):   0x4008
0.034693 print_debug_deviceinfo      (2):   0x4009
0.034695 print_debug_deviceinfo      (2): Device Properties Supported:
0.034696 print_debug_deviceinfo      (2):   0x5001
0.034698 print_debug_deviceinfo      (2):   0xd303
0.034699 print_debug_deviceinfo      (2):   0xd406
0.034701 print_debug_deviceinfo      (2):   0xd407
```

msmeissn, 2017-01-24:
In in earlier times used SnoopyPro. Not sure if it still lives..

Can you try

gphoto2 --capture-tethered

and press shutter button and see if something happens?

jbreiden, 2017-01-24:
Nothing.

```
31.309643 camera_wait_for_event       (2): no events received.
31.309653 camera_wait_for_event       (2): waiting for events timeout 1000 ms
31.309661 gp_port_get_timeout         (2): Current port timeout is 20000 milliseconds.
31.309667 gp_port_set_timeout         (2): Setting port timeout to 150 milliseconds.
31.309674 gp_port_check_int           (3): Reading 24 = 0x18 bytes from interrupt endpoint...
31.459909 gp_port_check_int           (3): Reading 24 = 0x18 bytes from interrupt endpoint...
31.610150 gp_port_set_timeout         (2): Setting port timeout to 20000 milliseconds.
31.610200 ptp_usb_event [usb.c:530]   (0): Reading PTP event failed: Timeout reading from or writing to the port (-10)
```

jbreiden, 2017-01-24:
Now with the camera in the correct mode (Connection Setting > PC Shoot Mode > USB Auto) we get a lot more.

```
0.035849 print_debug_deviceinfo      (2): Device info:
0.035852 print_debug_deviceinfo      (2): Manufacturer: FUJIFILM
0.035854 print_debug_deviceinfo      (2):   Model: X-T2
0.035855 print_debug_deviceinfo      (2):   device version: 1.10
0.035857 print_debug_deviceinfo      (2):   serial number: '5935333130321611296B3020112361'
0.035859 print_debug_deviceinfo      (2): Vendor extension ID: 0x0000000e
0.035861 print_debug_deviceinfo      (2): Vendor extension version: 100
0.035864 print_debug_deviceinfo      (2): Vendor extension description: fujifilm.co.jp: 1.0; 
0.035866 print_debug_deviceinfo      (2): Functional Mode: 0x0000
0.035868 print_debug_deviceinfo      (2): PTP Standard Version: 100
0.035869 print_debug_deviceinfo      (2): Supported operations:
0.035873 print_debug_deviceinfo      (2):   0x1001 (Get device info)
0.035875 print_debug_deviceinfo      (2):   0x1002 (Open session)
0.035878 print_debug_deviceinfo      (2):   0x1003 (Close session)
0.035880 print_debug_deviceinfo      (2):   0x1004 (Get storage IDs)
0.035882 print_debug_deviceinfo      (2):   0x1005 (Get storage info)
0.035885 print_debug_deviceinfo      (2):   0x1006 (Get number of objects)
0.035887 print_debug_deviceinfo      (2):   0x1007 (Get object handles)
0.035889 print_debug_deviceinfo      (2):   0x1008 (Get object info)
0.035892 print_debug_deviceinfo      (2):   0x1009 (Get object)
0.035894 print_debug_deviceinfo      (2):   0x100a (Get thumbnail)
0.035896 print_debug_deviceinfo      (2):   0x100b (Delete object)
0.035898 print_debug_deviceinfo      (2):   0x100e (Initiate capture)
0.035900 print_debug_deviceinfo      (2):   0x100f (Format storage)
0.035903 print_debug_deviceinfo      (2):   0x1014 (Get device property description)
0.035905 print_debug_deviceinfo      (2):   0x1015 (Get device property value)
0.035908 print_debug_deviceinfo      (2):   0x1016 (Set device property value)
0.035910 print_debug_deviceinfo      (2):   0x1017 (Reset device property value)
0.035912 print_debug_deviceinfo      (2):   0x1018 (Terminate open capture)
0.035915 print_debug_deviceinfo      (2):   0x101b (Get partial object)
0.035917 print_debug_deviceinfo      (2):   0x101c (Initiate open capture)
0.035919 print_debug_deviceinfo      (2):   0x900c (Unknown VendorExtensionID)
0.035922 print_debug_deviceinfo      (2):   0x900d (Unknown VendorExtensionID)
0.035924 print_debug_deviceinfo      (2):   0x901d (Unknown VendorExtensionID)
0.035926 print_debug_deviceinfo      (2):   0x9801 (Unknown VendorExtensionID)
0.035928 print_debug_deviceinfo      (2):   0x9802 (Unknown VendorExtensionID)
0.035931 print_debug_deviceinfo      (2):   0x9803 (Unknown VendorExtensionID)
0.035933 print_debug_deviceinfo      (2):   0x9805 (Unknown VendorExtensionID)
0.035935 print_debug_deviceinfo      (2): Events Supported:
0.035937 print_debug_deviceinfo      (2):   0x4002
0.035938 print_debug_deviceinfo      (2):   0x4003
0.035940 print_debug_deviceinfo      (2):   0x4004
0.035941 print_debug_deviceinfo      (2):   0x4005
0.035943 print_debug_deviceinfo      (2):   0x400d
0.035944 print_debug_deviceinfo      (2):   0xc006
0.035946 print_debug_deviceinfo      (2): Device Properties Supported:
0.035947 print_debug_deviceinfo      (2):   0x5001
0.035949 print_debug_deviceinfo      (2):   0x5003
0.035950 print_debug_deviceinfo      (2):   0x5005
0.035952 print_debug_deviceinfo      (2):   0x500a
0.035953 print_debug_deviceinfo      (2):   0x500b
0.035955 print_debug_deviceinfo      (2):   0x500c
0.035956 print_debug_deviceinfo      (2):   0x500e
0.035958 print_debug_deviceinfo      (2):   0x500f
0.035959 print_debug_deviceinfo      (2):   0x5011
0.035961 print_debug_deviceinfo      (2):   0x5012
0.035962 print_debug_deviceinfo      (2):   0x5015
0.035964 print_debug_deviceinfo      (2):   0x5018
0.035965 print_debug_deviceinfo      (2):   0x5019
0.035967 print_debug_deviceinfo      (2):   0x501c
0.035968 print_debug_deviceinfo      (2):   0xd001
0.035970 print_debug_deviceinfo      (2):   0xd002
0.035971 print_debug_deviceinfo      (2):   0xd003
0.035973 print_debug_deviceinfo      (2):   0xd004
0.035974 print_debug_deviceinfo      (2):   0xd005
0.035976 print_debug_deviceinfo      (2):   0xd007
0.035977 print_debug_deviceinfo      (2):   0xd008
[... 123 lines cut ...]
0.036163 print_debug_deviceinfo      (2):   0xd206
0.036165 print_debug_deviceinfo      (2):   0xd207
0.036166 print_debug_deviceinfo      (2):   0xd208
0.036168 print_debug_deviceinfo      (2):   0xd209
0.036169 print_debug_deviceinfo      (2):   0xd20a
0.036171 print_debug_deviceinfo      (2):   0xd20b
0.036172 print_debug_deviceinfo      (2):   0xd20c
0.036174 print_debug_deviceinfo      (2):   0xd20d
0.036175 print_debug_deviceinfo      (2):   0xd20e
0.036177 print_debug_deviceinfo      (2):   0xd20f
0.036178 print_debug_deviceinfo      (2):   0xd210
0.036179 print_debug_deviceinfo      (2):   0xd211
0.036181 print_debug_deviceinfo      (2):   0xd212
0.036182 print_debug_deviceinfo      (2):   0xd213
0.036184 print_debug_deviceinfo      (2):   0xd214
0.036185 print_debug_deviceinfo      (2):   0xd215
0.036187 print_debug_deviceinfo      (2):   0xd216
0.036188 print_debug_deviceinfo      (2):   0xd217
0.036190 print_debug_deviceinfo      (2):   0xd218
0.036191 print_debug_deviceinfo      (2):   0xd219
0.036193 print_debug_deviceinfo      (2):   0xd21a
0.036194 print_debug_deviceinfo      (2):   0xd21b
0.036196 print_debug_deviceinfo      (2):   0xd406
0.036197 print_debug_deviceinfo      (2):   0xd407
```

jbreiden, 2017-01-24:
Wait, maybe I am getting something. Need to charge battery.

jbreiden, 2017-01-24:
```
$ gphoto2 --capture-tethered --camera="Fuji Fujifilm X-T2" --debug --debug-logfile=/tmp/debug.txt --debug-loglevel=debug
Waiting for events from camera. Press Ctrl-C to abort.                         
Saving file as DSCF0001.jpg                                                    
^C
Cancelling...
```

[debug.txt](https://github.com/gphoto/libgphoto2/files/727477/debug.txt)

msmeissn, 2017-01-25:
0.035898 print_debug_deviceinfo      (2):   0x100e (Initiate capture)

gphoto2 --capture-image now should also work

jbreiden, 2017-01-26:
```
$ gphoto2 --capture-image
                                                                               
*** Error ***              
PTP Device Busy
ERROR: Could not capture image.
ERROR: Could not capture.
*** Error (-110: 'I/O in progress') ***       

For debugging messages, please use the --debug option.
Debugging messages may help finding a solution to your problem.
If you intend to send any error or debug messages to the gphoto
developer mailing list <gphoto-devel@lists.sourceforge.net>, please run
gphoto2 as follows:

    env LANG=C gphoto2 --debug --debug-logfile=my-logfile.txt --capture-image

Please make sure there is sufficient quoting around the arguments.
```

[debug.txt](https://github.com/gphoto/libgphoto2/files/731345/debug.txt)

msmeissn, 2017-01-27:
usually that should just work as is when called. hard to say why it reports busy :( 

can you get output of:

gphoto2 --summary > summary.txt
gphoto2 --list-all-config > allconfig.txt

jbreiden, 2017-01-27:
Had to reschedule USB tracing, should have that by the end of Monday.

[summary.txt](https://github.com/gphoto/libgphoto2/files/736378/summary.txt)
[allconfig.txt](https://github.com/gphoto/libgphoto2/files/736379/allconfig.txt)

jbreiden, 2017-01-30:
https://drive.google.com/drive/folders/0B1NxYWevtHcMdHJmcVBLVEdzalU?usp=sharing

jbreiden, 2017-02-02:
Decoded USB traces added to same location. Note the double call to InitiateCapture.

COMMAND : code:PTP_OC_InitiateCapture
RESPONSE: code:PTP_RC_GeneralError
COMMAND : code:PTP_OC_InitiateCapture

jbreiden, 2017-02-02:
I think the camera is unhappy and becomes "PTP Device Busy" because when we try to list the files on the camera. Here is what gphoto2 is doing.

```
gphoto2 --debug --capture-image |& grep Sending | cut -f2 -d:
 Sending PTP_OC 0x1002 (Open session) (0x1) request...
 Sending PTP_OC 0x1001 (Get device info) request...
 Sending PTP_OC 0x1007 (Get object handles) (0xffffffff,0x0,0xffffffff) request...
 Sending PTP_OC 0x1004 (Get storage IDs) request...
 Sending PTP_OC 0x1007 (Get object handles) (0x10000001,0x0,0xffffffff) request...
 Sending PTP_OC 0x1007 (Get object handles) (0x10000002,0x0,0xffffffff) request...
 Sending PTP_OC 0x100e (Initiate capture) (0x0,0x0) request...
 FAIL
```

Comparing to the working USB trace, there isn't any attempt to get object handles until after capture is complete.  And when it does happen,  it looks different to my eye.

```
$ grep COMMAND ~/Downloads/Single\ Shot.txt | cut -f4- -d: | less
PTP_OC_GetObjectHandles        trans:8365 payload:'\xff\xff\xff\xff\x00\x00\x00\x00\x00\x00\x00\x00'
PTP_OC_GetObjectInfo           trans:8366 payload:'\x07\x00\x00\x00'
PTP_OC_GetObjectInfo           trans:8367 payload:'\x07\x00\x00\x00'
PTP_OC_GetObjectInfo           trans:8368 payload:'\x07\x00\x00\x00'
PTP_OC_MTP_GetObjPropList      trans:8369 payload:'\x07\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x01\x00\x00\x00\x00\x00\x00\x00'
PTP_OC_GetObject               trans:836a payload:'\x07\x00\x00\x00'
PTP_OC_DeleteObject            trans:836b payload:'\x07\x00\x00\x00'
```

msmeissn, 2017-02-03:
This is very helpful, thanks. It is setting various properties, that is probably more the reason.

(currently a bit busy witjh work and FOSDEM travel, hope to have some time next week to check it out more)

jbreiden, 2017-02-06:
This appears to be the magical incantation to allow InitiateCapture() to work. Only works only per camera powercycle. Testing with `gphoto2 --capture-tethered --debug`. If you don't set those properties we get a camera busy error.

```
        propval.u16 = 0x0002; C_PTP_REP (ptp_setdevicepropvalue (params, 0xd207, &propval, PTP_DTC_UINT16));
        propval.u16 = 0x0200; C_PTP_REP (ptp_setdevicepropvalue (params, 0xd208, &propval, PTP_DTC_UINT16));
        C_PTP_REP(ptp_initiatecapture(params, 0x00000000, 0x00000000));
```

msmeissn, 2017-02-07:
ignore last commit, it seems per-captuyre specific.

msmeissn, 2017-02-07:
(thanks for your work so far ... still not yet found time to properly go over it :/ )

jbreiden, 2017-04-19:
Looking at Fuji XT-2 again. This is what it takes to make a "beep" then a "click" using `gphoto2 --trigger-capture`. Contrary to my earlier reports above, the only property I'm messing with is `0xd208`

```c++
--- a/camlibs/ptp2/library.c
+++ b/camlibs/ptp2/library.c
@@ -4577,7 +4577,15 @@ camera_trigger_capture (Camera *camera, GPContext *context)
                        _("Sorry, your camera does not support generic capture"));
                return GP_ERROR_NOT_SUPPORTED;
        }
-       C_PTP_REP (ptp_initiatecapture(params, 0x00000000, 0x00000000));
+        {
+          PTPPropertyValue     propval;
+          propval.u16 = 0x0200; C_PTP_REP (ptp_setdevicepropvalue (params, 0xd208, &propval, PTP_DTC_UINT16));
+          C_PTP_REP (ptp_initiatecapture(params, 0x00000000, 0x00000000));
+          usleep(1000*1000);
+          propval.u16 = 0x0304; C_PTP_REP (ptp_setdevicepropvalue (params, 0xd208, &propval, PTP_DTC_UINT16));
+          C_PTP_REP(ptp_initiatecapture(params, 0x00000000, 0x00000000));
+        }
+
        return GP_OK;
 }
 ```

jbreiden, 2017-04-19:
... And now  I can't communicate with the camera under any circumstance. 

```
*** Error ***              
PTP Device Busy
0.089729 gp_camera_trigger_capture [gphoto2-camera.c:1369](0): 'camera->functions->trigger_capture (camera, context)' failed: -110
ERROR: Could not trigger capture.
```

jbreiden, 2017-04-24:
Flashed camera firmware to from v1.10 to v2.00; no change. Camera does want to talk to me anymore.

msmeissn, 2017-04-24:
urks :( sorry to hear :(

and sorry i currently did not pursue this issue  further .. quite busy and i focused a bit more on pentax and canon eos m, but wanted to get back here

msmeissn, 2017-04-24:
check for possible causes of busy condit9ion ... sd card full? bttery empty? ;)

jbreiden, 2017-05-04:
Talking to camera again.  For camera initialization, we do need to set device property `0xd207` to `0x002` one time only. To trigger capture, we need something like this. Still working out what to do next, will update the comment as I go.

```c++
            // focus
            propval.u16 = 0x0200;
            C_PTP_REP (ptp_setdevicepropvalue (params, 0xd208, &propval, PTP_DTC_UINT16));
            C_PTP_REP(ptp_initiatecapture(params, 0x00000000, 0x00000000));

            // poll camera until it is ready
            propval.u16 = 0x0001;
            while (propval.u16 == 0x0001) {
              ptp_getdevicepropvalue (params, 0xd209, &propval, PTP_DTC_UINT16);
            }

            // shoot
            propval.u16 = 0x0304;
            C_PTP_REP (ptp_setdevicepropvalue (params, 0xd208, &propval, PTP_DTC_UINT16));
            C_PTP_REP(ptp_initiatecapture(params, 0x00000000, 0x00000000));
```

jbreiden, 2017-05-05:
The hacked together code above is happy to run for a long time on USB2. When we get to USB3 we run for a few shots then crash.

```
4.217176 get_file_func               (2): Getting file 'DSCF0004.jpg'.
4.217189 ptp_usb_sendreq             (2): Sending PTP_OC 0x1009 (Get object) (0x4) request...
4.220294 ptp_usb_getdata             (2): Reading PTP_OC 0x1009 (Get object) data...
4.737728 gp_libusb1_read [libusb1.c:583](0): 'libusb_bulk_transfer (port->pl->dh, port->settings.usb.inep, (unsigned char*)bytes, size, &curread, port->timeout)' failed: Overflow (-8)
4.737748 gp_port_read [gphoto2-port.c:441](0): Reading 209408 = 0x33200 bytes from port failed: Error reading from the port (-34)
4.737828 ptp_usb_getdata [usb.c:431] (0): PTP_OC 0x1009 receiving data failed: PTP I/O Error (0x02ff)
4.737833 get_file_func [library.c:6698](0): 'ret' failed: 'PTP I/O Error' (0x02ff)
4.737839 gp_context_error            (0): PTP I/O Error
```

msmeissn, 2017-05-05:
overflow is an error I have not seen yet ... But I also currently have no usb3 hosts or cameras :/

jbreiden, 2017-05-05:
The crash was my fault. I had been screwing around with the USB max packet size as part of my explorations. Removing that local change made the crash go away.

jbreiden, 2017-05-05:
This is what things look like right now. Obviously this can't be an official patch because it replaces the default capture mechanisms with Fuji X-T2 specific ones. But hey, it's working and it should get the general idea across.

```c++
--- a/camlibs/ptp2/library.c
+++ b/camlibs/ptp2/library.c
@@ -2007,7 +2007,7 @@ static struct {
        /* https://github.com/gphoto/libgphoto2/issues/32 */
        {"Fuji:Fujifilm X-T10",                 0x04cb, 0x02c8, 0},
        /* with new updated firmware 1.1 */
-       {"Fuji:Fujifilm X-T2",                  0x04cb, 0x02cd, PTP_CAP},
+       {"Fuji:Fujifilm X-T2",                  0x04cb, 0x02cd, PTP_CAP|PTP_NO_CAPTURE_COMPLETE},
 
        {"Ricoh:Caplio R5 (PTP mode)",          0x05ca, 0x0110, 0},
        {"Ricoh:Caplio GX (PTP mode)",          0x05ca, 0x0325, 0},
@@ -4022,8 +4022,41 @@ fallback:
         * indicating that the capure has been completed may occur after
         * few seconds. moving down the code. (kil3r)
         */
-       C_PTP_REP (ptp_initiatecapture(params, 0x00000000, 0x00000000));
-       CR (gp_port_set_timeout (camera->port, capture_timeout));
+        {
+          PTPPropertyValue     propval;
+
+          GP_LOG_D ("XXX Attempting Fuji X-T2 capture");
+
+          // ask camera to listen to tethering commands instead of button
+          propval.u16 = 0x0002;
+          C_PTP_REP (ptp_setdevicepropvalue (params, 0xd207, &propval, PTP_DTC_UINT16));
+
+          // focus
+          propval.u16 = 0x0200;
+          C_PTP_REP (ptp_setdevicepropvalue (params, 0xd208, &propval, PTP_DTC_UINT16));
+          C_PTP_REP(ptp_initiatecapture(params, 0x00000000, 0x00000000));
+
+          // poll camera until it is ready
+          propval.u16 = 0x0001;
+          while (propval.u16 == 0x0001) {
+            ptp_getdevicepropvalue (params, 0xd209, &propval, PTP_DTC_UINT16);
+            GP_LOG_D ("XXX Ready to shoot? %X", propval.u16);
+          }
+
+          // shoot
+          propval.u16 = 0x0304;
+          C_PTP_REP (ptp_setdevicepropvalue (params, 0xd208, &propval, PTP_DTC_UINT16));
+          C_PTP_REP(ptp_initiatecapture(params, 0x00000000, 0x00000000));
+
+          // poll camera until it is ready
+          propval.u16 = 0x0000;
+          while (propval.u16 == 0x0000) {
+            ptp_getdevicepropvalue (params, 0xd212, &propval, PTP_DTC_UINT64);
+            GP_LOG_D ("XXX Ready after shooting? %lx", propval.u64);
+          }
+        }
+
+        CR (gp_port_set_timeout (camera->port, capture_timeout));
        /* A word of comments is worth here.
         * After InitiateCapture camera should report with ObjectAdded event
         * all newly created objects. However there might be more than one
@@ -4105,6 +4138,7 @@ fallback:
        while (done != 3) {
                uint16_t ret;
 
+#if 0
                C_PTP_REP (ptp_wait_event (params));
 
                if (!ptp_get_one_event(params, &event)) {
@@ -4114,6 +4148,21 @@ fallback:
                                        break;
                        continue;
                }
+#endif
+
+            {
+              PTPObjectHandles handles;
+              int i;
+
+              GP_LOG_D ("XXX Begin Fuji X-T2 fake event creation.");
+              C_PTP_REP (ptp_getobjecthandles (params, 0xffffffff, 0x000000, 0x000000, &handles));
+              while (handles.n == 0) {
+                C_PTP_REP (ptp_getobjecthandles (params, 0xffffffff, 0x000000, 0x000000, &handles));
+              }
+              event.Code = PTP_EC_ObjectAdded;
+              event.Param1 = handles.Handler[0];
+              event.Nparam = 1;
+            }
                GP_LOG_D ("event.Code is %04x / param %08x", event.Code, event.Param1);
 
                switch (event.Code) {
@@ -4577,7 +4626,39 @@ camera_trigger_capture (Camera *camera, GPContext *context)
                        _("Sorry, your camera does not support generic capture"));
                return GP_ERROR_NOT_SUPPORTED;
        }
-       C_PTP_REP (ptp_initiatecapture(params, 0x00000000, 0x00000000));
+        {
+          GP_LOG_D ("XXX Attempting Fuji X-T2 trigger-capture");
+          PTPPropertyValue     propval;
+
+          // ask camera to listen to tethering commands instead of button
+          propval.u16 = 0x0002;
+          C_PTP_REP (ptp_setdevicepropvalue (params, 0xd207, &propval, PTP_DTC_UINT16));
+
+          // focus
+          propval.u16 = 0x0200;
+          C_PTP_REP (ptp_setdevicepropvalue (params, 0xd208, &propval, PTP_DTC_UINT16));
+          C_PTP_REP(ptp_initiatecapture(params, 0x00000000, 0x00000000));
+
+          // poll camera until it is ready
+          propval.u16 = 0x0001;
+          while (propval.u16 == 0x0001) {
+            ptp_getdevicepropvalue (params, 0xd209, &propval, PTP_DTC_UINT16);
+            GP_LOG_D ("XXX Ready to shoot? %X", propval.u16);
+          }
+
+          // shoot
+          propval.u16 = 0x0304;
+          C_PTP_REP (ptp_setdevicepropvalue (params, 0xd208, &propval, PTP_DTC_UINT16));
+          C_PTP_REP(ptp_initiatecapture(params, 0x00000000, 0x00000000));
+
+          // poll camera until it is ready
+          propval.u16 = 0x0000;
+          while (propval.u16 == 0x0) {
+              ptp_getdevicepropvalue (params, 0xd212, &propval, PTP_DTC_UINT64);
+              GP_LOG_D ("XXX Ready after shooting? %lx", propval.u64);
+          }
+        }
+
        return GP_OK;
 }
 
```

jbreiden, 2017-05-06:
I'm seeing maybe 8 shots per minute performance increase from adjusting USB read sizes. I also notice `camera_wait_for_event` basically never works, so disable that. It seems get called in between every shot for some reason, haven't bother tracing. With these two changes, I'm seeing somewhere around 73 shots per minute for 10MB JPEG.

```diff
diff --git a/camlibs/ptp2/usb.c b/camlibs/ptp2/usb.c
index 065c753f3..1fa6ad73e 100644
--- a/camlibs/ptp2/usb.c
+++ b/camlibs/ptp2/usb.c
@@ -256,7 +256,7 @@ retry:
        return PTP_ERROR_IO;
 }
 
-#define READLEN 512*1024 /* read blob size, mostly to avoid reading all of it at once. */
+#define READLEN 32*1024*1024 /* read blob size, mostly to avoid reading all of it at once. */
 
 uint16_t
 ptp_usb_getdata (PTPParams* params, PTPContainer* ptp, PTPDataHandler *handler)
```

```diff
diff --git a/camlibs/ptp2/library.c b/camlibs/ptp2/library.c
index bc586525d..fab3cc5bc 100644
--- a/camlibs/ptp2/library.c
+++ b/camlibs/ptp2/library.c

@@ -4604,6 +4684,8 @@ camera_wait_for_event (Camera *camera, int timeout,
        *eventtype = GP_EVENT_TIMEOUT;
        *eventdata = NULL;
 
+        return GP_OK;
+
        if (params->device_flags & DEVICE_FLAG_OLYMPUS_XML_WRAPPED) {
                GP_LOG_D ("olympus setcameracontrolmode 2\n");
                C_PTP_REP (ptp_olympus_setcameracontrolmode (params, 2));
```

msmeissn, 2017-05-06:
camera_wait_for_event is called by gphoto2 itself. but it will wait only the assigned time.

Increasing the usb readsize from 512kb to 32MB is a bit much

can you check the debuglogfile for events? will be visible around lines with 
_cb_irq  in them
can you attach one debuglogfile perhaps?

msmeissn, 2017-05-06:
I created a "fuji" branch in GIT where I integrated your code.

The change done is that I do it in camera_fuji_capture and I reuse the nikon brokencap code for object detection currently.

jbreiden, 2017-05-08:
This is a debug log for the Fuji branch, as per request.

```
$ gphoto2 --capture-image-and-download  --debug-loglevel=debug --debug --force-overwrite  --debug-logfile=fuji-branch.txt -I 2 -F 3
Time-lapse mode enabled (interval: 2s).                                        
Capturing frame #1/3...
New file is in location /store_10000001/DSCF0001.jpg on the camera
Saving file as DSCF0001.jpg                                                    
Deleting file /store_10000001/DSCF0001.jpg on the camera
Waiting for next capture slot 1 seconds...

*** Error ***              
PTP Invalid Object Handle
Capturing frame #2/3...
New file is in location /store_10000001/DSCF0002.jpg on the camera
Saving file as DSCF0002.jpg                                                    
Deleting file /store_10000001/DSCF0002.jpg on the camera
Waiting for next capture slot 0 seconds...
Capturing frame #3/3...
New file is in location /store_10000001/DSCF0003.jpg on the camera
Saving file as DSCF0003.jpg                                                    
Deleting file /store_10000001/DSCF0003.jpg on the camera
```

[fuji-branch.txt](https://github.com/gphoto/libgphoto2/files/984672/fuji-branch.txt)

msmeissn, 2017-05-08:
there is event happening, but the data is not in the dump. can you look for lines with _cb_irq in a full dump with data and quote the data inside? likely is 16 bytes or so
0.547019 _cb_irq                     (2): 0x1b32718 with status 0
0.547037 _cb_irq                     (2): requeuing complete transfer 0x1b32718

jbreiden, 2017-05-08:
Three images, but only one interrupt. `gphoto2 --capture-image-and-download -I 5 -F 3 --debug --debug-logfile=full.txt`

```
0.618652 _cb_irq                     (2): 0x180b518 with status 0
0.618667 _cb_irq                     (3): interrupt (hexdump of 16 bytes)
0000  10 00 00 00 04 00 02 40-ff ff ff ff 01 00 00 00  .......@........

0.618681 _cb_irq                     (2): requeuing complete transfer 0x180b518
0.619260 gp_port_read                (3): Read    20 = 0x14 out of 1024 bytes from port: (hexdump of 20 bytes)
0000  14 00 00 00 02 00 15 10-22 01 00 00 01 00 0e d2  ........".......
0010  12 00 00 00
[...]
18.750844 _cb_irq                     (2): 0x180b518 with status 3
18.750864 _cb_irq                     (2): 0x180b6e8 with status 3
18.750874 _cb_irq                     (2): 0x1813898 with status 3
18.750883 _cb_irq                     (2): 0x1813a68 with status 3
18.750892 _cb_irq                     (2): 0x1813c38 with status 3
18.750918 _cb_irq                     (2): 0x1813e08 with status 3
18.750928 _cb_irq                     (2): 0x1814018 with status 3
18.750937 _cb_irq                     (2): 0x1814228 with status 3
18.750947 _cb_irq                     (2): 0x1814438 with status 3
18.750956 _cb_irq                     (2): 0x1814648 with status 3
```

jbreiden, 2017-05-09:
Well, I still don't understand how to use git/github,. Here's a patch for the Fuji branch.

```
commit 999ddaa326202b033308d469bf16a399e4cba710 (HEAD -> fuji)
Author: Jeff Breidenbach <jbreiden@google.com>
Date:   Mon May 8 15:18:31 2017 -0700

    Speed up captures per minute from about 35 to about 70.
    
    WARNING: This is tested on a camera that does not have an
    SD card inserted. It is very likely that this code will
    do the wrong thing if an SD card is involved.
    
    Also some minor changes:
     * Remove some debug messages
     * Make one of the polling loops less tight
     * Move a camera initialization command to camera_prepare_capture.

diff --git a/camlibs/ptp2/config.c b/camlibs/ptp2/config.c
index 0b758fbbf..447830a49 100644
--- a/camlibs/ptp2/config.c
+++ b/camlibs/ptp2/config.c
@@ -492,6 +492,13 @@ camera_prepare_capture (Camera *camera, GPContext *context)
                        return camera_prepare_canon_eos_capture(camera,context);
                gp_context_error(context, _("Sorry, your Canon camera does not support Canon capture"));
                return GP_ERROR_NOT_SUPPORTED;
+        case PTP_VENDOR_FUJI:
+               {
+                       PTPPropertyValue propval;
+                       propval.u16 = 0x0002;
+                       return ptp_setdevicepropvalue (params, 0xd207, &propval, PTP_DTC_UINT16);
+               }
+               break;
        default:
                /* generic capture does not need preparation */
                return GP_OK;
diff --git a/camlibs/ptp2/library.c b/camlibs/ptp2/library.c
index 32d7c1423..b62b059f8 100644
--- a/camlibs/ptp2/library.c
+++ b/camlibs/ptp2/library.c
@@ -3923,18 +3923,10 @@ camera_fuji_capture (Camera *camera, CameraCaptureType type, CameraFilePath *pat
 {
        PTPParams               *params = &camera->pl->params;
        PTPPropertyValue        propval;
-       PTPObjectHandles        handles, beforehandles;
-       int                     tries;
-       uint32_t                newobject;
+       PTPObjectHandles        handles;
 
        GP_LOG_D ("camera_fuji_capture");
 
-       C_PTP (ptp_getobjecthandles (params, PTP_HANDLER_SPECIAL, 0x000000, 0x000000, &beforehandles));
-
-       /* ask camera to listen to tethering commands instead of button */
-       propval.u16 = 0x0002;
-       C_PTP_REP (ptp_setdevicepropvalue (params, 0xd207, &propval, PTP_DTC_UINT16));
-
        /* focus */
        propval.u16 = 0x0200;
        C_PTP_REP (ptp_setdevicepropvalue (params, 0xd208, &propval, PTP_DTC_UINT16));
@@ -3944,7 +3936,6 @@ camera_fuji_capture (Camera *camera, CameraCaptureType type, CameraFilePath *pat
        propval.u16 = 0x0001;
        while (propval.u16 == 0x0001) {
                ptp_getdevicepropvalue (params, 0xd209, &propval, PTP_DTC_UINT16);
-               GP_LOG_D ("XXX Ready to shoot? %X", propval.u16);
        }
 
        /* shoot */
@@ -3952,77 +3943,23 @@ camera_fuji_capture (Camera *camera, CameraCaptureType type, CameraFilePath *pat
        C_PTP_REP (ptp_setdevicepropvalue (params, 0xd208, &propval, PTP_DTC_UINT16));
[... 92 lines cut ...]
 
-               /* ask camera to listen to tethering commands instead of button */
-               propval.u16 = 0x0002;
-               C_PTP_REP (ptp_setdevicepropvalue (params, 0xd207, &propval, PTP_DTC_UINT16));
-
                /* focus */
                propval.u16 = 0x0200;
                C_PTP_REP (ptp_setdevicepropvalue (params, 0xd208, &propval, PTP_DTC_UINT16));
@@ -4695,7 +4628,6 @@ camera_trigger_capture (Camera *camera, GPContext *context)
                propval.u16 = 0x0001;
                while (propval.u16 == 0x0001) {
                        ptp_getdevicepropvalue (params, 0xd209, &propval, PTP_DTC_UINT16);
-                       GP_LOG_D ("XXX Ready to shoot? %X", propval.u16);
                }
 
                /* shoot */
@@ -4707,7 +4639,6 @@ camera_trigger_capture (Camera *camera, GPContext *context)
                propval.u16 = 0x0000;
                while (propval.u16 == 0x0000) {
                        ptp_getdevicepropvalue (params, 0xd212, &propval, PTP_DTC_UINT64);
-                       GP_LOG_D ("XXX Ready after shooting? %lx", propval.u64);
                }
        }
 
```

msmeissn, 2017-05-09:
The camera sends a ObjectAdded event with objectid 1, which is what we need.

so i think we do not need the ptp_get_object_handles() poll loop

I commited the d207 move from your patch to fuji branch, but changed the camera_fuji_capture to just wait for the ObjectAdded event now. (rest is in #if 0)

jbreiden, 2017-05-09:
But we only get one ObjectAdded event per camera power cycle.

msmeissn, 2017-05-09:
hmm. this object ... is it in SDRAM of the camera? perhaps it needs to be deleted before the next capture?

jbreiden, 2017-05-09:
Yes, this is all SDRAM, no SD card installed.  The camera will take a second shot, but I can't get it off because there is no second ObjectAdded event. I know there was a second shot because the camera clicked, and the status LED flashes.

On the other hand, polling works fine. Using polling I took 5000 consecutive shots this morning before the battery drained.

msmeissn, 2017-05-09:
hmm.

will it always start with a empty object handles array like in your change above as you are just checking for any object? 

this will probably also not work with SD Card inserted

jbreiden, 2017-05-09:
Here's a polling approach that works quickly and safely.

```diff
diff --git a/camlibs/ptp2/library.c b/camlibs/ptp2/library.c
index be3eddd19..b06f0da79 100644
--- a/camlibs/ptp2/library.c
+++ b/camlibs/ptp2/library.c
@@ -3926,31 +3926,22 @@ camera_fuji_capture (Camera *camera, CameraCaptureType type, CameraFilePath *pat
 {
        PTPParams               *params = &camera->pl->params;
        PTPPropertyValue        propval;
-#if 0
        PTPObjectHandles        handles, beforehandles;
-       int                     tries;
-#endif
-       PTPContainer            event;
-       uint32_t                newobject;
-       struct timeval          event_start;
-       int                     back_off_wait = 0;
 
        GP_LOG_D ("camera_fuji_capture");
 
-#if 0
+       /* see what images are already around */
        C_PTP (ptp_getobjecthandles (params, PTP_HANDLER_SPECIAL, 0x000000, 0x000000, &beforehandles));
-#endif
 
        /* focus */
        propval.u16 = 0x0200;
        C_PTP_REP (ptp_setdevicepropvalue (params, 0xd208, &propval, PTP_DTC_UINT16));
        C_PTP_REP(ptp_initiatecapture(params, 0x00000000, 0x00000000));
 
-       /* poll camera until it is ready */
+       /* poll until camera ready */
        propval.u16 = 0x0001;
        while (propval.u16 == 0x0001) {
                ptp_getdevicepropvalue (params, 0xd209, &propval, PTP_DTC_UINT16);
-               GP_LOG_D ("XXX Ready to shoot? %X", propval.u16);
        }
 
        /* shoot */
@@ -3958,100 +3949,23 @@ camera_fuji_capture (Camera *camera, CameraCaptureType type, CameraFilePath *pat
        C_PTP_REP (ptp_setdevicepropvalue (params, 0xd208, &propval, PTP_DTC_UINT16));
        C_PTP_REP(ptp_initiatecapture(params, 0x00000000, 0x00000000));
 
-       /* poll camera until it is ready */
-       propval.u16 = 0x0000;
-       while (propval.u16 == 0x0000) {
-               C_PTP_REP (ptp_getdevicepropvalue (params, 0xd212, &propval, PTP_DTC_UINT64));
-               GP_LOG_D ("XXX Ready after shooting? %lx", propval.u64);
-               C_PTP_REP (ptp_check_event (params));
-       }
-
-       /* there is a ObjectAdded event being sent */
-       do {
-               C_PTP_REP (ptp_check_event (params));
-
-               while (ptp_get_one_event(params, &event)) {
-                       switch (event.Code) {
-                       case PTP_EC_ObjectAdded:
-                               newobject = event.Param1;
-                               goto downloadfile;
-                       default:
-                               GP_LOG_D ("unexpected unhandled event Code %04x, Param 1 %08x", event.Code, event.Param1);
-                               break;
-                       }
-               }
-       }  while (waiting_for_timeout (&back_off_wait, event_start, 2000)); /* wait for 2 more seconds after busy is no longer signaled */
-
-       return GP_ERROR;
[... 70 lines cut ...]
-               /* delete last / or we get confused later. */
-               path->folder[ strlen(path->folder)-1 ] = '\0';
-               return gp_filesystem_append (camera->fs, path->folder, path->name, context);
-       }
-       return GP_ERROR;
+       PTPObject       *ob;
+       C_PTP_REP (ptp_object_want (params, handles.Handler[handles.n - 1], PTPOBJECT_OBJECTINFO_LOADED, &ob));
+       strcpy  (path->name,  ob->oi.Filename);
+       sprintf (path->folder,"/"STORAGE_FOLDER_PREFIX"%08lx/",(unsigned long)ob->oi.StorageID);
+       get_folder_from_handle (camera, ob->oi.StorageID, ob->oi.ParentObject, path->folder);
+       path->folder[ strlen(path->folder)-1 ] = '\0';
+       return gp_filesystem_append (camera->fs, path->folder, path->name, context);
 }
 
 static int
@@ -4730,7 +4644,7 @@ camera_trigger_capture (Camera *camera, GPContext *context)
                /* poll camera until it is ready */
                propval.u16 = 0x0000;
                while (propval.u16 == 0x0000) {
-                       C_PTP_REP (ptp_getdevicepropvalue (params, 0xd212, &propval, PTP_DTC_UINT64));
+                       C_PTP_REP (ptp_getdevicepropvalue (params, 0xd212, &propval, PTP_DTC_UINT16));
                }
        }
 
```

jbreiden, 2017-05-15:
Looks like we get ourselves in trouble at shot ten thousand. Not sure if this is a probably with the camera or with gphoto2.

```
Capturing frame #9999...
New file is in location /store_10000001/DSCF9999.jpg on the camera
Saving file as DSCF9999.jpg                                                    
Deleting file /store_10000001/DSCF9999.jpg on the camera
not sleeping (0 seconds behind schedule)
Capturing frame #10000...
New file is in location /store_10000001/DSCF.jpg on the camera
Saving file as DSCF.jpg                                                        
Deleting file /store_10000001/DSCF.jpg on the camera
not sleeping (0 seconds behind schedule)
Capturing frame #10001...
New file is in location /store_10000001/DSCF.jpg on the camera
File DSCF.jpg exists. Overwrite? [y|n]          
```

msmeissn, 2017-05-15:
On Mon, May 15, 2017 at 08:56:34AM -0700, jbreiden wrote:
> Looks like we get ourselves in trouble at shot ten thousand. Not sure if this is a probably with the camera or with gphoto2.

Does the camera create a new folder on this occasion? That would cause trouble on name decoding.

msmeissn, 2017-05-15:
There is config.c:camera_unprepare_capture  for this purpose :)

msmeissn, 2017-05-15:
btw, i merged the fuji branch back into master branch

jbreiden, 2017-05-16:
Okay, so while filenames get weird at shot number 10000 we are able to still takes them and get them off the camera.  Unfortunately everything grinds to a halt on shot 35634.

```
Capturing frame #35633...
New file is in location /store_10000001/DSCF.jpg on the camera
Saving file as latest.jpg                                                      
Deleting file /store_10000001/DSCF.jpg on the camera
not sleeping (0 seconds behind schedule)
Capturing frame #35634...
```

jbreiden, 2017-05-16:
I can't afford this half second delay waiting for an interrupt. 

https://github.com/gphoto/libgphoto2/blob/master/camlibs/ptp2/library.c#L3983

Could it be removed, perhaps like this patch?

https://github.com/gphoto/libgphoto2/issues/133#issuecomment-300251622

jbreiden, 2017-05-17:
There is definitely is a second folder involved. Don't know yet if it happens on image 10000 or  35634, my guess is the latter.

```
$ gphoto2 --list-files
There is no file in folder '/'.                                                
There is no file in folder '/store_10000001'.
There is no file in folder '/store_10000002'.
```

Here's how to release control of physical camera buttons when we are done. This helps a lot.

```diff
diff --git a/camlibs/ptp2/config.c b/camlibs/ptp2/config.c
index 447830a49..1f0891e2d 100644
--- a/camlibs/ptp2/config.c
+++ b/camlibs/ptp2/config.c
@@ -572,6 +572,14 @@ camera_unprepare_capture (Camera *camera, GPContext *context)
                gp_context_error(context,
                _("Sorry, your Canon camera does not support Canon capture"));
                return GP_ERROR_NOT_SUPPORTED;
+        case PTP_VENDOR_FUJI:
+               {
+                       PTPPropertyValue propval;
+                        PTPParams *params = &camera->pl->params;
+                       propval.u16 = 0x0001;
+                       return ptp_setdevicepropvalue (params, 0xd207, &propval, PTP_DTC_UINT16);
+               }
+               break;
        default:
                /* generic capture does not need unpreparation */
                return GP_OK;
diff --git a/camlibs/ptp2/library.c b/camlibs/ptp2/library.c
index b62b059f8..3321ae8f8 100644
--- a/camlibs/ptp2/library.c
+++ b/camlibs/ptp2/library.c
@@ -2456,6 +2456,9 @@ camera_exit (Camera *camera, GPContext *context)
                                C_PTP (ptp_sony_9280(params, 0x4,0,5,0,0,0,0));
                        }
                        break;
+               case PTP_VENDOR_FUJI:
+                       camera_unprepare_capture (camera, context);
+                       break;
                }
 
                if (camera->pl->checkevents)
```

msmeissn, 2017-05-17:
See camera_fuji_capture, the polling loop is already there.

I applied your changes (plus some adjustments) and also removed  the 0.5 second event wait loop in git master.

jbreiden, 2017-05-17:
Thank you! The checked in code works, but is still too slow. I'm not totally sure where the speed problem is, but I have this variation running right now at 75+ shots per minute. We'll see what happens when it reaches 32K shots.

```c++
static int
camera_fuji_capture (Camera *camera, CameraCaptureType type, CameraFilePath *path, GPContext *context)
{
	PTPParams		*params = &camera->pl->params;
	PTPPropertyValue	propval;
	PTPObjectHandles	handles, beforehandles;
	PTPObject               *ob;

	GP_LOG_D ("camera_fuji_capture");

	C_PTP (ptp_getobjecthandles (params, PTP_HANDLER_SPECIAL, 0x000000, 0x000000, &beforehandles));

	/* focus */
	propval.u16 = 0x0200;
	C_PTP_REP (ptp_setdevicepropvalue (params, 0xd208, &propval, PTP_DTC_UINT16));
	C_PTP_REP(ptp_initiatecapture(params, 0x00000000, 0x00000000));

	/* poll camera until it is ready */
	propval.u16 = 0x0001;
	while (propval.u16 == 0x0001) {
		ptp_getdevicepropvalue (params, 0xd209, &propval, PTP_DTC_UINT16);
	}

	/* shoot */
	propval.u16 = 0x0304;
	C_PTP_REP (ptp_setdevicepropvalue (params, 0xd208, &propval, PTP_DTC_UINT16));
	C_PTP_REP (ptp_initiatecapture(params, 0x00000000, 0x00000000));

        /* poll until image is ready */
	C_PTP_REP (ptp_getobjecthandles (params, 0xffffffff, 0x000000, 0x000000, &handles));
	while (1) {
          usleep(20 * 1000);
          C_PTP_REP (ptp_getobjecthandles (params, 0xffffffff, 0x000000, 0x000000, &handles));
          if (handles.n == beforehandles.n) continue;
          C_PTP_REP (ptp_object_want (params, handles.Handler[handles.n - 1], PTPOBJECT_OBJECTINFO_LOADED, &ob));
          if (ob->oi.ObjectFormat != PTP_OFC_Association) break;
	}

	/* extract image path */
	path->name[0]='\0';
	path->folder[0]='\0';
        strcpy  (path->name,  ob->oi.Filename);
	sprintf (path->folder,"/"STORAGE_FOLDER_PREFIX"%08lx/",(unsigned long)ob->oi.StorageID);
	get_folder_from_handle (camera, ob->oi.StorageID, ob->oi.ParentObject, path->folder);
	path->folder[ strlen(path->folder)-1 ] = '\0';

        /* clean up and write image data */
        free (handles.Handler);
	free (beforehandles.Handler);
        return gp_filesystem_append (camera->fs, path->folder, path->name, context);
}
```

jbreiden, 2017-06-07:
Unless I am very confused, I think these should be PTP_DTC_UINT16

https://github.com/gphoto/libgphoto2/blob/master/camlibs/ptp2/library.c#L3978
https://github.com/gphoto/libgphoto2/blob/master/camlibs/ptp2/library.c#L4748

Also, when polling the camera, it we currently look for anything and ignore directories. It's more elegant to ask the camera for handles from images only, which can be done by feeding the parameter 0xFFFFU to the third parameter of GetObjectHandles instead of 0x000000.

https://github.com/gphoto/libgphoto2/blob/master/camlibs/ptp2/library.c#L4035

msmeissn, 2017-06-15:
the last ... i doubt it makes any kind of speed difference.

for the first, i used your patch. The 0xD212 property seems to occasionaly return more data in your traces, but this might be a mistake of the tracing tool?

jbreiden, 2017-06-20:
FujiFilm GFX 50 S is 04cb:02d3 on USB. Works fine.

msmeissn, 2017-06-20:
does the GFX 50S support capture?

jbreiden, 2017-07-10:
Yes, the GFX 50S supports capture.

jbreiden, 2017-07-10:
Here are the results of querying the camera itself about properties. There are a few weird ones. As far as I can tell `0xd212` completely violates the PTP specification; amongst other things it has an invalid datatype code.  `0xd153` is missing the form field, which I think is required. All the string properties are also missing the form field, but I don't know if that violates the PTP specification or not.

```text
 Working on prop_code 0x5001
   Device Property Not Supported response_code=0x200a
 Working on prop_code 0x5003
{
  property_code=0x5003
  datatype_code=0xFFFF
  settable=y
  factory_default_value=6000x4000
  current_value=6000x4000
  supported_value=3008x2000
  supported_value=3008x1688
  supported_value=2000x2000
  supported_value=4240x2832
  supported_value=4240x2384
  supported_value=2832x2832
  supported_value=6000x4000
  supported_value=6000x3376
  supported_value=4000x4000
}
 Working on prop_code 0x5005
{
  property_code=0x5005
  datatype_code=0x0004
  settable=y
  factory_default_value=2
  current_value=2
  supported_value=2
  supported_value=4
  supported_value=6
  supported_value=8
  supported_value=32769
  supported_value=32770
  supported_value=32771
  supported_value=32774
  supported_value=32775
  supported_value=32776
  supported_value=32777
  supported_value=32778
}
 Working on prop_code 0x500a
{
  property_code=0x500A
  datatype_code=0x0004
  settable=y
  factory_default_value=1
  current_value=32769
  supported_value=1
  supported_value=32769
  supported_value=32770
}
 Working on prop_code 0x500b
{
  property_code=0x500B
  datatype_code=0x0004
  settable=y
  factory_default_value=1
  current_value=3
  supported_value=1
  supported_value=2
  supported_value=3
  supported_value=4
}
 Working on prop_code 0x500c
   Device Property Not Supported response_code=0x200a
 Working on prop_code 0x500e
{
  property_code=0x500E
[... 689 lines cut ...]
 Working on prop_code 0xd218
   Device Property Not Supported response_code=0x200a
 Working on prop_code 0xd219
   Device Property Not Supported response_code=0x200a
 Working on prop_code 0xd21a
   Device Property Not Supported response_code=0x200a
 Working on prop_code 0xd21b
   Device Property Not Supported response_code=0x200a
 Working on prop_code 0xd406
{
  property_code=0xD406
  datatype_code=0xFFFF
  settable=y
  factory_default_value=
  current_value=
}
 Working on prop_code 0xd407
{
  property_code=0xD407
  datatype_code=0x0006
  settable=n
  factory_default_value=1
  current_value=1
}
```

xoul, 2017-08-31:
A friend has lent me a Fuji X-T2 to test with my code, but I'm having problem with gphoto2.
As you can see, gphoto2 2.5.14 is able to capture and download images from the camera, even if it returns an error on strating. But as long as I just use "--capture-image" it reports more severe errors and I have to reboot che computer to be able to shoot again!

Having tried also "--reset" with no results. Is there something I can try, softer that reboot?

Also, is there a way to simulate the same behaviour as "--capture-image-and-download" using the Python binding to libghoto2 since just capturing an image would not work?

Debug files are very big, even when zipped. I can send/attache the relevant parts if needed.

# env LANG=C gphoto2 --debug --debug-logfile=fujixt2-1.txt --port usb:001,005 --capture-image-and-download

*** Error ***
PTP General Error
ERROR: Could not capture image.
ERROR: Could not capture.
New file is in location /store_10000001/DSCF0001.jpg on the camera
Saving file as DSCF0001.jpg
Deleting file /store_10000001/DSCF0001.jpg on the camera
# env LANG=C gphoto2 --debug --debug-logfile=fujixt2-2.txt --port usb:001,005 --capture-image-and-download
New file is in location /store_10000001/DSCF0002.jpg on the camera
Saving file as DSCF0002.jpg
Deleting file /store_10000001/DSCF0002.jpg on the camera
# env LANG=C gphoto2 --debug --debug-logfile=fujixt2-3.txt --port usb:001,005 --capture-image
New file is in location /store_10000001/DSCF0003.jpg on the camera
# env LANG=C gphoto2 --debug --debug-logfile=fujixt2-4.txt --port usb:001,005 --capture-image

*** Error ***
An error occurred in the io-library ('Unknown error'): No error description available
ERROR: Could not capture image.
ERROR: Could not capture.
*** Error (-110: 'I/O in progress') ***

# env LANG=C gphoto2 --debug --debug-logfile=fujixt2-5.txt --port usb:001,005 --capture-image

*** Error ***
An error occurred in the io-library ('Unknown error'): No error description available
ERROR: Could not capture image.
ERROR: Could not capture.
*** Error (-110: 'I/O in progress') ***

# env LANG=C gphoto2 --debug --debug-logfile=fujixt2-6.txt --port usb:001,005 --capture-image-and-download

*** Error ***
An error occurred in the io-library ('Unknown error'): No error description available
ERROR: Could not capture image.
ERROR: Could not capture.
*** Error (-110: 'I/O in progress') ***

elijahparker, 2017-09-22:
I've been doing a bit of work with Fuji here this week and it's been quite interesting.  I've got full camera control (ISO, shutter, aperture) as well as live view and reliable capture.  Like @jbreiden noted, it seems to violate standards in many ways.  For switching to PC mode (to then allow setting the camera parameters), it requires setting a property, 0xd38c, to 1, which isn't even listed in the camera's supported property codes.  @msmeissn -- is there a system in place to handle properties that are not reported?  Right now to make it work I hacked _set_config in config.c to allow a list of hidden properties if the vendor is Fuji, but it's messy. 

Also, capture would fail for shutter speeds longer than a few seconds.  This is due to detecting the "object added" event.  0xd212 polls events, but the current code was triggering any time a single event was detected.  It actually turns out that first byte is the number of events, followed by a array with an item size of 6 bytes, the first 2 the property, and the last 4 the value.  Here's the code I'm using to parse this:

```
uint16_t ptp_fuji_getevents (PTPParams* params, uint16_t** events, uint16_t* count)
{
        PTPContainer    ptp;
        unsigned char   *data;
        unsigned int    size;
        uint16_t        ret;                  

        PTP_CNT_INIT(ptp, PTP_OC_GetDevicePropValue, 0xd212);
        CHECK_PTP_RC(ptp_transaction(params, &ptp, PTP_DP_GETDATA, 0, &data, &size));
        ptp_debug(params, "ptp_fuji_getevents");
        int i;                    
        *count = 0;                         
        if(size >= 2)
        {
                memcpy(count, data, 2);
                ptp_debug(params, "event count: %d", *count);
                *events = calloc(*count, sizeof(uint16_t));
                if(size >= 2 + *count * 6)
                {
                        uint16_t param;
                        uint32_t value;
                        for(i = 0; i < *count; i++)
                        {
                                memcpy(&param, &data[2 + 6 * i], 2);
                                memcpy(&value, &data[2 + 6 * i + 2], 4);
                                *events[i] = param;
                                ptp_debug(params, "param: %02x, value: %d ", param, value);
                        }
                }             
        }
        free(data);
        return PTP_RC_OK;        
}
```
The event property that appears to signify the object added is 0xd20d, and this seems to be very reliable on the Fuji X-T1 I'm using for testing.  Here's code in the capture routine that looks for this:

```
/* poll camera until it is ready */        
propval.u16 = 0x0000;
int loops = 0, i;
uint16_t *events;
uint16_t count, ready = 0;  
while (loops < 3300) { // loop for about 33 seconds max
        ptp_fuji_getevents (params, &events, &count);
        if(count > 0) {
                for(i = 0; i < count; i++) {
                        if(events[i] == 0xd20d) {
                                ready = 1;
                                break;        
                        }
                }                   
                free(events);
                if(ready) break;
        }
        C_PTP_REP (ptp_check_event (params));
        usleep(10000);       
        i++;              
}
```
I'm still trying to figure out how to best incorporate everything now.  The main thing is how unlisted properties are handled.

Thanks @jbreiden and @msmeissn for getting it this far!

[37 more replies]

closed 2021-06-19 by msmeissn