gphoto/libgphoto2 #135: Cannot record video using Canon EOS 650D

nmrmak, 2017-01-25:
Hi Marcus,

It seems that video recording functionality for Canon is broken. When I try to do (as described here: http://gphoto.org/doc/remote/)

gphoto2 --set-config viewfinder=1 --set-config movierecordtarget=Card --wait-event=10s --set-config movierecordtarget=None --wait-event-and-download=2s

I get (full log warning, question continued below):

gphoto2 --set-config viewfinder=1 --set-config movierecordtarget=Card --wait-event=10s --set-config movierecordtarget=None --wait-event-and-download=2s

*** Error ***
The property 'Recording Destination' / 0xd1b8 was not set (0x2019: PTP Device Busy).
Waiting for 10 seconds for events from camera. Press Ctrl-C to abort.
UNKNOWN PTP Property d105 changed
UNKNOWN PTP Property d108 changed
UNKNOWN PTP Property d106 changed
UNKNOWN PTP Property d107 changed
UNKNOWN PTP Property d109 changed
UNKNOWN PTP Property d10a changed
UNKNOWN PTP Property d10b changed
UNKNOWN PTP Property d10c changed
UNKNOWN PTP Property d10d changed
UNKNOWN PTP Property d10e changed
UNKNOWN PTP Property d10f changed
UNKNOWN PTP Property d11b changed
UNKNOWN PTP Property d114 changed
UNKNOWN PTP Property d116 changed
UNKNOWN PTP Property d119 changed
UNKNOWN PTP Property d110 changed
UNKNOWN PTP Property d101 changed
UNKNOWN PTP Property d102 changed
UNKNOWN PTP Property d103 changed
UNKNOWN PTP Property d104 changed
UNKNOWN PTP Property d11d changed
UNKNOWN PTP Property d111 changed
UNKNOWN PTP Property d112 changed
UNKNOWN PTP Property d113 changed
UNKNOWN PTP Property d120 changed
UNKNOWN PTP Property d122 changed
UNKNOWN PTP Property d156 changed
UNKNOWN PTP Property d150 changed
UNKNOWN PTP Property d151 changed
UNKNOWN PTP Property d152 changed
UNKNOWN PTP Property d153 changed
UNKNOWN PTP Property d154 changed
UNKNOWN PTP Property d155 changed
UNKNOWN PTP Property d160 changed
UNKNOWN PTP Property d161 changed
UNKNOWN PTP Property d162 changed
UNKNOWN PTP Property d11c changed
UNKNOWN PTP Property d1a0 changed
UNKNOWN PTP Property d1a1 changed
UNKNOWN PTP Property d1a8 changed
UNKNOWN PTP Property d1a7 changed
UNKNOWN PTP Property d1ab changed
UNKNOWN PTP Property d1b0 changed
UNKNOWN PTP Property d1b1 changed
UNKNOWN PTP Property d1b2 changed
UNKNOWN PTP Property d1b3 changed
UNKNOWN PTP Property d1b4 changed
UNKNOWN PTP Property d1b5 changed
UNKNOWN PTP Property d1b6 changed
UNKNOWN PTP Property d1a9 changed
UNKNOWN PTP Property d146 changed
UNKNOWN PTP Property d1ac changed
UNKNOWN PTP Property d1aa changed
UNKNOWN PTP Property d11e changed
UNKNOWN PTP Property d11f changed
UNKNOWN PTP Property d1d9 changed
UNKNOWN PTP Property d1ba changed
UNKNOWN PTP Property d1cc changed
UNKNOWN PTP Property d1bc changed
UNKNOWN PTP Property d1b8 changed
UNKNOWN PTP Property d1d3 changed
UNKNOWN PTP Property d1d8 changed
UNKNOWN PTP Property d1b7 changed
UNKNOWN PTP Property d1cb changed
UNKNOWN PTP Property d1db changed
UNKNOWN PTP Property d1dc changed
UNKNOWN PTP Property d1a4 changed
UNKNOWN PTP Property d1df changed
UNKNOWN PTP Property d1bd changed
UNKNOWN PTP Property d1c1 changed
UNKNOWN PTP Property d1c0 changed
UNKNOWN PTP Property d1bf changed
UNKNOWN PTP Property d1c4 changed
UNKNOWN PTP Property d1c2 changed
UNKNOWN PTP Property d1c5 changed
UNKNOWN PTP Property d194 changed
UNKNOWN PTP Property d195 changed
UNKNOWN PTP Property d196 changed
UNKNOWN PTP Property d197 changed
UNKNOWN PTP Property d198 changed
UNKNOWN PTP Property d1dd changed
UNKNOWN PTP Property d1c7 changed
UNKNOWN PTP Property d199 changed
UNKNOWN PTP Property d138 changed
UNKNOWN PTP Property d139 changed
UNKNOWN PTP Property d13a changed
UNKNOWN PTP Property d13c changed
UNKNOWN PTP Property d13d changed
UNKNOWN PTP Property d14d changed
UNKNOWN PTP Property d19a changed
UNKNOWN PTP Property d19c changed
UNKNOWN PTP Property d17c changed
UNKNOWN PTP Property d17d changed
UNKNOWN PTP Property d17e changed
UNKNOWN PTP Property d1c6 changed
UNKNOWN PTP Property d1c8 changed
UNKNOWN PTP Property d178 changed
UNKNOWN PTP Property d179 changed
UNKNOWN PTP Property d17a changed
UNKNOWN PTP Property d17b changed
UNKNOWN PTP Property d1c9 changed
UNKNOWN Button 1
UNKNOWN PTP Property d102 changed
UNKNOWN PTP Property d101 changed
UNKNOWN PTP Property d103 changed
UNKNOWN OLCInfo event 0x0010 content 04011723
UNKNOWN OLCInfo event 0x0020 content 000000000000
UNKNOWN OLCInfo exposure indicator 1,1,0.0 (00000000)
UNKNOWN OLCInfo event 0x0080 content 00000000
UNKNOWN OLCInfo event 0x0400 content 00000000000000
UNKNOWN OLCInfo event 0x0800 content 0000000000080000
UNKNOWN OLCInfo event 0x1000 content 00
UNKNOWN OLCInfo event mask=1fff
UNKNOWN PTP Property d115 changed
UNKNOWN PTP Property d1d0 changed
UNKNOWN PTP Property d1d1 changed
UNKNOWN PTP Property d1af changed
UNKNOWN PTP Property d1d9 changed
UNKNOWN PTP Property d1d5 changed
UNKNOWN PTP Property d1d9 changed
UNKNOWN PTP Property d1d5 changed
UNKNOWN PTP Property d11b changed
UNKNOWN PTP Property d11c changed
UNKNOWN PTP Property d11b changed
UNKNOWN PTP Property d1d9 changed
UNKNOWN PTP Property d1d5 changed
UNKNOWN PTP Property d1d9 changed
UNKNOWN PTP Property d1d5 changed
UNKNOWN PTP Property d1d9 changed
UNKNOWN PTP Property d1d5 changed
UNKNOWN PTP Property d1d9 changed
UNKNOWN PTP Property d1d5 changed
UNKNOWN PTP Property d11b changed
UNKNOWN PTP Property d11c changed
UNKNOWN PTP Property d11b changed
UNKNOWN PTP Property d11c changed
UNKNOWN PTP Property d11b changed
UNKNOWN PTP Property d11b changed
UNKNOWN PTP Property d1d9 changed
UNKNOWN PTP Property d1d5 changed
UNKNOWN PTP Property d1d9 changed
UNKNOWN PTP Property d1d5 changed
UNKNOWN PTP Property d1d9 changed
UNKNOWN PTP Property d1d5 changed
UNKNOWN PTP Property d1d9 changed
UNKNOWN PTP Property d1d5 changed
UNKNOWN PTP Property d1b0 changed
UNKNOWN PTP Property d1b0 changed
UNKNOWN PTP Property d1d9 changed
UNKNOWN PTP Property d1d5 changed
UNKNOWN PTP Property d1d9 changed
UNKNOWN PTP Property d1d5 changed
UNKNOWN PTP Property d1d9 changed
UNKNOWN PTP Property d1d5 changed
UNKNOWN PTP Property d1d9 changed
UNKNOWN PTP Property d1d5 changed
UNKNOWN PTP Property d1d9 changed
UNKNOWN PTP Property d1d5 changed
UNKNOWN PTP Property d1d9 changed
UNKNOWN PTP Property d1d5 changed

*** Error ***
The property 'Recording Destination' / 0xd1b8 was not set (0x2019: PTP Device Busy).
Waiting for 2 seconds for events from camera. Press Ctrl-C to abort.
UNKNOWN PTP Property d1d9 changed
UNKNOWN PTP Property d1d5 changed
UNKNOWN PTP Property d1d9 changed
UNKNOWN PTP Property d1d5 changed
UNKNOWN PTP Property d1d9 changed
UNKNOWN PTP Property d1d5 changed
UNKNOWN PTP Property d1d9 changed
UNKNOWN PTP Property d1d5 changed

========== END OF LOG ==========

The camera physical switch is set to recording mode, not photo mode. I know that recording functionality is possible because I've tested it with qDslrDashboard on Windows.

Do you know what could cause this?
As always, if you need more info, please let me know and I'll provide it. I can also provide wireshark capture of what qDslrDashboard does, although the capture is polluted by livestream data.

Thanks,
Nebojsa

msmeissn, 2017-02-23:
busy means we might need to something before, or change ordering or similar :/

can you capture a trace of what qdslrdasboard does? that might help

msmeissn, 2017-02-23:
One thing the qdslrdashboard seems to do is to set the UI Lock

can you put 
--set-config uilock=1   
at the begin of the parameters?  and --set-config uilock=0   at the end

jhampton, 2017-07-14:
Bumping this thread:

We were having similar issues with a 7D.  It turns out that you need to run a --set-config capturetarget=<WHATEVER IS THE CARD> otherwise it will default to RAM even though the moiecapturetarget is set to Card.

To get the values for capturetarget, run

```
gphoto2 --get-config capturetarget
```

and it will list the values.

HTH!

msmeissn, 2017-07-15:
if that helps also for 650D it would be great. I have added a note to the documentation.

nmrmak, 2017-07-15:
I will check this in a few hours and get back to you. I'm still interested
in the solution.

Nebojsa

nmrmak, 2017-07-15:
Hi,

Even with setting the uilock and capturetarget, I still get the same issue, PTP errorcode 0x2019.

Here is the full output along with the exact command line I used:

root@hostname:~# gphoto2 --set-config=uilock=1 --set-config=capturetarget=1 --set-config=viewfinder=1 --set-config=movierecord=0 --wait-event=10s --set-config movierecord=1 --set-config uilock=0 --set-config=capturetarget=0 --wait-event-and-download=2s

*** Error ***
The viewfinder widget is not configurable.

*** Error ***
The property 'Movie Recording' / 0xd1b8 was not set, PTP errorcode 0x2019.

*** Error ***
Failed to set new configuration value 0 for configuration entry movierecord.
Waiting for 10 seconds for events from camera. Press Ctrl-C to abort.
UNKNOWN PTP Property d1b0 changed
UNKNOWN PTP Property d1b0 changed
UNKNOWN PTP Property d105 changed
UNKNOWN PTP Property d108 changed
UNKNOWN PTP Property d106 changed
UNKNOWN PTP Property d107 changed
UNKNOWN PTP Property d109 changed
UNKNOWN PTP Property d10a changed
UNKNOWN PTP Property d10b changed
UNKNOWN PTP Property d10c changed
UNKNOWN PTP Property d10d changed
UNKNOWN PTP Property d10e changed
UNKNOWN PTP Property d10f changed
UNKNOWN PTP Property d11b changed
UNKNOWN PTP Property d114 changed
UNKNOWN PTP Property d116 changed
UNKNOWN PTP Property d119 changed
UNKNOWN PTP Property d110 changed
UNKNOWN PTP Property d101 changed
UNKNOWN PTP Property d102 changed
UNKNOWN PTP Property d103 changed
UNKNOWN PTP Property d104 changed
UNKNOWN PTP Property d11d changed
UNKNOWN PTP Property d111 changed
UNKNOWN PTP Property d112 changed
UNKNOWN PTP Property d113 changed
UNKNOWN PTP Property d120 changed
UNKNOWN PTP Property d122 changed
UNKNOWN PTP Property d156 changed
UNKNOWN PTP Property d150 changed
UNKNOWN PTP Property d151 changed
UNKNOWN PTP Property d152 changed
UNKNOWN PTP Property d153 changed
UNKNOWN PTP Property d154 changed
UNKNOWN PTP Property d155 changed
UNKNOWN PTP Property d160 changed
UNKNOWN PTP Property d161 changed
UNKNOWN PTP Property d162 changed
UNKNOWN PTP Property d11c changed
UNKNOWN PTP Property d1a0 changed
UNKNOWN PTP Property d1a1 changed
UNKNOWN PTP Property d1a8 changed
UNKNOWN PTP Property d1a7 changed
UNKNOWN PTP Property d1ab changed
UNKNOWN PTP Property d1b0 changed
UNKNOWN PTP Property d1b1 changed
UNKNOWN PTP Property d1b2 changed
UNKNOWN PTP Property d1b3 changed
UNKNOWN PTP Property d1b4 changed
UNKNOWN PTP Property d1b5 changed
UNKNOWN PTP Property d1b6 changed
UNKNOWN PTP Property d1a9 changed
UNKNOWN PTP Property d146 changed
UNKNOWN PTP Property d1ac changed
UNKNOWN PTP Property d1aa changed
UNKNOWN PTP Property d11e changed
UNKNOWN PTP Property d11f changed
UNKNOWN PTP Property d1d9 changed
UNKNOWN PTP Property d1ba changed
UNKNOWN PTP Property d1cc changed
UNKNOWN PTP Property d1bc changed
UNKNOWN PTP Property d1b8 changed
UNKNOWN PTP Property d1d3 changed
UNKNOWN PTP Property d1d8 changed
UNKNOWN PTP Property d1b7 changed
UNKNOWN PTP Property d1cb changed
UNKNOWN PTP Property d1db changed
UNKNOWN PTP Property d1dc changed
UNKNOWN PTP Property d1a4 changed
UNKNOWN PTP Property d1df changed
UNKNOWN PTP Property d1bd changed
UNKNOWN PTP Property d1c1 changed
UNKNOWN PTP Property d1c0 changed
UNKNOWN PTP Property d1bf changed
UNKNOWN PTP Property d1c4 changed
UNKNOWN PTP Property d1c2 changed
UNKNOWN PTP Property d1c5 changed
UNKNOWN PTP Property d194 changed
UNKNOWN PTP Property d195 changed
UNKNOWN PTP Property d196 changed
UNKNOWN PTP Property d197 changed
UNKNOWN PTP Property d198 changed
UNKNOWN PTP Property d1dd changed
UNKNOWN PTP Property d1c7 changed
UNKNOWN PTP Property d199 changed
UNKNOWN PTP Property d138 changed
UNKNOWN PTP Property d139 changed
UNKNOWN PTP Property d13a changed
UNKNOWN PTP Property d13c changed
UNKNOWN PTP Property d13d changed
UNKNOWN PTP Property d14d changed
UNKNOWN PTP Property d19a changed
UNKNOWN PTP Property d19c changed
UNKNOWN PTP Property d17c changed
UNKNOWN PTP Property d17d changed
UNKNOWN PTP Property d17e changed
UNKNOWN PTP Property d1c6 changed
UNKNOWN PTP Property d1c8 changed
UNKNOWN PTP Property d178 changed
UNKNOWN PTP Property d179 changed
UNKNOWN PTP Property d17a changed
UNKNOWN PTP Property d17b changed
UNKNOWN PTP Property d1c9 changed
UNKNOWN PTP Property d1d9 changed
UNKNOWN PTP Property d1d5 changed
UNKNOWN PTP Property d1d9 changed
UNKNOWN PTP Property d1d5 changed
UNKNOWN PTP Property d115 changed
UNKNOWN PTP Property d1d0 changed
UNKNOWN PTP Property d1d1 changed
UNKNOWN PTP Property d1af changed

*** Error ***
The property 'Movie Recording' / 0xd1b8 was not set, PTP errorcode 0x2019.

*** Error ***
Failed to set new configuration value 1 for configuration entry movierecord.
Waiting for 2 seconds for events from camera. Press Ctrl-C to abort.
UNKNOWN PTP Property d1b0 changed
UNKNOWN PTP Property d1b0 changed

nmrmak, 2017-07-15:
To anticipate the question:

gphoto2 2.5.14.1
[...]
This version of gphoto2 is using the following software versions and options:
gphoto2         2.5.14.1       gcc, popt(m), exif, no cdk, no aa, jpeg, no readline
libgphoto2      2.5.14.1       all camlibs, gcc, ltdl, EXIF
libgphoto2_port 0.12.0         gcc, ltdl, USB, serial without locking

tobiasBora, 2017-09-09:
Hello,

I've a similar issue, and I cannot find a solution:

    $ LANG=C gphoto2 --set-config viewfinder=1 --set-config movierecordtarget=Card --set-config capturetarget=1 --wait-event=5s --set-config movierecordtarget=None --set-config capturetarget=1 --wait-event-and-download=2s
                                                                                   
    *** Error ***              
    The property 'Recording Destination' / 0xd1b8 was not set (0x2019: PTP Device Busy).
    Waiting for 5 seconds for events from camera. Press Ctrl-C to abort.
    UNKNOWN PTP Property d105 changed
    UNKNOWN PTP Property d108 changed
    UNKNOWN PTP Property d106 changed
    UNKNOWN PTP Property d107 changed
    UNKNOWN PTP Property d109 changed
    UNKNOWN PTP Property d10a changed
    UNKNOWN PTP Property d10b changed
    UNKNOWN PTP Property d10c changed
    UNKNOWN PTP Property d10d changed
    UNKNOWN PTP Property d10e changed
    UNKNOWN PTP Property d10f changed
    UNKNOWN PTP Property d11b changed
    UNKNOWN PTP Property d114 changed
    UNKNOWN PTP Property d116 changed
    UNKNOWN PTP Property d119 changed
    UNKNOWN PTP Property d110 changed
    UNKNOWN PTP Property d101 changed
    UNKNOWN PTP Property d102 changed
    UNKNOWN PTP Property d103 changed
    UNKNOWN PTP Property d104 changed
    UNKNOWN PTP Property d11d changed
    UNKNOWN PTP Property d111 changed
    UNKNOWN PTP Property d112 changed
    UNKNOWN PTP Property d113 changed
    UNKNOWN PTP Property d120 changed
    UNKNOWN PTP Property d122 changed
    UNKNOWN PTP Property d150 changed
    UNKNOWN PTP Property d151 changed
    UNKNOWN PTP Property d152 changed
    UNKNOWN PTP Property d153 changed
    UNKNOWN PTP Property d154 changed
    UNKNOWN PTP Property d155 changed
    UNKNOWN PTP Property d160 changed
    UNKNOWN PTP Property d161 changed
    UNKNOWN PTP Property d162 changed
    UNKNOWN PTP Property d11c changed
    UNKNOWN PTP Property d1a0 changed
    UNKNOWN PTP Property d1a0 changed
    UNKNOWN PTP Property d1a0 changed
    UNKNOWN PTP Property d1a0 changed
    UNKNOWN PTP Property d1a1 changed
    UNKNOWN PTP Property d1a8 changed
    UNKNOWN PTP Property d1ab changed
    UNKNOWN PTP Property d1b0 changed
    UNKNOWN PTP Property d1b1 changed
    UNKNOWN PTP Property d1b2 changed
    UNKNOWN PTP Property d1b3 changed
    UNKNOWN PTP Property d1b4 changed
    UNKNOWN PTP Property d1b5 changed
    UNKNOWN PTP Property d1b6 changed
    UNKNOWN PTP Property d1a9 changed
    UNKNOWN PTP Property d145 changed
    UNKNOWN PTP Property d1ac changed
    UNKNOWN PTP Property d11e changed
    UNKNOWN PTP Property d11f changed
    UNKNOWN PTP Property d1d9 changed
    UNKNOWN PTP Property d1ba changed
    UNKNOWN PTP Property d1be changed
    UNKNOWN PTP Property d1bc changed
    UNKNOWN PTP Property d1b8 changed
    UNKNOWN PTP Property d1d3 changed
[... 137 lines cut ...]
    7.318603 _cb_irq                     (2): 0x561da49fb340 with status 3
    7.318610 _cb_irq                     (2): 0x561da49fb570 with status 3
    7.318616 _cb_irq                     (2): 0x561da49fb7a0 with status 3
    7.318622 _cb_irq                     (2): 0x561da49fb9d0 with status 3
    7.318628 _cb_irq                     (2): 0x561da49fbc00 with status 3
    7.318854 gp_filesystem_reset         (2): resetting filesystem
    7.318860 gp_filesystem_lru_clear     (2): Clearing fscache LRU list...
    7.318865 gp_filesystem_lru_clear     (2): fscache LRU list already empty
    7.318869 delete_all_folders          (2): Internally deleting all folders from '/'...
    7.318874 lookup_folder               (2): Lookup folder '/'...
    7.318879 lookup_folder               (2): Found! / is 0x561da49b9420
    7.318883 recurse_delete_folder       (2): Recurse delete folder 0x561da49b9420//
    7.318888 gp_port_free                (2): Freeing port...
    7.318892 gp_port_close               (2): Closing port...
    7.319026 gp_filesystem_reset         (2): resetting filesystem
    7.319032 gp_filesystem_lru_clear     (2): Clearing fscache LRU list...
    7.319034 gp_filesystem_lru_clear     (2): fscache LRU list already empty
    7.319039 delete_all_folders          (2): Internally deleting all folders from '/'...
    7.319044 lookup_folder               (2): Lookup folder '/'...
    7.319048 lookup_folder               (2): Found! / is 0x561da49b9420
    7.319053 recurse_delete_folder       (2): Recurse delete folder 0x561da49b9420//
    
Any idea?

Thank you!

Rensodiaz, 2019-08-01:
I'm having this same issue anyone have an idea of what's going on, thank you.

toaarnio, 2019-08-20:
I had the same problem on a Canon 5D Mk IV. Got it working with this command line:

```
gphoto2 \
    --set-config viewfinder=1 \
    --set-config capturetarget=Card \
    --set-config movierecordtarget=Card \
    --wait-event 10s \
    --set-config movierecordtarget=None \
    --wait-event-and-download 2s \
    --set-config viewfinder=0 \
```

The ordering is important. Before starting to record (`movierecordtarget=Card`), the viewfinder needs to be enabled and `capturetarget` set to `Card`.

picheli20, 2023-01-16:
I'm having the same problem, is any solid solution for this issue? I've been playing with the answer above, but nothing actually helped me.

Main issue is that I can't defined the `Recording Destination` for movies (for pictures works fine)
```
*** Error ***              
The property 'Recording Destination' / 0xd1b8 was not set (0x2019: PTP Device Busy).
```

msmeissn, 2023-01-17:
which camera? 
Make sure it is in a Dial setting that allows liveview also on the camera (e.g. some EOS do not do this in Auto mode)

there is no solid solution i fear.

inversion, 2023-09-25:
> I had the same problem on a Canon 5D Mk IV. Got it working with this command line:
> 
> ```
> gphoto2 \
>     --set-config viewfinder=1 \
>     --set-config capturetarget='Memory card' \
>     --set-config movierecordtarget=Card \
>     --wait-event 10s \
>     --set-config movierecordtarget=None \
>     --wait-event-and-download 2s \
>     --set-config viewfinder=0 \
> ```
> 
> The ordering and the exact spelling of the targets are important. Before starting to record (`movierecordtarget=Card`), the viewfinder needs to be enabled and `capturetarget` set to "Memory card" (not "Card"). You can use `--get-config capturetarget` and `--get-config movierecordtarget` to figure out the exact names of the targets, in case there are differences between camera models.

This also worked for me on the 5D MKIII (used the ordinals for record target):
`gphoto2 --set-config viewfinder=1 --set-config capturetarget=1 --set-config movierecordtarget=0 --wait-event=10s --set-config movierecordtarget=1 --wait-event-and-download=2s --set-config viewfinder=0`

Thanks @toaarnio !