gphoto/libgphoto2 #136: Livestream slow on Canon EOS 650D

nmrmak, 2017-01-25:
Hi Marcus,

I made a simple application that tries to get livestream frames from the camera. I am basically doing nothing but calling gp_camera_capture_preview() in a loop and then using gp_file_get_data_and_size() to extract JPEG data, but I'm not performing any other operations such as displaying or saving the data anywhere.

Using this setup, I get around 10 frames per second from the camera (on Nikon, the same setup gives around 24 fps). 

This is a bit slow, and I've tested with qDslrDashboard on Windows and with it I get a percievably higher framerate (wasn't able to measure it exactly, but it's much higher than 10 fps). 

Am I doing this right, and is there a way to make this faster?

I've done some profiling and the only call that blocks for a significant amount of time is gp_camera_capture_preview().

Thanks,
Nebojsa

rlamarche, 2017-02-05:
Hi,

What is the resolution of the jpeg preview file ?
On a Nikon (D800), I get 40 fps in photo mode, and up to 60fps in video mode.

msmeissn, 2017-02-11:
--capture-movie also uses capture_preview in a loop, so this is consistent. There might be an option where we could select the preview size.

nmrmak, 2017-07-16:
Possibly related to #185

jcward, 2020-03-26:
My 5D Mark II generates preview images at 1024x680 (which is an ok res), and I'm piping it to FFMPEG with:

```
gphoto2 --stdout --capture-movie | ffmpeg -i - -vcodec rawvideo -pix_fmt yuv420p -threads 0 -f v4l2 /dev/video0
```

I can also specify the ffmpeg framerate with `-filter:v fps=fps=30`, though it seems to default to 24, but that's just the output fps. It's hard to tell exactly how fast it's coming in -- it looks reasonable for sitting and speaking. Maybe 15-20 fps? But I wonder if I could make it go faster. There also seems to be some slight tearing, any ideas for that?

faern, 2021-02-16:
Any news on how to get higher FPS from the 650D using gphoto2? I get around ~8-9 fps using
```
gphoto2 --stdout --capture-movie | ffmpeg -i - -vcodec rawvideo -pix_fmt yuv420p -threads 0 -f v4l2 /dev/video10
```
The image quality is good :+1:, but the FPS is way too low for a usable stream :-1:

I have magic lantern on mine if that can help in any way. Previous posts (and in the other thread) seem to suggest that the preview image transmitted is way too large compared to other brands. Any camera setting that can lower the resolution of it to make it go faster? :thinking:

luxpir, 2021-03-03:
Interested in any solutions too, subscribing here. I'm also getting 10fps on a Canon EOS 1200D. The fps filter command given by jcward above gave no change, but if there's an update to gphoto2 that somehow allows for better frame rates hopefully it can be posted here.