gphoto/libgphoto2 #138: Nikon Keymission 360 failure with capture-image-and-download

esavolle, 2017-02-06:
Hi, 

Got my hands on a Nikon Keymission 360 (Firmware Version 1.2).
Gphoto2 is 2.5.10 and libgphoto is 0.12.0

I've succeeded to get some pictures with --trigger-capture and download it after.
Note that whatever is the value of capturetarget, the pictures are always saved in memory card.

But when I've tried the command capture-image-and-download, there is a failure. Seems that the library is waiting for an event the Keymission is not sending.

Find the log in attachment.

If more tests are needed I can try to get my hands again on the Keymission.

[capture-and-download-keymission360-failure.txt](https://github.com/gphoto/libgphoto2/files/755000/capture-and-download-keymission360-failure.txt)
[trigger-capture-keymission360.txt](https://github.com/gphoto/libgphoto2/files/755001/trigger-capture-keymission360.txt)

msmeissn, 2017-02-07:
Thanks for the report!

there is camera flag which might allow it to work, which I just added in library.c in curren git.
       {"Nikon:KeyMission 360 (PTP mode)",0x04b0, 0x019f, PTP_CAP|PTP_NIKON_BROKEN_CAP},

esavolle, 2017-02-07:
Thanks for the quick reply Marcus.
Do I have to build a new library or can the flag be added in udev rules and/or hwdb ?

I will try as soon as I can and will let you know.

msmeissn, 2017-02-07:
it is a flag in the library, you need to rebuild it

esavolle, 2017-02-10:
Hi Marcus

I have built the library with libusb-0.1.12
But it is still not working.

I had to cancel the operation to exit.

see attachment
[capture-image-and-download-keymission360-failure-with-lib-flag.txt](https://github.com/gphoto/libgphoto2/files/767316/capture-image-and-download-keymission360-failure-with-lib-flag.txt)

msmeissn, 2017-02-10:
hmm. can you try building with libusb1 ? I see an image being added, but a lot of 60 seconds delays :/