gphoto/libgphoto2 #139: libgphoto2 can't detect camera on Theta S (live mode)

tiepviet, 2017-02-07:
When i turn on Live mode on Theta S camera, i can't detect camera port with `gphoto2 --auto-detect`.

msmeissn, 2017-02-07:
is the camera showing on lsusb? 

it might just not be possible to access it when in live mode.

tiepviet, 2017-02-07:
yes, it's shown on lusb. Thank you for answer. Maybe libgphoto2 don't support live mode now.

msmeissn, 2017-02-07:
can you attach lsusb -v   output, the Theta part?

tiepviet, 2017-02-07:
Amazing. You reply soon. That's great! 
```
 RICOH THETA S:

          Product ID: 0x2711
          Vendor ID: 0x05ca  (Ricoh Company Ltd.)
          Version: 0.00
          Serial Number: 00140481
          Speed: Up to 480 Mb/sec
          Manufacturer: Ricoh Company, Ltd.
          Location ID: 0x14200000 / 9
          Current Available (mA): 500
          Current Required (mA): 500
          Extra Operating Current (mA): 0
```

msmeissn, 2017-02-07:
this does not seem to be the "lsusb -v" output... as what kind of device does it show up?
what OS is this?

tiepviet, 2017-02-07:
I'm so sorry. My computer is Macbook with Mac OS. I run `lsusb -v` with theta S live mode.

msmeissn, 2017-02-07:
can you run gphoto2 --auto-detect --debug --debug-logfile=xx.log   and attach xx.log ?

tiepviet, 2017-02-07:
Yes, it's here. 

[debug.txt](https://github.com/gphoto/libgphoto2/files/759360/debug.txt)

tiepviet, 2017-02-14:
Sorry. Can I detect camera list port with OpenCV?

msmeissn, 2017-02-14:
the debug.txt sadly did not have the info i needed. :/ currently no good idea how to show this :/

tiepviet, 2017-02-15:
Thank you very much!

msmeissn, 2017-02-15:
so its probably not gphoto2 usable in this mode. not much to do riught now

closed 2017-02-15 by msmeissn