gphoto/libgphoto2 #140: Trouble capturing using gphoto2 with Nikon D3100

pavelzag, 2017-02-09:
Hi there!
I am trying to use gphoto2 with my Mac and the D3100 and I am getting the following error when running 
gphoto2 --trigger-capture
                                                                               
*** Error ***              
PTP Operation Not Supported
ERROR: Could not trigger capture.
*** Error (-6: 'Unsupported operation') ***    

http://pastebin.com/QJGk0ct8

Thanks!

msmeissn, 2017-02-10:
you trace shows you tried --capture-preview and not --trigger-capture or --capture-image  ?

pavelzag, 2017-02-12:
I'll try the otjher ones as well, thanks!

toxik, 2018-01-04:
Hi, I have similar problems..
Running on OS X High Sierra 10.13.2, gphoto2 installed via homebrew:

➜  ~ gphoto2 --version
gphoto2 2.5.15

Copyright (c) 2000-2017 Lutz Mueller and others

gphoto2 comes with NO WARRANTY, to the extent permitted by law. You may
redistribute copies of gphoto2 under the terms of the GNU General Public
License. For more information about these matters, see the files named COPYING.

This version of gphoto2 is using the following software versions and options:
gphoto2         2.5.15         clang, popt(m), no exif, no cdk, no aa, jpeg, readline
libgphoto2      2.5.16         all camlibs, clang, ltdl, no EXIF
libgphoto2_port 0.12.0         iolibs: disk ptpip serial usb1 usbdiskdirect usbscsi, clang, ltdl, USB, serial without locking

I'm also attaching the output of `gphoto2 --summary --debug --debug-logfile=logfiled3100.txt`
[logfiled3100.txt](https://github.com/gphoto/libgphoto2/files/1602846/logfiled3100.txt)

Any idea why I can't capture images? Thanks!

lucio507986, 2019-02-27:
Hi,

I have the same problem. Any solution yet?

treflip, 2019-04-28:
me too. Nikon D3000

msmeissn, 2019-04-29:
@toxik logfile does not allow GetVendorPropCodes opcode on the D3100. 
@pavelzag does not allow SetControlMode on the D3100.

For the D3100 we "guess" and override opcodes as the camera hides them from us. 

If this is the current firmware, Nikon might have removed them from the camera altogether.
If it is not current firmware, please try to update.

Aroweeri, 2019-05-23:
Experiencing same issue with Nikon D3100, with --capture-movie, --capture-image.
Error is same "PTP Operation Not Supported". 

Firmware on camera has been updated last month, so it's current.

Output of gphoto2 --capture-image --stdout --debug is at https://pastebin.com/xyqSyYKg

I see some lines of interest which lead me to believe it's the same SetControlMode thing that pavelzag was dealing with. See lines 420, 425, and PTP error on line 434 of paste.

You mention overriding opcodes. Is there anything I can try or sections of code that are relevant? Or is this sort of a lost cause for the D3100.

Kesiunas, 2019-12-27:
Hello,

I have the same issue. Can not capture image with Nikon D3100.
Using octoprint with raspberry pi4.
Done everything by this tutorial: https://github.com/FormerLurker/Octolapse/wiki/Configuring-an-External-Camera

Output of gphoto2 --capture-image --stdout --debug is at https://pastebin.com/qpg0W965

Can't find a solution to the problem. Maybe someone could help me out? Or there is no chance getting D3100 working with gphoto2?

Aroweeri, 2019-12-27:
@Kesiunas I never did get it working, but I also just don't know what to try.

Kesiunas, 2019-12-27:
Trying to work it out all day with no success. I will give a try with another camera: Nikon D7000 and get back here

Aroweeri, 2019-12-27:
I suspect that will work since it's a higher end model. From what I've read though the D3100 just doesn't support this stuff.

Kesiunas, 2019-12-27:
I would like to confirm that it works with D7000 with no issues.
It seems that it is true what you said about D3100 not supporting gphoto2 though I saw one person in forum getting it to work with older version of gphoto. Maybe they have done some kind of changes and it doesn't support some older cameras any more.

Aroweeri, 2019-12-27:
Can you link that forum? I could try checking out a version before that post was made.

Kesiunas, 2019-12-30:
http://gphoto-software.10949.n7.nabble.com/Nikon-D3100-Can-t-use-capture-image-and-download-td16393.html

Aroweeri, 2019-12-30:
Seems that that user was just looking for the --capture-image-and-download feature. I've just tested it again on my camera and it does work. version 2.5.23. Anything about capturing video continues to fail. Most features must simply not be supported by it.

Kesiunas, 2019-12-30:
I've tried this command and it doesn't work for me. I just need this command. At the moment I'm using D7000. It would be great to switch back to D3100 as D7000 is used for other works.
Can you suggest something to try to get "capture image" working?

Aroweeri, 2019-12-30:
I run lsusb to see that my camera is detected, then gphoto2 --auto-detect to see that gphoto2 recognizes it, and then gphoto2 --capture-image-and-download just works for me

I have a memory card in my camera as well, I notice that it doesn't work if I take it out.

Kesiunas, 2020-01-07:
After some fiddling around I think I figured it out one step closer.
The thing is with DSLR's (I'm totally new to using DSLR cameras) that sometimes it doesn't focus (if object is too dark, autofocus can not get the object in focus with that kind of lens or any other problem that I don't know), when you get it to focus and take picture normally (without connecting it to USB), then it works with SSH commands. So I got my D3100 to take picutures and save them but it still gives error and after that it makes photos and saves them.

`gphoto2 --capture-image`

```
*** Error ***
PTP Operation Not Supported
New file is in location /store_00010001/DCIM/220D3100/DSC_0008.JPG on the camera
```

`gphoto2 --capture-image-and-download
`
```
*** Error ***
PTP Operation Not Supported
New file is in location /store_00010001/DCIM/220D3100/DSC_0007.JPG on the camera
Saving file as DSC_0007.JPG
Deleting file /store_00010001/DCIM/220D3100/DSC_0007.JPG on the camera
```

So Gphoto2 works but I'm not able to get it working with 3Dprinting timelapses plugin for Octopi, continued research there:
https://github.com/FormerLurker/Octolapse/issues/422

jsnguyen, 2020-04-11:
I think I figured out the problem after mucking around a bit. The D3000/D3100 don't seem to support some of the operations that the camera library seems to think they do. 

All I did was look in the debug output and looked for lines with "PTP Operation Not Supported" and commented out the operations that errored out.

Comment out the following lines in the file `libgphoto2/camlibs/ptp2/library.c`:

**(Note the change in indicies and the realloc size)**
```
if ((nikond >= 3000) && (nikond < 3199)) {
  GP_LOG_D("The D3xxx series hides commands from us ... adding all D3000 ones");
  if (!ptp_operation_issupported(&camera->pl->params, PTP_OC_NIKON_GetVendorPropCodes)) {
    C_MEM (di->OperationsSupported = realloc(di->OperationsSupported,sizeof(di->OperationsSupported[0])*(di->OperationsSupported_len + 8)));
    //di->OperationsSupported[di->OperationsSupported_len+0]  = PTP_OC_NIKON_GetVendorPropCodes;
    //di->OperationsSupported[di->OperationsSupported_len+1]  = PTP_OC_NIKON_CheckEvent;
    //di->OperationsSupported[di->OperationsSupported_len+2]  = PTP_OC_NIKON_Capture;
    di->OperationsSupported[di->OperationsSupported_len+0]  = PTP_OC_NIKON_AfDrive;
    //di->OperationsSupported[di->OperationsSupported_len+4]  = PTP_OC_NIKON_SetControlMode;
    di->OperationsSupported[di->OperationsSupported_len+1]  = PTP_OC_NIKON_DeviceReady;
    //di->OperationsSupported[di->OperationsSupported_len+6]  = PTP_OC_NIKON_AfCaptureSDRAM;
    di->OperationsSupported[di->OperationsSupported_len+2]  = PTP_OC_NIKON_DelImageSDRAM;

    di->OperationsSupported[di->OperationsSupported_len+3]  = PTP_OC_NIKON_GetPreviewImg;
    //di->OperationsSupported[di->OperationsSupported_len+9]  = PTP_OC_NIKON_StartLiveView;
    //di->OperationsSupported[di->OperationsSupported_len+10] = PTP_OC_NIKON_EndLiveView;
    //di->OperationsSupported[di->OperationsSupported_len+11] = PTP_OC_NIKON_GetLiveViewImg;
    di->OperationsSupported[di->OperationsSupported_len+4] = PTP_OC_NIKON_MfDrive;
    di->OperationsSupported[di->OperationsSupported_len+5] = PTP_OC_NIKON_ChangeAfArea;
    di->OperationsSupported[di->OperationsSupported_len+6] = PTP_OC_NIKON_AfDriveCancel;
    /* only opcodes up to 9206 ... I think. */
    di->OperationsSupported_len += 7;
  }
}
```
After recompiling and reinstalling, running:
```
gphoto2 --camera="Nikon DSC D3000 (PTP mode)" --capture-image
```
no longer produces any errors, and an image is taken. I have not tested the D3100, but the above code seems to affect it as well. Commenting out the above lines has not affected any of the other functionality from what I can tell.

msmeissn, 2020-04-12:
did you test this for both D3000 and D3100?

Do the cameras have the latest firmware?

jsnguyen, 2020-04-13:
I tested this for the D3000 successfully, and it seems Nikon never made updated firmware for the cameras. I do not own a D3100, unfortunately.

Holyshititsjustin, 2020-04-13:
I am new and rather confused. How would I recompile this? using putty or with notepad++?

jsnguyen, 2020-04-13:
Download one of the latest releases or git clone. Then change the file in the way that I showed above. After that follow all of the instructions in the `INSTALL` file to reinstall the library. 

You can use notepad++ (or any other text editor) to modify `libgphoto2/camlibs/ptp2/library.c`.

msmeissn, 2020-04-14:
I have applied the above change ... the d3000 and d3100 no longer override to have the special capture modes of nikon.

lucio507986, 2020-04-15:
I have also applied the changes in libgphoto2/camlibs/ptp2/library.c and compiled it again. I have a D3100 and it works perfect for capturing images. Thanks a lot jsnguyen. Nonetheless I tried to capture the LiveView but I get:

Capturing preview frames as movie to 'stdout'. Press Ctrl-C to abort.

*** Error ***              
Sorry, your Nikon camera does not support LiveView mode

I think maybe the Nikon D3100 just does not support this feature. Should I give up?

jsnguyen, 2020-04-16:
Hmm, the D3000 does not support Live View while the D3100 does. Try uncommenting the lines with `PTP_OC_NIKON_StartLiveView`, `PTP_OC_NIKON_EndLiveView`, and `PTP_OC_NIKON_GetLiveViewImg` and recompiling.

There might have to be a separate `if` statement for the D3100.

I unfortunately do not have a D3100 to test this myself.

msmeissn, 2020-04-16:
I split the sections in git and added liveview to the D3100 section.

if you can update libgphoto2 to current master and test it, we could find out.

alternatively you can try:

gphoto2 --set-config opcode=0x9201

and report the output of tghis command, to see if the opcode is supported at all.

msmeissn, 2020-04-16:
Can you perhaps also run

gphoto2 --set-config opcode=0x90ca --debug --debug-logfile=vendorcodes.debug.txt
and attach vendorcodes.debug.txt

Aroweeri, 2020-04-16:
`gphoto2 --set-config opcode=0x90ca --debug --debug-logfile=vendorcodes.debug.txt`

[vendorcodes.debug.txt](https://github.com/gphoto/libgphoto2/files/4487542/vendorcodes.debug.txt)

`gphoto2 --set-config opcode=0x9201 --debug --debug-logfile=vendorcodes9201.debug.txt`

[vendorcodes9201.debug.txt](https://github.com/gphoto/libgphoto2/files/4487550/vendorcodes9201.debug.txt)

Both commands are accompanied by this output:
```
*** Error ***              
Failed to set new configuration value 0x9201 for configuration entry opcode.
*** Error (-6: 'Unsupported operation') *** 
```

msmeissn, 2020-04-17:
Thanks!

D3100: 0x90ca PTP Operation Not Supported (0x2005)
D3100: 0x9201 PTP Operation Not Supported (0x2005) (Start LiveView)

So liveview is also not in the "hidden" commandset on D3100.

msmeissn, 2020-04-18:
I have adjusted also the mark ..

- d3000 and d3100 do not support liveview capture.

Sorry for the incorrect overrides up to now.

closed 2020-04-18 by msmeissn

andrealmeid, 2020-10-20:
> I have adjusted also the mark ..
> 
>     * d3000 and d3100 do not support liveview capture.
> 
> 
> Sorry for the incorrect overrides up to now.

hi @msmeissn, is this a hardware/firmware limitation or is something that could be implemented in gphoto?

msmeissn, 2020-10-20:
it is a camera firmware side limitation

andrealmeid, 2020-10-20:
@msmeissn thanks!

cbaii, 2021-01-08:
Could someone help me edit the file like jsnguyen did? I believe I got into the file in notepad++ however it looks very different than what his looks like. Half of it is no longer there including the parts that  he commented out. Thanks

jsnguyen, 2021-01-09:
@cbaii The change just removed all of the commands which are no longer relevant since the D3100 and D3000 do not support them. You should not have to edit the file at all anymore, and it should work.

Double check the version of libgphoto2 that you're using. If you're still having the error in this thread it is likely old. Do a fresh git clone to get the latest version and compile from source if necessary.

cbaii, 2021-01-09:
@jsnguyen Thank you for replying so fast, I figured out that I needed to download the latest development version instead of the last stable version.

Stephan5097, 2021-08-25:
*** Error ***              
An error occurred in the io-library ('Could not claim the USB device'): Could not claim interface 0 (Device or resource busy). Make sure no other program (gvfs-gphoto2-volume-monitor) or kernel module (such as sdc2xx, stv680, spca50x) is using the device and you have read/write access to the device.
ERROR: Could not capture image.
ERROR: Could not capture.
*** Error (-53: 'Could not claim the USB device') ***       

For debugging messages, please use the --debug option.
Debugging messages may help finding a solution to your problem.
If you intend to send any error or debug messages to the gphoto
developer mailing list <gphoto-devel@lists.sourceforge.net>, please run
gphoto2 as follows:

    env LANG=C gphoto2 --debug --debug-logfile=my-logfile.txt --capture-image

Please make sure there is sufficient quoting around the arguments.

Need some help please

mkoetzing, 2021-09-29:
> Download one of the latest releases or git clone. Then change the file in the way that I showed above. After that follow all of the instructions in the `INSTALL` file to reinstall the library.
> 
> You can use notepad++ (or any other text editor) to modify `libgphoto2/camlibs/ptp2/library.c`.

if you are running it on an raspberry pi you can easily run following script for updating without building from source: 
https://github.com/gonzalo/gphoto2-updater