gphoto/libgphoto2 #141: Problems with Nikon Coolpix A Image Capture

Smau4, 2017-02-16:
Hello,

I have a Nikon Coolpix A connected to my Raspberry Pi.  The gphoto2 is version 2.5.11, libgphoto2 version 2.5.4.  I want to run the capture-image-and-download command, but I
keep getting the following error:

*** Error ***              
Sorry, your camera does not support generic capture
ERROR: Could not capture image.
ERROR: Could not capture.
*** Error (-6: 'Unsupported operation') ***   

I checked the camera summary, image capture is not supported (I've also attached the debug log file in
[logfile.txt](https://github.com/gphoto/libgphoto2/files/778886/logfile.txt)).  However, on the gphoto2 supported camera page, the Nikon Coolpix A can capture images.   I also read that there was a way to manually override opcodes in version 2.5.8, but I am unsure of how to do this.

Thanks,
Stephen

msmeissn, 2017-02-16:
you wil need at least libgphoto2 2.5.8 , i added Coolpix A capture support there

Smau4, 2017-02-17:
I updated libgphoto2 to version 2.5.11.  It's partially working now.  The capture-preview command works fine, as does trigger capture.  However, when I try the capture-image command, I get a new error:

*** Error ***              
Not in Liveview
ERROR: Could not capture image.
ERROR: Could not capture.

Debug file: [log.txt](https://github.com/gphoto/libgphoto2/files/784349/log.txt)

I've tried turning on the viewfinder using the command "gphoto2 --set-config viewfinder=On", but to no avail.  I feel like I'm missing something really obvious here.

msmeissn, 2017-02-20:
i added a patch ... can you try building? perhaps it helps. if not, a logfile would be useful again

msmeissn, 2017-09-06:
no reply so far ... lets hope its fixed. feel free to reopen if not.

skypuppy, 2018-05-19:
Not fixed.  :(  Just got an A900.  While the camera emits a click when you tell it to capture. the lens "protector" still does not open, so the pics are always just black.  Help!

msmeissn, 2018-05-21:
can you capture debug output?

add --debug --debug-logfile=debug.log    to the commandline

skypuppy, 2018-05-21:
Learning this phone email, dang it.  Ok, I kept experimenting and tried every reasonable command in the list  (manual is 240+ pages wow).  Finally found a shell command that worked for getting pics.  Woohoo.  However the lens stay extended and the shots take a second or more.  Wah.  Will have to experiment more to find commands that recess the lens and close the cover when that series of shots are complete, then repeat it all a few seconds later when the next target comes into view -- all by Rip command, while saving every shot taken.  Can't wait to get into zoom/retract issues!  :(  thanks

⁣Sent from BlueMail ​

skypuppy, 2018-05-21:
Will try.

⁣Sent from BlueMail ​

skypuppy, 2018-05-21:
"Rip" should be "RPi."  Dang autocorrect.

⁣Sent from BlueMail ​

skypuppy, 2018-06-12:
msmeissn,
Here is the debug log for the Nikon A900, assuming I can figure out how to upload it!  :)

[debug.log](https://github.com/gphoto/libgphoto2/files/2094269/debug.log)

msmeissn, 2018-06-14:
This one has to be explicitly listed, which I just did for git. you likel yneed git state or the next release when it comes out to make it work

msmeissn, 2018-06-17:
can you manually protrude the lens before? or is it motor driven? (dont remember from the a5000 testing) i dont remember this as prolem

that debian has issues is weird. :/

dlwalter, 2019-11-06:
I'm having the same problem - ubuntu 16.04 and macOS.  Camera capture runs fine but lens cover does not open and lens does not extend.  

I just get blank pictures of the back of the lens cover.