gphoto/libgphoto2 #143: Timeout on wait_for_event

Kev-J, 2017-03-15:
Hi,

I have a custom camera accessed by libgphoto2 through ptp2 camlib.
The gp_camera_wait_for_event() function does not wait timeout expiration and return setting the event type to GP_EVENT_TIMEOUT and throwing this kind of error : 

[ptp_usb_event [usb.c:537]]Reading PTP event failed: Timeout reading from or writing to the port (-10) (level 0)

After reading the code, I figured out that the timeout value was not used in my case (e.g. without a Canon or Nikon or Sony camera). The call to ptp_get_one_event() in camlibs/ptp2/library.c:L5008 fail and the function return GP_OK with a GP_EVENT_TIMEOUT event type.

IMO, the expected behaviour would be to wait for the event during the timeout.

Cheers,
Kevin

msmeissn, 2017-03-16:
C_PTP_REP (ptp_check_event(params));
does the checking, but you are correct. it only does a short timeout wait and not a long one.

Kev-J, 2017-03-16:
I submitted a pull request. I tested my modifications and it is working like a charm.

msmeissn, 2017-03-18:
Thanks! i accepted the pull request and did my changes on top.

can you check if that still does what you want?

closed 2017-03-18 by msmeissn

Kev-J, 2017-03-20:
Perfect! Thanks a lot Marcus!