gphoto/libgphoto2 #145: compiled source slower than package

punkoffice, 2017-03-17:
I've successfully compiled the source code from the bz2 file.  When I run this gphoto2 its very slow to detect cameras.  I'm working with 40 cameras so its so slow its unusable.  However, when I install gphoto2 from apt-get it detects all the cameras instantly.  The latest version from apt-get is 2.5.9.  I tried source code from 2.5.12, 2.5.11, 2.5.10 and also 2.5.9 and I have the same issue with all of them.  Is there any difference in the way the apt-get version is built?

msmeissn, 2017-03-17:
tgere might be a difference. if you capture a debug trace, it will have timestamps in front and you will see larger jumps.

gphoto2 --auto-detect --debug --debug-logfile=log.txt

punkoffice, 2017-03-18:
Ok, funny thing is I ran that line for debug output and now its not slow to detect anymore.  I can no longer replicate this problem

closed 2017-03-18 by punkoffice

msmeissn, 2017-03-18:
It might be a  unrelated device (printer?)  on the usb bus that caused slowdowns.

weird

punkoffice, 2017-03-18:
There is nothing else on the server.  They are built only to control a lot of cameras.  There's not even a monitor or keyboard connected to them.  It could be hardware related.  I have a lot of hubs connected to 2 USB cards.  I have had hubs cause delays before.

msmeissn, 2017-03-18:
yeah, hard to say if not one or more hubs might ause trouble then :/