gphoto/libgphoto2 #146: Timeout with Nikon D5200

punkoffice, 2017-03-18:
I have 40 Nikon D5200 cameras attached to one server.  I'm calling gphoto2 --get-all-files on each camera and all as a separate process which makes it faster (using ampersand in BASH script).  This was working 100% of the time with gphoto2 2.5.9.  I'm now using the latest gphoto2 2.5.11 and libghoto2 2.5.12 and I get timeout errors on some of the cameras.  I've attached the debug output here:
[nikond5200log.txt](https://github.com/gphoto/libgphoto2/files/852152/nikond5200log.txt)
Does anyone know how I can avoid these timeouts?

msmeissn, 2017-03-18:
I looked at it, the communication with the camera does time out during session open.

as the camera got detected, its probably not hardware related.

we try probing all ports though, even though we specified the port explicilty, there is some room for improvent.  if this all runs parallel we might be overloading the usb bus with traffic.

are these always the same cameras or different ones?

can you try to downgrade?

punkoffice, 2017-03-18:
I'm using 40 different cameras together.  The purpose is for photogrammetry.
Yes I can downgrade.  This is not a show-stopper for me.  I'm happy to use 2.5.9 for now.