gphoto/libgphoto2 #148: Add support Canon EOS M10

dhamp, 2017-03-23:
gphoto2 --version
gphoto2 2.5.11.1

Copyright (c) 2000-2016 Lutz Mueller and others

gphoto2 comes with NO WARRANTY, to the extent permitted by law. You may
redistribute copies of gphoto2 under the terms of the GNU General Public
License. For more information about these matters, see the files named COPYING.

This version of gphoto2 is using the following software versions and options:
gphoto2         2.5.11.1       gcc, popt(m), exif, no cdk, aa, jpeg, readline
libgphoto2      2.5.12.1       all camlibs, gcc, ltdl, EXIF
libgphoto2_port 0.12.0         gcc, ltdl, USB, serial without locking

gphoto2 - adb1b27 from git(current master)
libgphoto2 - db534554f from git(current master)

USB ID: 04a9:32a0 Canon, Inc.

summary, config, debug log with try capture:
https://gist.github.com/dhamp/36b68bdb08848cd9217fd3f005d6f59b

msmeissn, 2017-05-05:
(FWIW, EOS m10 is now supported in master, I bought one and integrated a patch from one other contributor)

Anachron, 2018-05-26:
Since the latest release I cannot use my `Canon EOS M10` anymore.

https://gist.github.com/Anachron/688c2914c0132b6be3e77ac2f342dc8a

```
lsusb | grep Canon
Bus 001 Device 018: ID 04a9:32a0 Canon, Inc.
```

But no udev-rules for it in `/usr/lib/udev/rules.d/40-gphoto.rules`.

argosvr, 2018-08-13:
it is still not working ! we tested it and there is a real problem with the EOS M10. What can we do ?

msmeissn, 2018-08-13:
0.068721 gp_libusb1_open [libusb1.c:392](0): 'libusb_open (port->pl->d, &port->pl->dh)' failed: Access denied (insufficient permissions) (-3)

its an access denied, so a permission issue.

you first can try as root and see if it works there. 

what distribution do you use? perhaps contact them to clarify this permission setting?

argosvr, 2018-08-13:
sorry but EOS M10 is not working properly.. you can take pictures but can't change the settings. It is ust not working at that time.

argosvr, 2018-08-13:
we use the last distribution

msmeissn, 2019-02-10:
This is a unresolved issue , somehow settings do not persist longer than the current session on the Canon EOS M