gphoto/libgphoto2 #149: Complete overhaul

milani, 2017-03-25:
I was searching for a PTP library to use in my android app when I came across libgphoto2. Also its large support for DSRLs makes it great, I had a hard time understanding the structure of the project. It seems that over time, some functionalities are moved between different files and some files are now obsolete.

If my understanding is correct, do you think it is necessary or do you have a plan to re-organize the project?

msmeissn, 2017-03-25:
I do not think there is much stuff obsolete.

camlibs/ptp2/ would cover the PTP based cameras and none of the files are obsolete in there :)

msmeissn, 2017-03-25:
it is however optimized for embedding in libgphoto2 and making it standalone again would be hard.

FWIW, qDSLRDashboard is I think the leading Non-Vendor Android App for camera control (it is not libgphoto2 based). Perhaps you can use that already for your purposes.

milani, 2017-03-25:
A standalone library would be great. By the way, I saw qDSLRDashboard and found https://github.com/hubaiz/DslrDashboard (also I think qDSLRdashboard is not opensource and these two are not the same). But libgphoto2 is much more interesting as it is widely used and supports more cameras.