gphoto/libgphoto2 #150: Nikon P900 stopped working after upgrade from 2.5.4 to 2.5.12

spiral27, 2017-03-27:
Hello and thank you very much for gphoto2!

I am using gphoto2 for months with a Nikon P900 and a Raspberry 3. This combo makes a great surveillance camera because of the incredible 2000mm zoom on the P900.

I wanted to improve the focus on pictures so I thought the first step would be to upgrade from 2.5.4 (lastest version installed via apt-get) to 2.5.12 (installed via https://github.com/gonzalo/gphoto2-updater) but I can't take pictures anymore:

```
pi@raspberrypi:~ $ gphoto2 --capture-image
                                                                              
*** Error ***             
PTP Invalid Parameter
ERROR: Could not capture image.
ERROR: Could not capture.
```
I have attached the log file to this thread. Does anyone know what is happening? Thank you very much for any help
[logfile.txt](https://github.com/gphoto/libgphoto2/files/871207/logfile.txt)

msmeissn, 2017-03-27:
I added a small patch to current libgphoto2 git. can you try that? gphoto updater shoyld be ablke to check it out

spiral27, 2017-03-28:
Hello msmeissn and thanks for your help! I just updated via https://github.com/gonzalo/gphoto2-updater, rebooted and tried again:

```
pi@raspberrypi:~ $ gphoto2 --debug --debug-logfile=logfile2.txt --capture-image
ERROR: Could not capture image.                                                
ERROR: Could not capture.
```
Here is the new logfile.
[logfile2.txt](https://github.com/gphoto/libgphoto2/files/876671/logfile2.txt)

Any clue?

Sija, 2017-03-28:
@spiral27 did you use `--development` flag when running `gphoto2-updater` script?

spiral27, 2017-03-29:
@Sija, good point, I forgot that. I now updated with the --development flag and it works. Thank you very much!

elyjason, 2017-06-15:
I can get it to show the camera, but I can't get my Nikon P900 to turn on whenever it is plugged in.  As soon I plug the camera in it goes to mass storage mode.   I can't find any settings on the camera.   How did you connect your camera and pi, how did you get the camera to stay in photo taking mode?

```
gphoto2 --auto-detect
Model                          Port
----------------------------------------------------------
USB PTP Class Camera           usb:001,006
```

msmeissn, 2017-06-15:
@spiral27 thanks for confirming!

@elyjason  ... no idea though :/ also you will need a newer libgphoto2 version

closed 2017-06-15 by msmeissn

elyjason, 2017-06-16:
Thank you for your response.  I was able to get the Nikon P900 to finally turn on by killing the gphoto2 volume monitor service and then running gphoto2 --capture-image-and-download.  That allows the communication between my Pi and the P900.   Now the problem I am troubleshooting is a PTP issue.

` $ gphoto2 --capture-image-and-download --debug
0.000377 main(2): ALWAYS INCLUDE THE FOLLOWING LINES WHEN SENDING DEBUG MESSAGES TO THE MAILING LIST:
0.000482 main(2): gphoto2 2.5.4
0.000546 main(2): gphoto2 has been compiled with the following options:
0.000574 main(2):  + gcc (C compiler used)
0.000610 main(2):  + popt (mandatory, for handling command-line parameters)
0.000637 main(2):  + exif (for displaying EXIF information)
0.000672 main(2):  + cdk (for accessing configuration options)
0.000706 main(2):  + aa (for displaying live previews)
0.000740 main(2):  + jpeg (for displaying live previews in JPEG format)
0.000772 main(2):  + readline (for easy navigation in the shell)
0.000845 main(2): libgphoto2 2.5.4
0.000895 main(2): libgphoto2 has been compiled with the following options:
0.000931 main(2):  + all camlibs
0.000964 main(2):  + gcc (C compiler used)
0.000987 main(2):  + ltdl (for portable loading of camlibs)
0.001005 main(2):  + EXIF (for special handling of EXIF files)
0.001030 main(2): libgphoto2_port 0.10.0
0.001060 main(2): libgphoto2_port has been compiled with the following options:
0.001082 main(2):  + gcc (C compiler used)
0.001099 main(2):  + ltdl (for portable loading of camlibs)
0.001133 main(2):  + USB (libusb1, for USB cameras)
0.001158 main(2):  + serial (for serial cameras)
0.001175 main(2):  + no resmgr (serial port access and locking)
0.001215 main(2):  + no ttylock (serial port locking)
0.001249 main(2):  + no lockdev (serial port locking)
0.001288 main(2): CAMLIBS env var not set, using compile-time default instead
0.001309 main(2): IOLIBS env var not set, using compile-time default instead
0.001327 main(2): invoked with following arguments:
0.001344 main(2):   --capture-image-and-download
0.001360 main(2):   --debug
0.001485 setting/gphoto2-setting.c(2): Creating $HOME/.gphoto
0.001760 setting/gphoto2-setting.c(2): Loading settings from file "/home/pi/.gphoto/settings"
0.002039 main(2): The user has not specified both a model and a port. Try to figure them out.
0.002125 gphoto2-port-info-list(2): Using ltdl to load io-drivers from '/usr/lib/arm-linux-gnueabihf/libgphoto2_port/0.10.0'...
0.002619 gphoto2-port-info-list(2): Called for filename '/usr/lib/arm-linux-gnueabihf/libgphoto2_port/0.10.0/disk'.
0.003749 gphoto2-port/disk(2): found fstab fsname proc
0.003813 gphoto2-port/disk(2): found fstab fsname /dev/mmcblk0p6
0.003904 gphoto2-port/disk(2): found fstab fsname /dev/mmcblk0p7
0.004506 gphoto2-port/disk(2): found mtab fsname /dev/root
0.004617 gphoto2-port/disk(2): found mtab fsname devtmpfs
0.004667 gphoto2-port/disk(2): found mtab fsname sysfs
0.004717 gphoto2-port/disk(2): found mtab fsname proc
0.004769 gphoto2-port/disk(2): found mtab fsname tmpfs
0.004851 gphoto2-port/disk(2): found mtab fsname devpts
0.004896 gphoto2-port/disk(2): found mtab fsname tmpfs
0.004984 gphoto2-port/disk(2): found mtab fsname tmpfs
0.005074 gphoto2-port/disk(2): found mtab fsname tmpfs
0.005185 gphoto2-port/disk(2): found mtab fsname cgroup
0.005217 gphoto2-port/disk(2): found mtab fsname cgroup
0.005249 gphoto2-port/disk(2): found mtab fsname cgroup
0.005278 gphoto2-port/disk(2): found mtab fsname cgroup
0.005313 gphoto2-port/disk(2): found mtab fsname cgroup
0.005505 gphoto2-port/disk(2): found mtab fsname cgroup
0.005549 gphoto2-port/disk(2): found mtab fsname cgroup
0.005578 gphoto2-port/disk(2): found mtab fsname tmpfs
0.005639 gphoto2-port/disk(2): found mtab fsname systemd-1
0.005669 gphoto2-port/disk(2): found mtab fsname mqueue
0.005702 gphoto2-port/disk(2): found mtab fsname debugfs
0.005780 gphoto2-port/disk(2): found mtab fsname fusectl
0.005813 gphoto2-port/disk(2): found mtab fsname configfs
0.005902 gphoto2-port/disk(2): found mtab fsname /dev/mmcblk0p6
0.005976 gphoto2-port/disk(2): found mtab fsname tmpfs
0.006058 gphoto2-port/disk(2): found mtab fsname gvfsd-fuse
0.006089 gphoto2-port/disk(2): found mtab fsname /dev/mmcblk0p5
0.006508 gphoto2-port-info-list(2): Could not load port driver list: 'Unspecified error'.
0.006577 gphoto2-port-info-list(2): Loaded '' ('^disk:') from '/usr/lib/arm-linux-gnueabihf/libgphoto2_port/0.10.0/disk'.
0.006640 gphoto2-port-info-list(2): Called for filename '/usr/lib/arm-linux-gnueabihf/libgphoto2_port/0.10.0/ptpip'.
[... 785 lines cut ...]
5.050280 gphoto2-port(2): Reading 1024=0x400 bytes from port...
5.050303 libusb1(2): reading with timeout 20000
5.201745 libusb1(2): ret = 0
5.201790 gphoto2-port(2): Could only read 12 out of 1024 byte(s)
5.201831 gphoto2-port(3): Hexdump of 12 = 0xc bytes follows:
0000  0c 00 00 00 03 00 01 20-19 00 00 00              ....... ....

5.201928 gphoto2-port(2): Closing port...
5.202670 gphoto2-filesystem(2): resetting filesystem
5.202702 libgphoto2/gphoto2-filesys.c(2): Clearing fscache LRU list...
5.202727 libgphoto2/gphoto2-filesys.c(2): fscache LRU list already empty
5.202751 gphoto2-filesystem(2): Internally deleting all folders from '/'...
5.202775 gphoto2-filesystem(2): Lookup folder '/'...
5.202798 gphoto2-filesystem(2): Found! / is 0xb8ab18
5.202820 gphoto2-filesystem(2): Recurse delete folder 0xb8ab18//
5.202857 gphoto2-port(2): Freeing port...
5.202889 gphoto2-port(2): Closing port...
5.203466 gphoto2-filesystem(2): resetting filesystem
5.203495 libgphoto2/gphoto2-filesys.c(2): Clearing fscache LRU list...
5.203519 libgphoto2/gphoto2-filesys.c(2): fscache LRU list already empty
5.203542 gphoto2-filesystem(2): Internally deleting all folders from '/'...
5.203566 gphoto2-filesystem(2): Lookup folder '/'...
5.203587 gphoto2-filesystem(2): Found! / is 0xb8ab18
5.203609 gphoto2-filesystem(2): Recurse delete folder 0xb8ab18//
`

elyjason, 2017-06-16:
--set-config capturetarget=1 fixed my above error.

msmeissn, 2017-07-15:
nice to hear! yes, we might now support more capture methods thats why that changed here