gphoto/libgphoto2 #151: Canon EOS 550D, 600D: gp_camera_wait_for_event blocks for a long time after shutter

Ruuttu, 2017-03-27:
I'm using the python-gphoto2 interface: https://github.com/jim-easterbrook/python-gphoto2

I have an infinite loop calling gp_camera_wait_for_event with a timeout of a few milliseconds. Soon after capturing a photo with a hardware trigger, the function returns a GP_EVENT_FILE_ADDED event. When this event is produced, the function call first blocks for ~1.2 seconds. Because this event is produced ~0.8 seconds after the shot, I'm basically restricted to a shot-to-shot time of 2 seconds if triggering via code.

So this (paraphrasing):
```
while True:
    time_before = time.time()
    etype, event = self.check_result( gp.gp_camera_wait_for_event( camera, 5, context ) )
    time_after = time.time()
    
    print "Took", time_after - time_before
    
    # Timed out waiting for event
    if etype == gp.GP_EVENT_TIMEOUT:
        continue
    
    # New image
    if etype == gp.GP_EVENT_FILE_ADDED:
        print "FILE ADDED"

```

Would produce this for example:
```
Took 0.0054...
Took 0.0051...
Took 0.0060...
Took 1.2440...
FILE ADDED
Took 0.0055...
```

Here's the camera summary and debug printout:

```
Camera summary:
Manufacturer: Canon Inc.
Model: Canon EOS 600D
  Version: 3-1.0.2
  Serial Number: a02e63eedea843bcb12a1e24e7eeb36
Vendor Extension ID: 0xb (2.0)

Capture Formats: JPEG
Display Formats: Association/Directory, Script, DPOF, MS AVI, MS Wave, JPEG, CRW, Unknown(b103), Unknown(bf02), Defined Type, Unknown(b104), Unknown(b105)

Device Capabilities:
        File Download, File Deletion, File Upload
        No Image Capture, No Open Capture, Canon EOS Capture, Canon EOS Shutter Button

Storage Devices Summary:

Device Property Summary:
Property 0xd402:(read only) (type=0xffff) 'Canon EOS 600D'
Property 0xd407:(read only) (type=0x6) 1
Property 0xd406:(readwrite) (type=0xffff) 'Unknown Initiator'
Property 0xd303:(read only) (type=0x2) 1
Battery Level(0x5001):(read only) (type=0x2) Enumeration [100,0,75,0,50] value: 100% (100)
```

```
0.000071 main                        (2): ALWAYS INCLUDE THE FOLLOWING LINES WHEN SENDING DEBUG MESSAGES TO THE MAILING LIST:
0.000270 main                        (2): gphoto2 2.5.11
0.000333 main                        (2): gphoto2 has been compiled with the following options:
0.000381 main                        (2):  + gcc (C compiler used)
0.000471 main                        (2):  + popt (mandatory, for handling command-line parameters)
0.000587 main                        (2):  + exif (for displaying EXIF information)
0.000664 main                        (2):  + no cdk (for accessing configuration options)
0.000726 main                        (2):  + no aa (for displaying live previews)
0.000802 main                        (2):  + jpeg (for displaying live previews in JPEG format)
0.000857 main                        (2):  + no readline (for easy navigation in the shell)
0.000946 main                        (2): libgphoto2 2.5.12
0.001044 main                        (2): libgphoto2 has been compiled with the following options:
0.001118 main                        (2):  + all camlibs
0.001179 main                        (2):  + gcc (C compiler used)
0.001260 main                        (2):  + ltdl (for portable loading of camlibs)
0.001323 main                        (2):  + EXIF (for special handling of EXIF files)
0.001401 main                        (2): libgphoto2_port 0.12.0
0.001507 main                        (2): libgphoto2_port has been compiled with the following options:
0.001593 main                        (2):  + gcc (C compiler used)
0.001682 main                        (2):  + ltdl (for portable loading of camlibs)
0.001729 main                        (2):  + USB (libusb1, for USB cameras)
0.001799 main                        (2):  + serial (for serial cameras)
0.001879 main                        (2):  + no resmgr (serial port access and locking)
0.001939 main                        (2):  + no ttylock (serial port locking)
0.001987 main                        (2):  + no lockdev (serial port locking)
0.002062 main                        (2): CAMLIBS env var not set, using compile-time default instead
0.002123 main                        (2): IOLIBS env var not set, using compile-time default instead
0.002180 main                        (2): invoked with following arguments:
0.002246 main                        (2):   --debug
0.002386 load_settings               (2): Creating gphoto config directory ('/root/.gphoto')
0.002753 load_settings               (2): Loading settings from file '/root/.gphoto/settings'.
0.003259 gp_camera_free              (2): Freeing camera...
0.003340 gp_port_free                (2): Freeing port...
0.003410 gp_filesystem_reset         (2): resetting filesystem
0.003497 gp_filesystem_lru_clear     (2): Clearing fscache LRU list...
0.003564 gp_filesystem_lru_clear     (2): fscache LRU list already empty
0.003651 delete_all_folders          (2): Internally deleting all folders from '/'...
0.003718 lookup_folder               (2): Lookup folder '/'...
0.003781 lookup_folder               (2): Found! / is 0x38318
0.003867 recurse_delete_folder       (2): Recurse delete folder 0x38318//
```

msmeissn, 2017-03-27:
I think it is in "sdram" capture mode, where it downloads the image from the camera sdram.

so 0.8 for in camera processing and 1.2 seconds for download?

you can change the capturetarget to "card" to avoid the implicit download

Ruuttu, 2017-03-27:
The Canon will happily hold up to 6 RAW images without and SD card, so it's a shame they can't be downloaded more granularly - I believe the camera would be capable of it. 

I am trying to run a rig of 30 Canons and SD cards were not in the plans, so I'd like to avoid that. :) I'll test it anyway.

msmeissn, 2017-03-28:
hmm, this could potentially be implemented in libgphoto2, which so far downloads the SDRAM image as soon as it sees the event in wait_for_event. WOuld need some redesign though :/

Ruuttu, 2017-03-28:
Yeah, that was 1.2 seconds for downloading an 18 megapixel RAW over USB. Not bad at all. 

With capturetarget set to Memory Card, the GP_EVENT_FILE_ADDED event blocks for ~7 milliseconds. 
After that, calling gp_file_get_data_and_size took ~1.1 seconds. I wasn't able to use the gp_camera_file_read before, which I'm assuming would enable the incremental download I'm looking for. Perhaps the SD card would make that work. Will test.

alex73, 2017-06-15:
I also have too big delay on the capture. Does it correspond with this issue ?
See details in the https://sourceforge.net/p/gphoto/mailman/message/35891757/