gphoto/libgphoto2 #153: sony a6300: The adjusted value togle between two values

inquito, 2017-04-03:
Hello again.
when I send to a sony a6300 this command:
photo2 --set-config iso=100  --set-config exposurecompensation=1; gphoto2 --get-config iso | grep Current
I get
Current: 100      (value is ok)
if I repeat the same command I get
Current: Auto ISO   (value is not correct)
if I repeat again I get
Current: 100      (value is ok again)
and this change in adjusted value happens each time than I send that order.

But if I break up the command in two commands...
gphoto2 --set-config iso=100
gphoto2 --set-config exposurecompensation=1
all values are adjusted to the correct ones and that values don't change if I repeat this two commands again.
Please, tell me what kind of log may I send you if you need it to debug.
Best Regards.
Inco.

inquito, 2017-04-03:
hello
doing more test and have seen that its more simple to obtain the toggle adjusted value.

[root@mundin:~]$ gphoto2 --get-config iso | grep "Current"
Current: Auto ISO
[root@mundin:~]$ gphoto2 --set-config iso=100
[root@mundin:~]$ gphoto2 --get-config iso | grep "Current"                     
Current: 100
[root@mundin:~]$ gphoto2 --set-config iso=100
[root@mundin:~]$ gphoto2 --get-config iso | grep "Current"                     
Current: Auto ISO
[root@mundin:~]$ gphoto2 --set-config iso=100
[root@mundin:~]$ gphoto2 --get-config iso | grep "Current"                     
Current: 100
[root@mundin:~]$ gphoto2 --set-config iso=100
[root@mundin:~]$ gphoto2 --get-config iso | grep "Current"                     
Current: Auto ISO

Inco.

msmeissn, 2017-04-23:
I did some fixes for this in libgphoto2 2.5.13, can you try that?

inquito, 2017-04-25:
Hello Marcus
I have been doing a lot of test and your new version works fine, I have not found any strange result.
thank you very much.
inco.
This version of gphoto2 is using the following software versions and options:
gphoto2         2.5.11         gcc, popt(m), exif, no cdk, no aa, no jpeg, no readline
libgphoto2      2.5.13         all camlibs, gcc, ltdl, EXIF
libgphoto2_port 0.12.0         gcc, ltdl, USB, serial without locking

msmeissn, 2017-04-25:
thx for checking back!

closed 2017-04-25 by msmeissn