gphoto/libgphoto2 #154: Need to free camera after download

punkoffice, 2017-04-04:
I download like this:
gp_camera_folder_list_files 
gp_camera_file_get
gp_file_free
gp_camera_file_delete

but if I take more photos with the camera then do this download process again it cannot find the photos on the camera.  gp_camera_folder_list_files tells me there's no photos.  The only way I can get it to refresh is if I run gp_camera_exit then gp_camera_free then create that camera again.  If I just run gp_camera_exit I get realloc segfault when it automatically tries to do a gp_camera_init during list files.  Just wondering if I'm doing something wrong here.  It seems crazy to have to exit and free after every download.

msmeissn, 2017-04-04:
the photos are addded via shutter press?

then try running gp_camera_wait_for_event inbetween

punkoffice, 2017-04-05:
Yes, I'm using shutter press.
I'm playing with gp_camera_wait_for_event.  It works for the first shutter press.  It blocks and waits.  When I run it again it doesn't block anymore.  It completes instantly.  If I disconnect (exit and free) then reconnect to the camera then it works again

msmeissn, 2017-04-05:
The gp_camera_wait_for_event needs to be called more often, as it only returns 1 event. 

use a short timeout and then run in a while loop until it returns GP_EVENT_TIMEOUT,

punkoffice, 2017-04-06:
ok, I got it working better using this kind of loop:

		CameraEventType eventType = GP_EVENT_UNKNOWN;
		while (eventType != GP_EVENT_FILE_ADDED) {
			gp_camera_wait_for_event(camera,10, &eventType, &eventData, SimpleCam::context);				
		}

I don't need to close and reconnect.  However, it only works with 23 cameras out of the 40 cameras I have connected to my computer.  Then if I disconnect and reconnect I can get the remaining 17 cameras.

msmeissn, 2017-04-07:
Only with 23 cameras in parallel...  or how are you calling it? It is hard to say where the issue is  :/
are there 40 programs running?

punkoffice, 2017-04-08:
This is serial (no multi-threading).  I am running "wait for event" in a for loop for all 40 cameras.  So it will just block for the 1st camera, then when I take photos with all 40 cameras all the waits will instantly trigger but when it gets to camera 23 it blocks forever.

The code is like:
for i = 0 to 40 do
    wait for event
    download photo
    delete photo
end for