gphoto/libgphoto2 #155: Any examples with multiple cameras?

punkoffice, 2017-04-06:
Hello.  I'm using libgphoto2 to control 40 cameras on a server.  I've tried to run commands to each camera using its own thread.  This is very quick but sometimes I get PTP errors but it will still work if I keep polling.  Other times I get a segfault around libusb.  Is it possible to use libgphoto2 for multiple cameras with multiple threads?   I also tried a single thread.  Its much slower but I still get issues downloading the photos off each camera.  I have to use gp_camera_wait_for_event for each camera and it blocks forever after the 23rd camera (never receives the GP_EVENT_FILE_ADDED event). Is there any example code for using libgphoto2 for multiple cameras?

msmeissn, 2017-04-07:
currently there is no such code yet, I could probably write a sample, but i did not do much thread programming my own either.

punkoffice, 2017-04-08:
I am not sure if there is a limitation with libusb or if I'm writing bad code.  But if there is a limitation that is official then I can work around it.  Maybe what I'm trying to do is impossible, I don't know.  If you want to test you can use 2 cameras and just run a multithreaded command in an endless loop.  You will eventually come to a segfault