gphoto/libgphoto2 #156: How to download from Canon RAW + JPG files in one go?

Sija, 2017-04-06:
Canon DSLRs have a [setting](https://github.com/gphoto/libgphoto2/blob/fe35d7878bb6e77692bb76cfd7c62b3cbb1268ad/camlibs/canon/library.c#L317) which allows to save RAW files with additional JPG.

How could I download JPG files along with RAW using for instance `gp_camera_capture`?

msmeissn, 2017-04-06:
this is not possible. after camera_capture returns 1 file, use gp_camera_wait_for_event to wait for a FILEADDED event to download the secondary image.

Sija, 2017-04-06:
So then I'll get 2 `FILEADDED` events, one for each file (RAW + JPG), right?

msmeissn, 2017-04-06:
Not if you use gp_camera_capture, there will be the first image returned by capture_image and the second via wait_event. 

you can however use gp_camera_trigger_capture ... and get both of the images via gp_camera_wait_for_event then.

Sija, 2017-04-06:
Now I see, thanks for explanation!

closed 2017-04-06 by Sija