gphoto/libgphoto2 #157: redefinition of typedef ‘GPPort’

ryandesign, 2017-04-11:
Hello, I'm the maintainer of gphoto2 in the MacPorts package management system. libgphoto2 2.5.13 fails to build with GCC 4.2 (the default compiler on Mac OS X 10.6) with this error:

    error: redefinition of typedef ‘GPPort’ 

This is a regression; 2.5.12 built fine.

I can work around the problem by requiring a newer compiler (e.g. clang) on Mac OS X 10.6 and earlier, but I'd rather not have to do that, especially since clang is no good at generating PowerPC code, so that would affect users of older systems. Seems like redefining a type is wrong, and it should be fixed properly.

Here's the MacPorts project's bug report on this issue: https://trac.macports.org/ticket/53965

msmeissn, 2017-04-12:
sorry for that. you can remove the typedef line from camlibs/pentax/pslr_scsi.h

closed 2017-04-12 by msmeissn

ryandesign, 2017-04-13:
Thanks, that works.