gphoto/libgphoto2 #161: error: could not capture image halfway thru my timelapse.

elijahlucian, 2017-04-20:
Hello, I tried making a timelapse script and all seemed well, but then suddenly after 270 frames, i got the error: 

ERROR: Could not capture image. 
ERROR: Could not capture.

I am using an a6000, but gphoto2 also doesn't seem to recognize my camera for some reason. it just calls it a "USB PTP Class Camera"  (rude)

so, what's the deal? is it my camera, did I set it up wrong?

elijahparker, 2017-04-20:
I've had Sony cameras lock up when polling for events.  I've disabled this
in my fork: https://github.com/timelapseplus/libgphoto2

I'm not sure what the best overall solution is.  My fork disables focus
checking, which for time-lapse is what I want, but it's not for everyone.

elijahlucian, 2017-04-20:
Interesting, yeah that should work for me, I set everything manually beforehand, essentially all I want it to do is hit the shutter and grab the image.

How do i disable a part of the installation. Sorry I ma suuuuper new to this, also I'm doing it on my raspberry pi...

elijahparker, 2017-04-20:
I would just clone my fork and then build it from source. You only need to
build libgphoto2, gphoto2 will then use the updated library (verify this
with gphoto2 --version before and after installing the update)

elijahlucian, 2017-04-21:
So i just realized that the problem occours right after it tries "deleting 'capt0000.jpg from folder '/'

my print command that says the print at the end of the sequence doesn't happen, meaning that it's not trying to focus, but instead it's unable to delete the file from the camera. 

orrrrrrrrr. not . as i was writing this, it seems like it just decided to give me the error after the print command. 

fack.

elijahparker, 2017-04-21:
I'm not sure where the issue is in that case, but do you have the very
latest libgphoto2? Many Sony-specific improvements are very recent, so if
you installed using apt-get, it's probably an outdated version. I would
recommend you build it from source, either from my fork or from the main
one.

Or, if you want to try a different approach, there's wifi:
https://github.com/timelapseplus/node-sony-camera

elijahlucian, 2017-04-21:
sorry i'm really new to this and trying to learn. bit overwhelming sometimes...

ok ill try and figure out how to build it from the source haha. I will definately check out this wifi thing soon.

elijahlucian, 2017-04-21:
OK just an update here. I re-installed Raspbian, and put gphoto on it, didn't change anything but the localization. 

I managed to get 1400 photos out of my 1800 script (updated it for 30fps but i digress)

After 1427 it gave me the double "could not capture" error all the way to 1800. Now When i look at my camera and press the "play" button on the back (to view images) it says "Writing to memory card, unable to operate" 

so this seems like a bug in the memory card writing process... rather than the focus. - unless somehow the two are linked. any idea how i could get past this? maybe it just gets confused and needs more time between C commands? like hitting delete before the file is fully written or something, could lock it up.