gphoto/libgphoto2 #162: sony a6000 possible to just use shutter press and leave files on the camera?

elijahlucian, 2017-04-21:
i have been having a ton of problems trying to get this to work on my a6000, i was wondering if there was a simple shutter press option for this, one where it just shoots a "shutter" button to the camera, it seems simple, and if I knew where to look in the code, i might be able to hack together something that could work. Just leave all the pics on the card and let the camera do all the work. 

then again i don't really know what magic is happening behind the scenes here...

elijahparker, 2017-04-21:
Unfortunately, no. This is the most annoying "feature" of Sony USB control
-- it doesn't allow saving to the camera while connected via USB.

This, however, is possible with wifi, and shorter intervals are possible
due to this as well.

msmeissn, 2017-04-21:
so lets close as "unsupported"

closed 2017-04-21 by msmeissn