gphoto/libgphoto2 #164: Nikon KeyMission 360 (PTP mode) picture resolution

ibandominguez, 2017-04-30:
Hi everyone, 

Beforehand I wanna thank you for all the great work that has been put into this library. 

I have been playing with a Nikon KeyMission 360 for a experimental project, it all works awesome but I cant find a way to change the image resolution, the camera takes quite long to grab the picture and I guess it has to do with the image size.

Do anybody know how to do reduce image resolution?

Cheers, 
Ibán

msmeissn, 2017-04-30:
there are quite some unknown properties, but none seem to show image size ... is the phone/windows access softrware able to do different resolutions?

ibandominguez, 2017-05-02:
Thank you for your quick response. 

We ended up processing the image before sending it and It all worked out fine.

Thanks for putting this library out there, keep up the good work!
cheers

closed 2017-05-02 by ibandominguez