gphoto/libgphoto2 #165: memory leak

teo666, 2017-05-05:
hi
I have started play with libgphoto2 but I am stumbled in a memory leak.
I have wrote a simple snippet of code and i tested it with valgrind, giving me a memory leak. The question is: is that memory leak normal? What's wrong in my code or in me?

```
#include "gphoto2/gphoto2-camera.h"

int main() {
    CameraList * clst = NULL;
    GPContext * context = NULL;

    gp_list_new(&clst);
    gp_camera_autodetect(clst,context);
    gp_list_free(clst);
    return 0;
}
```

note: it gives me problem with camera connected and not

VALGRIND OUTPUT with `valgrind --leak-check=full`
```
==4510== Memcheck, a memory error detector
==4510== Copyright (C) 2002-2015, and GNU GPL'd, by Julian Seward et al.
==4510== Using Valgrind-3.12.0 and LibVEX; rerun with -h for copyright info
==4510== Command: ./cameraleak
==4510== 
==4510== 
==4510== HEAP SUMMARY:
==4510==     in use at exit: 128,768 bytes in 41 blocks
==4510==   total heap usage: 6,244 allocs, 6,203 frees, 77,618,641 bytes allocated
==4510== 
==4510== 4,096 bytes in 1 blocks are definitely lost in loss record 14 of 20
==4510==    at 0x4C2AF1F: malloc (in /usr/lib/valgrind/vgpreload_memcheck-amd64-linux.so)
==4510==    by 0x405799F: ???
==4510==    by 0x4055EA0: ???
==4510==    by 0x680923C: ???
==4510==    by 0x6807A92: ???
==4510==    by 0x67FF2CC: ???
==4510==    by 0x65F469B: ???
==4510==    by 0x546388B: ??? (in /usr/lib/libgphoto2_port.so.12.0.0)
==4510==    by 0x588DF09: ??? (in /usr/lib/libltdl.so.7.3.1)
==4510==    by 0x588D8AB: ??? (in /usr/lib/libltdl.so.7.3.1)
==4510==    by 0x588E488: lt_dlforeachfile (in /usr/lib/libltdl.so.7.3.1)
==4510==    by 0x5463CB8: gp_port_info_list_load (in /usr/lib/libgphoto2_port.so.12.0.0)
==4510== 
==4510== 4,096 bytes in 1 blocks are definitely lost in loss record 15 of 20
==4510==    at 0x4C2AF1F: malloc (in /usr/lib/valgrind/vgpreload_memcheck-amd64-linux.so)
==4510==    by 0x405799F: ???
==4510==    by 0x405712B: ???
==4510==    by 0x406280B: ???
==4510==    by 0x40609B2: ???
==4510==    by 0x4055A3F: ???
==4510==    by 0x40636D6: ???
==4510==    by 0x405DCEB: ???
==4510==    by 0x680925A: ???
==4510==    by 0x6807A92: ???
==4510==    by 0x67FF2CC: ???
==4510==    by 0x65F469B: ???
==4510== 
==4510== 8,192 bytes in 2 blocks are definitely lost in loss record 17 of 20
==4510==    at 0x4C2AF1F: malloc (in /usr/lib/valgrind/vgpreload_memcheck-amd64-linux.so)
==4510==    by 0x405799F: ???
==4510==    by 0x4055EA0: ???
==4510==    by 0x784523C: ???
==4510==    by 0x7843A92: ???
==4510==    by 0x783B2CC: ???
==4510==    by 0x65F2CCB: ???
==4510==    by 0x546606E: gp_port_set_info (in /usr/lib/libgphoto2_port.so.12.0.0)
==4510==    by 0x4E3E654: gp_abilities_list_detect (in /usr/lib/libgphoto2.so.6.0.0)
==4510==    by 0x4E40ADB: gp_camera_autodetect (in /usr/lib/libgphoto2.so.6.0.0)
==4510==    by 0x4007EC: main (in /home/teo/Desktop/cameraleak)
==4510== 
==4510== 8,192 bytes in 2 blocks are definitely lost in loss record 18 of 20
==4510==    at 0x4C2AF1F: malloc (in /usr/lib/valgrind/vgpreload_memcheck-amd64-linux.so)
==4510==    by 0x405799F: ???
==4510==    by 0x405712B: ???
==4510==    by 0x406280B: ???
==4510==    by 0x40609B2: ???
==4510==    by 0x4055A3F: ???
==4510==    by 0x40636D6: ???
==4510==    by 0x405DCEB: ???
==4510==    by 0x784525A: ???
==4510==    by 0x7843A92: ???
==4510==    by 0x783B2CC: ???
==4510==    by 0x65F2CCB: ???
==4510== 
==4510== LEAK SUMMARY:
==4510==    definitely lost: 24,576 bytes in 6 blocks
==4510==    indirectly lost: 0 bytes in 0 blocks
==4510==      possibly lost: 0 bytes in 0 blocks
==4510==    still reachable: 104,192 bytes in 35 blocks
==4510==         suppressed: 0 bytes in 0 blocks
==4510== Reachable blocks (those to which a pointer was found) are not shown.
==4510== To see them, rerun with: --leak-check=full --show-leak-kinds=all
==4510== 
==4510== For counts of detected and suppressed errors, rerun with: -v
==4510== ERROR SUMMARY: 4 errors from 4 contexts (suppressed: 0 from 0)
```

Sija, 2017-05-05:
See [gphoto2.cr](https://github.com/Sija/gphoto2.cr) shard.

kadler15, 2017-05-05:
Not that this is causing your problem, but aren't you missing
```
context = gp_context_new();
```

msmeissn, 2017-05-06:
(context can be NULL)

I can confirm the leaks. These are leaks in libusb mostly, have to check how to avoid them.

Ruuttu, 2017-12-27:
This is troublesome, as there doesn't seem to be any possible workaround for the leak (?) when one needs to repeatedly check if a camera is present.

trumpetlicks, 2019-08-13:
I hope to just be another user of this API with a use-model of continuously checking camera alive status, so that this hopefully gets more attention.  I can definitely see (and I am using the gphoto2 python bound interface) memory climb when I initialize a camera.  Even if I close it prior to attempting to open again, I still see a leak.

msmeissn, 2019-10-12:
there was an issue in the python bindings, solved in another bug ... can you check if an update of the bindigns fixes it?

thexangle, 2020-08-12:
I know this thread is quite old but we can confirm that this leak is still present in 2.5.25, on raspberry pis at least. The code snippet provided by @teo666 still produces a leak, even when no cameras are connected... 
Any suggestion to work around this issue ? Calling gp_camera_autodetect every 2 sec in a loop causes a leak of approximately 30 Mb/h

msmeissn, 2020-09-19:
can you share your code that reproduces the leak?

or can you run it in valgrind --leak-check=full --show-leak-kinds=all  ./leakprogram

thexangle, 2020-09-30:
Unfortunately valgrind is not working so well on raspberry pis... However, when I monitor the process with htop, it shows a constant (and slow) increase in memory consumption... but only when no camera is connected. When a camera is connected OR when I remove the autodetect call, the leak disappears. 

Here is the code we use. It is called in a loop every 2 seconds. 
```
    GPContext* context = ThreadContext::GetGPhoto2Instance((int)ThreadContext::AUTO_DETECT_THREAD);
    CameraList	*list;
    int ret = gp_list_new (&list);
    if (ret >= GP_OK && gp_camera_autodetect (list, NULL) >= GP_OK)
    {
        //browse the cameras detected by gphoto2
        for (int i = 0; i < gp_list_count(list); i++) {
            const char *name, *value;
            gp_list_get_name (list, i, &name); //Canon EOS 100D (example)
            gp_list_get_value (list, i, &value); //usb:001,005 (example)

            //Check if we have this cam on the list based on its usb:xxxx address
            CameraIterator searchedCamIter = m_cameras.find(value);
            if (searchedCamIter == m_cameras.end()){ //if we dont...
		CameraPtr camera = std::make_shared<GPhoto2Camera>();
		GPhoto2CameraPtr gpcamptr = std::dynamic_pointer_cast<GPhoto2Camera>(camera);
                gpcamptr->SetStatusListener(GetStatusListener());
                Camera* targetDevice;
                ret = open_camera(&targetDevice, name, value, context); //Open USB connection
                if ( ret < GP_OK){
                    printf("[CameraManager] Cannot open camera: %s:%s - will retry later\n", name, value);
                }
                else{
                    ret = gpcamptr->Init(context, targetDevice); //Initialize it
                    if (ret >= GP_OK)
                    {
                        (m_cameras)[value] = camera; //Add it to the list
						gpcamptr->NotifyCameraConnected();
                    }
                }
            }
            else{ //We already know this cam. Flag it as detected
                searchedCamIter->second->ToggleDetected(true);
            }
        }
    } 
    gp_list_free (list);
```

msmeissn, 2020-10-05:
so it wont enter this for() loop and even leaks there.

thexangle, 2020-10-12:
correct, the leak seems to happen when `gp_list_count() ` returns 0